espressif/esp-idf #11929: ESP32-H2 HA_on_off_light does not connect to the Zigbee network.

luigidoria, 2023-07-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1805-g9a1cc59338

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have two boards esp32-h2-DevKitM-1, run light_sample, coordinator create a network and end device connect to it.

### What is the actual behavior?

Coordinator create a network, but the end device can't connet and return Network steering was not successful

### Steps to reproduce.

Terminal 1 - HA_on_off_switch:
1. idf.py set-target esp32h2 
2. idf.py -p /dev/ttyACM0 erase-flash
3. idf.py build 
4. idf.py -p /dev/ttyACM0 flash monitor

Terminal 2 - HA_on_off_light:
1. idf.py set-target esp32h2 
2. idf.py -p /dev/ttyACM1 erase-flash
3. idf.py build 
4. idf.py -p /dev/ttyACM1 flash monitor

### Build or installation Logs.

```plain
Coordinator:

I (366) main_task: Calling app_main()
I (376) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (396) phy: phy_version: 200,0, 8a31b65, Jun  6 2023, 20:13:10
I (396) phy: libbtbb version: 98b760f, Jun  6 2023, 20:13:22
I (406) main_task: Returned from app_main()
I (516) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 23, status: -1
I (516) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (516) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (9146) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 54, status: 0
I (9146) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: c0:6c:f7:fe:ff:f9:55:60, PAN ID: 0xb403, Channel:19)
I (9616) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 54, status: 0
I (9616) ESP_ZB_ON_OFF_SWITCH: Network steering started

End device:

I (367) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (407) phy: phy_version: 200,0, 8a31b65, Jun  6 2023, 20:13:10
I (407) phy: libbtbb version: 98b760f, Jun  6 2023, 20:13:22
I (407) main_task: Returned from app_main()
I (497) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 23, status: -1
I (497) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (497) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (41187) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
```

xieqinan, 2023-07-25:
Hi @luigidoria ,

 Could you try using the UART to monitor the Zigbee device? This is because the ACM port requires JTAG driver support.

luigidoria, 2023-07-26:
This resolve my problem thanks.

closed 2023-08-31 by chshu