espressif/esp-idf #11930: how to increase ipc0 Stack ?

diplfranzhoepfinger, 2023-07-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

.

diplfranzhoepfinger, 2023-07-24:
[image]

ESP-Marius, 2023-07-26:
1. Probably yes, did it?
2. Like you already found, there is a config for this: [CONFIG_ESP_IPC_TASK_STACK_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-ipc-task-stack-size)

ESP-Marius, 2023-08-14:
Marking this as resolved.

closed 2023-08-14 by ESP-Marius