espressif/esp-idf #11932: ESP32-C6 Cannot Call esp_sleep_enable_ext0_wakeup

schwefeljm, 2023-07-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1805-g9a1cc59338

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Per the documentation at https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/sleep_modes.html?highlight=esp_sleep_enable_ext0_wakeup , I expect to be able to call the 'esp_sleep_enable_ext0_wakeup' function.

### What is the actual behavior?

A -Werror=implicit-function-declaration is triggered during build if esp_sleep_enable_ext0_wakeup is called.

### Steps to reproduce.

1. Create new project from 'template_app'
2. main.c follows

#include <stdio.h>
#include "esp_sleep.h"
#include "driver/gpio.h"

void app_main(void)
{
    esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 1);
}

3. Build project
4. Fails

### Build or installation Logs.

```plain
/home/jschwefel/esp/projects/esp32-iot-wifi/main/sleep.c:113:12: error: implicit declaration of function 'esp_sleep_enable_ext0_wakeup'; did you mean 'esp_sleep_enable_ext1_wakeup'? [-Werror=implicit-function-declaration]
  113 |     return esp_sleep_enable_ext0_wakeup(GPIO_NUM_0, 1);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |            esp_sleep_enable_ext1_wakeup
```

### More Information.

Verified the same behavior in **v5.1-476-g3187b8b326**

This is my first bug report on any open source project. Please advise if I did anything improper or how to better search the issue tracker if this issue exists as I was unable to find it.

esp-lis, 2023-07-24:
@schwefeljm On the esp32c6 chip, I recommend you to use gpio wakeup, which includes digital GPIO wakeup (power off when no digital peripheral is configured) and LPIO wakeup (power off when digital peripheral is configured), see the following code for details:
```c
#if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
    /* Initialize LPIO */
    rtc_gpio_init(gpio_num);
    rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pullup_dis(gpio_num);
    rtc_gpio_pulldown_dis(gpio_num);

    /* Enable wake up from LPIO wakeup */
    rtc_gpio_wakeup_enable(gpio_num, GPIO_INTR_LOW_LEVEL);
#else
    /* Enable wake up from GPIO */
    gpio_wakeup_enable(gpio_num, GPIO_INTR_LOW_LEVEL);
#endif
```

esp32c6 also supports **ext1** wakeup, which will be synced to github's master branch at a later date.

schwefeljm, 2023-07-25:
> @schwefeljm On the esp32c6 chip, I recommend you to use gpio wakeup, which includes digital GPIO wakeup (power off when no digital peripheral is configured) and LPIO wakeup (power off when digital [...]

@esp-lis 
Thanks for the info. I did find use of the ext1 wakeup did work. I am still in the "playing around/learning" stage. Was not sure if it was a documentation or hardware issue. Figured I would bring it to Espressif's attention.

Thank you,

Jason

closed 2024-09-10 by Sherry616