espressif/esp-idf #11934: [Crash][v5.1] InstrFetchProhibited when characteristic subscribed

chipweinberger, 2023-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

No crash

### What is the actual behavior?

Crash

### Steps to reproduce.

Subscribe to BLE chr. Probably other conditions as well

### Debug Logs.

```plain
I (849163) pd ble: GAP subscribe event:
I (849163) pd ble:    gap conn handle: 1
I (849163) pd ble:    att handle:      31 (Ble Midi)
I (849163) pd ble:    cur indicate:    0
I (849163) pd ble:    cur notify:      1
I (849163) pd ble:    prev inficate:   0
I (849163) pd ble:    prev notify:     0
I (849163) pd ble:    reason:          Write
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xabba1234  PS      : 0x00060e30  A0      : 0x8209e591  A1      : 0x3fcc10e0  
A2      : 0x3fcc11c8  A3      : 0xabba1234  A4      : 0x3fca893c  A5      : 0x3c2e6b00  
A6      : 0x0000000a  A7      : 0x00000201  A8      : 0x8201a7d6  A9      : 0x3fcc10b0  
A10     : 0x00000001  A11     : 0x0000000a  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x485895cd  A15     : 0x485895cd  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0xabba1234  LBEG    : 0x4201a204  LEND    : 0x4201a21e  LCOUNT  : 0x00000000  
0x4201a204: gap_subscription_add(unsigned short, unsigned short, bool, bool) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:198

0x4201a21e: gap_subscription_add(unsigned short, unsigned short, bool, bool) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:209

Backtrace: 0x6bba1231:0x3fcc10e0 0x4209e58e:0x3fcc1160 0x4209e5ed:0x3fcc1190 0x420a03ae:0x3fcc11c0 0x420a15e3:0x3fcc1220 0x420a869b:0x3fcc1280 0x420a96b5:0x3fcc12b0 0x42099bd9:0x3fcc12f0 0x420999ea:0x3fcc1320 0x42098b88:0x3fcc1360 0x4037aa69:0x3fcc1390 0x4037aa0e:0x3fcc13c0 0x4201969c:0x3fcc13f0 0x40387012:0x3fcc1420
0x4209e58e: ble_gap_call_event_cb at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:758

0x4209e5ed: ble_gap_call_conn_event_cb at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:784

0x420a03ae: ble_gap_subscribe_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:6409

0x420a15e3: ble_gatts_subscribe_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:623
 (inlined by) ble_gatts_subscribe_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:618
 (inlined by) ble_gatts_clt_cfg_access at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:774

0x420a869b: ble_att_svr_write at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:536

0x420a96b5: ble_att_svr_write_handle at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:567
 (inlined by) ble_att_svr_rx_write at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2013

0x42099bd9: ble_att_rx at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:514

0x420999ea: ble_hs_hci_evt_acl_process at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1063

0x42098b88: ble_hs_process_rx_data_queue at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:241
 (inlined by) ble_hs_event_rx_data at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:538

0x4037aa69: npl_freertos_event_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:450

0x4037aa0e: ble_npl_event_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:246

0x4201969c: start_nimble_port_run(void*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:650

0x40387012: vPortTaskWrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

chipweinberger, 2023-07-23:
Note: the PC is 0xabba1234

ChatGPT:

> In the context of the ESP32 and its exception handling, 0xabba1234 is a special value used to mark the top of the stack when an exception occurs.
> 
> If you see this value in a stack trace, it means that you've reached the end of the relevant stack information. It is used to distinguish between valid stack data and "noise" that might be present beyond the end of the stack.
> 
> This specific value is chosen because it's distinctive and unlikely to be confused with valid memory addresses or other data.

SumeetSingh19, 2023-07-24:
Hi @chipweinberger ,
According to the logs you have sent, the crash appeared to have happened in `gap_subscription_add`. Since this is created by you I will need some assistance to debug this further.
Could you please answer these questions for me?
1. I can see the line numbers that the backtrace indicates, however, I can't figure that out in the function definition you have attached. Could you provide that information.
2. If you can send the entire app, it would become easy to reproduce on my end, otherwise, you could mention the steps to reproduce.
3. How did you declare the array `subscribedClients`? Is it static or dynamic?

I can see that you have written a logic to remember subscribed clients, might I add that NimBLE stack itself remembers all the clients that are subscribled, and if you want to send a notification/indication whenever the characteristic is written to, you could simply call `ble_gatts_chr_updated` with the attribute handle, the API will notify all the clients that have subscribed

chipweinberger, 2023-07-24:
Thanks for the reply.

> the crash appeared to have happened in gap_subscription_add

Are you sure? It is a `InstrFetchProhibited` which seems kind of strange for me to trigger.

> I can see the line numbers that the backtrace indicates, however, I can't figure that out in the function definition you have attached. Could you provide that information.

Okay. I've added lines numbers to the original code above. 

> How did you declare the array subscribedClients? Is it static or dynamic?

It is dynamic, but I am 100% sure it is allocated. This appears to be a heap corruption bug in my opinion. I'll have some heap integrity checks around my code.

>  might I add that NimBLE stack itself remembers all the clients that are subscribled

I did not know that!!!!

SumeetSingh19, 2023-07-26:
Hi @chipweinberger ,
I will need to reproduce the issue on my end to debug it any further, could you provide the application you have used?

chipweinberger, 2023-07-26:
I cannot provide application code sadly. 

If you cannot reproduce it is okay, I will attempt to fix it myself soon.

chipweinberger, 2023-08-09:
🎉🎉🎉 Found the bug. It was in my code.

It was memory corruption. Always tricky to know the culprit when that happens.

I'm glad I told you to wait to investigate =)

```
enum pd_ble_chr_t {
    PD_BLE_CHR_NULL = 0,
    .....
    PD_BLE_CHR_MIDI = 10,
};
#define PD_BLE_CHR_COUNT 10 // BUG: THIS HAD TO BE 11
```

closed 2023-08-09 by chipweinberger