espressif/esp-idf #11936: Wifi roaming does not work (802.11k 802.11v 802.11r or client-side roaming) in STA mode

xoores, 2023-07-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to get ESP32-S2 to have as stable wifi connection as possible - that means preiodic monitoring of network and roaming to neighboring AP if necessary (e.g. current AP has RSSI below certain threshold).

I thought I would use the [roaming example](https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c), but ESP32 just refuses to roam:
```
00:03:29.421|  +WIFI-ROAM: RSSI threshold hit (-81dBm), will try to look for better AP
00:03:29.530|  +WIFI-RRM: Sending neighbor report request
00:03:29.534|  I (13122) wpa: action frame sent
00:03:29.543|  +WIFI-RRM: Received neighbor report [LEN=31]
00:03:29.552|  0000 | 01 34 0D D0 21 F9 5D 39  36 8F 08 00 00 51 06 07 |.4..!.]9 6....Q..|
00:03:29.560|  0010 | 34 0D 78 45 58 9F 27 D1  8F 58 00 00 51 0B 0E    |4.xEX.'. .X..Q.. |
00:03:29.561|  
00:03:29.569|  +WLAN-RMM: Neigbor report bssid=d0:21:f9:5d:39:36 info=0x88f op_class=81 chan=6 phy_type=7
00:03:29.580|  +WLAN-RMM: Neigbor report bssid=78:45:58:9f:27:d1 info=0x588f op_class=81 chan=11 phy_type=14
00:03:29.582|  I (13172) wpa: action frame sent
00:03:29.590|  +WIFI-RRM: Sending transition mgmt query [RES=0]
00:03:29.595|  I (13182) wpa: scan issued at time=169747532768
00:03:29.714|  I (13302) wpa: scan done received
00:03:29.718|  I (13302) wpa: action frame sent
~~ after this nothing relevant happens, no roaming or WLAN-related events... ~~
```

I have Ubiquiti APs and other devices on the same network roam without any problem.  I also tried to change `REASON_FRAME_LOSS` to `REASON_RSSI` in `esp_wnm_send_bss_transition_mgmt_query()` but nothing has changed. The example mentioned explicitely that the station will be forced to roam but this seems to be false as the station will remain on the same AP.

I also enabled 802.11r in the network but no difference in ESP32 behavior. After that I was trying to figure out how to force ESP32 to initiate client-side roaming without dropping connection altogether (by calling `esp_wifi_disconnect()` and then `esp_wifi_connect()`) which seems rather unfortunate.

Right now it seems that ESP32 will roam only when the current connection fully drops, which is unfortunate.

## So to the question:
Is there any way to either get 802.11k/v/r working or debug why it is not working? Or is there any way to force client side roaming so I can have reliable connection without having to wait for the current connection to die (or kill it by calling disconnect)?

Has anyone managed to get ESP32 to have reliable and stable connection without having to manually scan every few seconds and then manually call disconnect()/connect() when better AP becomes available?

KlausPopp, 2023-10-09:
Hello @xoores : I experienced the same problem. I am using ESP-IDF v5.1.1 and AVM Fritzbox mesh. I used a Raspberry PI and wireshark as a network sniffer. What I saw, is that the ESP sends a "BTM transition query" to the AP, but the AP ignores it. Thats why the STA does not roam. In other situations, the AP sends a BTM transition request (without a prior query from the STA), which causes the STA to roam.

For me, the big problem is, that it seems not possible with the current Wifi stack to perform a SEAMLESS roaming (i.e. without a connection loss when roaming). The wifi stack always disconnects from the old AP, and then connects to the new AP, which takes at least 2 seconds.

Espressif Team: Do you have plan to change this in the future?

xoores, 2023-10-09:
Yeah it is a pain and so far nobody from Espressif did reach out to me so I don't think they are even aware of this issue... :shrug: 

Right now I have to scan every 30 seconds or so and if there is a better AP, I have to disconnect (tear down all the connections, MQTT etc) and re-connect to the new AP using its BSSID.

kapilkedawat, 2023-10-10:
Hello @xoores @KlausPopp , ESP devices right now only supports network assisted roaming(ie AP will tell the device on which it needs to move on using BTM request and device will blindly follow it).

We are in process of adding one application code which will do active roaming(essentially more scans and more trigger points to start roaming), that code will be available in esp-idf in coming months.

However please note that, in all these scenario station will always disconnect first from the AP then connect to the other AP. 

Also if AP doesn't reply to btm request, station won't do anything in network assisted roaming(since station is supposed to follow AP's directions).  However there will be different config options in station initiated roaming where station will take different decisions based on configuration options.

KlausPopp, 2023-10-10:
Hello @kapilkedawat : Thanks for your reply. Why is it necessary to disconnect from the old AP first before connecting to the new AP. I monitored an iPhone while roaming. I does it absolutely seamless, no IP interruption occurs. I noticed that the IPhone reassociates with the new AP while it is still associated with the old AP. After some seconds, the old AP sends a disassociation request to the iPhone due to missing ACKs.
Would such an approach be possible with ESP wifi stack?

kapilkedawat, 2023-10-10:
Hi @KlausPopp , This methodology is called [make-before-break](https://source.android.com/docs/core/connect/wifi-network-selection#make-before-break). However we don't have plans to implement it in near future. Only break-before-make is available with ESP devices.

KlausPopp, 2023-10-10:
Hi @kapilkedawat : Ok, could you explain why the feature is not on the raodmap. Is it due to project priority reasons, or due to a technical restriction in ESP-Hardware?

Saicharan67, 2023-11-17:
Hi @xoores have you got any leads on this issue ?... coz am facing the same issue and observed that the connected AP keeps reporting itself as the best candidate as the neighbour report is only consisting of info about connected AP

KlausPopp, 2023-11-17:
Hi @xoores : Do you know if make-before-brake isn't possible due to a hardware limitation? Or is it "just" a matter of the wifi driver?

xoores, 2023-11-17:
Hi @KlausPopp, I honestly don't know :shrug: It all depends whether the ESP is physically capable talking on (possibly) two different channels...

Since @kapilkedawat told us that they don't plan to implement it and only break-before-make is available, I personally suspect that it might very well the case of HW limitation. But that is only my guess, I don't have any hard data to back that claim up. But at the very least it seems that whole ESP-IDF WLAN stack was *not* really written in high availability in mind so implementing proper mechanism would mean rewriting big chunk of their codebase and that is why they are dodging any hard info when it comes to this :sweat_smile: 

But that is just my take on the matter and I only hope to be corrected by someone from Espressif :slightly_smiling_face:

kapilkedawat, 2023-11-20:
Hello @KlausPopp and @xoores,

Implementing "make before break" will consume significantly more memory with seemingly very little benefit, making it less important from the perspective of IoT devices (it also causes network disruption). However, we are currently working on the next generation ESP roaming app, which will offer users much more control and significantly reduce roaming time. It will utilize both BTM and legacy roaming mechanisms.

Please note that network disruption will always occur in roaming in single PHY systems since the device goes off-channel multiple times for:
1. scanning
2. connection

Additionally, it has to wait for WPA2 handshake as well as DHCP to resume data transfer.

KlausPopp, 2023-11-20:
Hello @kapilkedawat : Thanks for this info. In my application, the main issue is that I need to restart all existing socket connections in case of roaming. Do you know if there is a way to keep the socket connections intact during roaming?

Saicharan67, 2023-11-20:
Hi @kapilkedawat in my case...ESP32 seem to connect randomly to the router or a repeater but not the one with the strongest WiFi-signal
So my question is:
is there a way to force the EPS32 to connect to either

the strongest signal
or
to a specified MAC-Adress?

xoores, 2023-11-20:
Hi @kapilkedawat, yeah it may consume some memory but since you can get ESP with 8M of SRAM, it seems like a non-issue. :slightly_smiling_face:

Also even while it may consume more memory, I would trade that for more battery as few of my IoT nodes have quite unstable wifi connection and when they push data through the MQTT, it will disconnect several times, having to stay up way too long... :shrug: 

@Saicharan67 You can for sure use BSSID of AP to connect to - that is what I'm doing right now as the roaming does not work.

kapilkedawat, 2023-11-20:
@KlausPopp, IMO, it's always better to restart sockets in case of roaming (you may encounter more issues in the field if you don't do that :) ). We could have masked the [ROAMING disconnect event](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv4N17wifi_err_reason_t19WIFI_REASON_ROAMINGE), which will also prevent LWIP from doing an interface stop, and current sockets may continue working. However, restarting the sockets ensures that everything is refreshed, and things will work as they are supposed to be.

@xoores, as I explained earlier, for ESP devices, even after the implementation of "make before break," it will cause a network disruption since we have a single mac/rf, so multi-channel concurrency will always cause issues in connectivity. Also, the device has to disconnect before any data frame transmission (hardware limitation of station key indexes available). So, the time saved here is too low to even try that (ieee802.11 exchange generally happens under 100ms).

If you are seeing zero disruption in Apple devices, they may be using STA+STA concurrency, where the chip is capable of making two station connections simultaneously on different channels (essentially a dual mac WLAN chip).

@Saicharan67, we have a [configuration](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_sort_method_t) for that, which is by default set to RSSI. However, please make sure the [scan_method](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_method_t) is set to FULL_SCAN. If you are experiencing any issues with that, feel free to open a new issue. Please also attach a sniffer capture and debug logs to that.

xoores, 2023-11-20:
@kapilkedawat Yeah, I know that there will be slight outage, but that is not a problem as this will be transparently handled by the TCP stack (with retransmission etc).

So small outage in a range of few hundreds ms should not a problem for applications :slightly_smiling_face: There is no need to have dual wlan chip for this - that would also have to be supported by HW and I get that this would make it much more energy & memory hungry.

Problem is when there is a shutdown of all sockets during the change of AP. That way you have to connect & auth to server again (MQTT for example) and you have to basically re-do all the work.

KlausPopp, 2023-11-20:
@kapilkedawat as @xoores mentioned, the main problem is the disconnection of sockets while roaming.

Consider the following use case:
ESP32 device is installed on a forklifter. Forklifter is constantly moving between access points. Now a client tries to load recorded data from the device via http. This operation will fail when the device is roaming while the http request is active. Of course, I can think about workarounds, but in my opinion, this should work. 

So can you consider to change wifi stack NOT to close underlying LWIP connections while roaming?

Saicharan67, 2023-11-20:
@KlausPopp Thanks for the work around.

Currently I am evaluating the roaming_example in IDF. What i found out while debugging is that...the responce STA get from AP (ORBI )after sending BTM transition request contains information only about connected AP which causing it to not roam by saying Already associated with the preferred candidate.

`I (1727) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (1737) roaming_example: setting rssi threshold as -50

I (1737) roaming_example: RRM supported
I (1747) roaming_example: BTM supported
I (3357) roaming_example: esp_bss_rssi_low_handler:bss rssi is=-59
W (3357) wpa: action frame sent
W (3357) wpa: RRM: Notifying neighbor report (token = 1)
I (3357) roaming_example: rrm: neighbor report len=47
I (3367) roaming_example: 0x3fc9a282   01 34 15 9a a6 7e 60 10  0e 87 d8 00 00 0                                                                                                                                                             ..~`.........|
I (3377) roaming_example: 0x3fc9a292   06 03 01 05 00 03 01 00  34 15 9a a6 7e 5                                                                                                                                                             .....4...~Y.\|
I (3387) roaming_example: 0x3fc9a2a2   87 d8 00 00 00 03 0e 06  03 01 05 00 03 0                                                                                                                                                             ............|
I (3397) roaming_example: RMM neigbor report bssid=9a:a6:7e:60:10:0e info=0xd887                                                                                                                                                             han=3 phy_type=14
I (3407) roaming_example: RMM neigbor report bssid=9a:a6:7e:59:bc:5c info=0xd887                                                                                                                                                             
W (3477) wpa: WNM: Send BSS Transition Management Query to 9a:a6:7e:59:b9:c5 
I (3487) wnm_send_bss_transition_mgmt_query: WNM: Send BSS Transition Management                                                                                                                                                             a6:7e:59:b9:c5 query_reason=1
W (3497) wpa: action frame sent
W (3497) wpa: WNM: RX action 7 from 9a:a6:7e:59:b9:c5
W (3507) wpa: WNM: BSS Transition Management Request: dialog_token=1 request_mod                                                                                                                                                             c_timer=0 validity_interval=100
I (3517) ieee802_11_rx_bss_trans_mgmt_req: neighbor_report_elements length 0
W (3527) wpa: WNM: Neighbor report tag 52
W (3527) wpa: WNM:neighbor report bssid:9a:a6:7e:59:b9:c5
W (3537) wpa: WNM: Subelement id=6 len=3
W (3537) wpa: WNM: Subelement id=3 len=1
W (3547) wpa: WNM: Candidate list valid for 10240 ms
W (3547) wpa: WNM: Try to use recent scan results
I (3557) compare_scan_neighbor_results: getting into scan compare neighbors
I (3567) compare_scan_neighbor_results: number of neighbours - 09a:a6:7e:59:b9:c
W (3567) wpa: WNM: Found an acceptable preferred transition candidate BSS 9a:a6:7e:59:b9:c                                                                                                                                                            RSSI -59)
W (3577) wpa: WNM: Transition to BSS 9a:a6:7e:59:b9:c5 based on BSS Transition M                                                                                                                                                             uest after_new_scan=1)
W (3597) wpa: WNM: Sending successful BSS Transition Management Response
W (3597) wpa: WNM: Send BSS Transition Management Response to 9a:a6:7e:59:b9:c5                                                                                                                                                              1 status=0 reason=0 delay=0
W (3607) wpa: WNM: Add candidate list to BSS Transition Management Response frame
W (3617) wpa: action frame sent
W (3627) wpa: WNM: Already associated with the preferred candidate
W (3627) wpa: WNM: RX action 7 from 9a:a6:7e:59:b9:c5
W (3637) wpa: WNM: BSS Transition Management Request: dialog_token=1 request_mod                                                                                                                                                             c_timer=0 validity_interval=100
I (3647) ieee802_11_rx_bss_trans_mgmt_req: neighbor_report_elements length 0
W (3657) wpa: WNM: Neighbor report tag 52
W (3657) wpa: WNM:neighbor report bssid:9a:a6:7e:59:b9:c5
W (3667) wpa: WNM: Subelement id=6 len=3
W (3667) wpa: WNM: Subelement id=3 len=1
W (3677) wpa: WNM: Candidate list valid for 10240 ms
W (3677) wpa: WNM: Try to use recent scan results
I (3687) compare_scan_neighbor_results: getting into scan compare neighbors
I (3697) compare_scan_neighbor_results: number of neighbours - 09a:a6:7e:59:b9:c
W (3707) wpa: WNM: Found an acceptable preferred transition candidate BSS 9a:a6:                                                                                                                                                             RSSI -59)
[... cut ...]
I (119838) roaming_example: RRM supported
I (119848) roaming_example: BTM supported
I (119878) wifi:<ba-add>idx:0 (ifx:0, 60:a4:b7:b0:a7:12), tid:0, ssn:2, winSize:64
I (119908) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (121338) esp_netif_handlers: sta ip: 192.168.173.104, mask: 255.255.255.0, gw: 192.168.173.1
I (136798) roaming_example: esp_bss_rssi_low_handler:bss rssi is=-52
I (136798) roaming_example: setting rssi threshold as -50`

Here I can't see where  STA explicitly scanning for AP's.

Saicharan67, 2023-11-20:
@KlausPopp Thanks for the work around.

Currently I am evaluating the roaming_example in IDF. What i found out while debugging is that...the responce STA get from AP (ORBI )after sending BTM transition request contains information only about connected AP which causing it to not roam by saying Already associated with the preferred candidate.

I (1727) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (1737) roaming_example: setting rssi threshold as -50
I (1737) roaming_example: RRM supported
I (1747) roaming_example: BTM supported
I (3357) roaming_example: esp_bss_rssi_low_handler:bss rssi is=-59
W (3357) wpa: action frame sent
W (3357) wpa: RRM: Notifying neighbor report (token = 1)
I (3357) roaming_example: rrm: neighbor report len=47
I (3367) roaming_example: 0x3fc9a282   01 34 15 9a a6 7e 60 10  0e 87 d8 00 00 0                                                                                                                                                             
I (3377) roaming_example: 0x3fc9a292   06 03 01 05 00 03 01 00  34 15 9a a6 7e 5                                                                                                                                                             
I (3387) roaming_example: 0x3fc9a2a2   87 d8 00 00 00 03 0e 06  03 01 05 00 03 0                                                                                                                                                             
I (3397) roaming_example: RMM neigbor report bssid=9a:a6:7e:60:10:0e info=0xd887                                                                                                                                                             han=3 phy_type=14
I (3407) roaming_example: RMM neigbor report bssid=9a:a6:7e:59:bc:5c info=0xd887                                                                                                                                                             
W (3477) wpa: WNM: Send BSS Transition Management Query to 9a:a6:7e:59:b9:c5 
I (3487) wnm_send_bss_transition_mgmt_query: WNM: Send BSS Transition Management                                                                                                                                                             a6:7e:59:b9:c5 query_reason=1
W (3497) wpa: action frame sent
W (3497) wpa: WNM: RX action 7 from 9a:a6:7e:59:b9:c5
W (3507) wpa: WNM: BSS Transition Management Request: dialog_token=1 request_mod                                                                                                                                                             c_timer=0 validity_interval=100
I (3517) ieee802_11_rx_bss_trans_mgmt_req: neighbor_report_elements length 0
W (3527) wpa: WNM: Neighbor report tag 52
W (3527) wpa: WNM:neighbor report bssid:9a:a6:7e:59:b9:c5
W (3537) wpa: WNM: Subelement id=6 len=3
W (3537) wpa: WNM: Subelement id=3 len=1
W (3547) wpa: WNM: Candidate list valid for 10240 ms
W (3547) wpa: WNM: Try to use recent scan results
I (3557) compare_scan_neighbor_results: getting into scan compare neighbors
I (3567) compare_scan_neighbor_results: number of neighbours - 09a:a6:7e:59:b9:c
W (3567) wpa: WNM: Found an acceptable preferred transition candidate BSS 9a:a6:7e:59:b9:c                                                                                                                                                            RSSI -59)
W (3577) wpa: WNM: Transition to BSS 9a:a6:7e:59:b9:c5 based on BSS Transition M                                                                                                                                                             uest after_new_scan=1)
W (3597) wpa: WNM: Sending successful BSS Transition Management Response
W (3597) wpa: WNM: Send BSS Transition Management Response to 9a:a6:7e:59:b9:c5                                                                                                                                                              1 status=0 reason=0 delay=0
W (3607) wpa: WNM: Add candidate list to BSS Transition Management Response frame
W (3617) wpa: action frame sent
W (3627) wpa: WNM: Already associated with the preferred candidate
W (3627) wpa: WNM: RX action 7 from 9a:a6:7e:59:b9:c5
W (3637) wpa: WNM: BSS Transition Management Request: dialog_token=1 request_mod                                                                                                                                                             c_timer=0 validity_interval=100
I (3647) ieee802_11_rx_bss_trans_mgmt_req: neighbor_report_elements length 0
W (3657) wpa: WNM: Neighbor report tag 52
W (3657) wpa: WNM:neighbor report bssid:9a:a6:7e:59:b9:c5
W (3667) wpa: WNM: Subelement id=6 len=3
W (3667) wpa: WNM: Subelement id=3 len=1
W (3677) wpa: WNM: Candidate list valid for 10240 ms
W (3677) wpa: WNM: Try to use recent scan results
I (3687) compare_scan_neighbor_results: getting into scan compare neighbors
I (3697) compare_scan_neighbor_results: number of neighbours - 09a:a6:7e:59:b9:c
W (3707) wpa: WNM: Found an acceptable preferred transition candidate BSS 9a:a6:                                                                                                                                                             RSSI -59)
[... cut ...]
I (119838) roaming_example: RRM supported
I (119848) roaming_example: BTM supported
I (119878) wifi:<ba-add>idx:0 (ifx:0, 60:a4:b7:b0:a7:12), tid:0, ssn:2, winSize:64
I (119908) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (121338) esp_netif_handlers: sta ip: 192.168.173.104, mask: 255.255.255.0, gw: 192.168.173.1
I (136798) roaming_example: esp_bss_rssi_low_handler:bss rssi is=-52
I (136798) roaming_example: setting rssi threshold as -50

Here I can't see where  STA explicitly scanning for AP's.

xoores, 2023-11-20:
@Saicharan67 I would say that it might be better to post this log in the attachment and not as a huge wallpaper of badly formatted text :shrug:

xanthome, 2024-02-08:
I have a setup with google wifi and the roaming sample will not roam for me ever. it will **only** connect to the main nest router first. No matter how bad the RSSI. I have the threshold set to -50. I see it parsing one single rrm neighbor report. which contains the same bssid that I'm already connected to. Then nothing.

[noroam.log](https://github.com/espressif/esp-idf/files/14205113/noroam.log)

Using framework 5.1. Not sure if google problem or esp problem. I added the scan first to debug, but the same problem is present when not scanning like vanilla example.

jgujarathi, 2024-06-25:
Hi just summarizing before closing. 
1) Most complaints here revolve around the AP not responding to BTM Queries , either by not responding at all or responding with  a candidate list that contains the current AP only. We believe that there is nothing ESP devices can do in this case and recommend checking the setup of the network. 
2) Starting from v5.3 we have an experimental roaming application code that is designed to initiate a station initiated roaming using configurable methods and triggers. (You can try it out at examples/roaming/roaming_app)We will continue to improve and optimise this application and recommend that people facing issues with roaming here, check it out. Please note that this feature is still in experimental  stage.

closed 2024-06-25 by Indastri