espressif/esp-idf #11937: classic bt PCM data is not pcm format

zhou88mang, 2023-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

i have used hfp_ag demo , it works well ,  i set the BR/EDR Sync(SCO/eSCO) default data path  to  PCM ,  By measuring the waveform of the signal,  in fact the format of data output is not pcm format ,  It has always been i2s format .
can you provide some api to set the data format . Because the audio interface provided by most wifi Bluetooth chips is PCM format

esp-cqq, 2023-07-28:
Hello, according to your description, there are some points:
First, make sure that you have correctly configured the PCM data path according to the README document in the hfp_ag demo.

Then, regarding the output audio data format, PCM data format is output through the I2S interface. The waveform I test based on the esp32 is shown in the figure below. 
[image]

I'm not clear about the output waveform of your data and what your expected waveform looks like. If the issue is still not resolved, could you please describe it in more detail?

Finally, regarding the data path configuration, you can follow the instructions in the README document. In ESP-IDF, there might not be a direct API to output PCM format data. Instead, it is usually done through the I2S interface，and the PCM data format remains unchanged; it is simply transmitted according to the timing requirements of the I2S interface.

If you have any further questions or need more assistance, feel free to ask!

zhou88mang, 2023-08-03:
> Hello, according to your description, there are some points: First, make sure that you have correctly configured the PCM data path according to the README document in the hfp_ag demo.
>
> Then, [...]

Sorry it took so long to reply, the output waveform of data what i expected is like this 
[image]
and if i want to get the pcm format data , please tell me how to realize it through the I2S interface.

esp-cqq, 2023-08-07:
Hi,
The waveform you provided seems to be a clock waveform rather than a data waveform.
In the hfp-ag example, the default setting is to output PCM waveform through the I2S interface.

esp-cqq, 2023-11-08:
Hi, I haven’t received your reply for a long time. If there are no questions, I will close this issue after a week.

zhou88mang, 2023-11-15:
> Hi, The waveform you provided seems to be a clock waveform rather than a data waveform. In the hfp-ag example, the default setting is to output PCM waveform through the I2S interface.

I'm sorry for taking a long time to reply, i think the output waveform is like this
[image]
i know the  waveform is output through the I2S interface,but i want the i2s mode is pcm

esp-cqq, 2023-11-15:
Hi, according to the picture you sent, it seems to be a pulse synchronous clock form, and a mono output. 
The default PCM form in esp-idf is stereo, and you can change the rising edge or the falling edge to send data at the menuconfig , but there is currently no open interface to change other forms of PCM. 
If you only need to output the waveform of the picture, I can provide you with the changed lib file. But you need to clarify the following requirements: What are the synchronization clock frequency and data clock frequency respectively of PCM?  left or right channel output? pcm as central or peripheral? 16bit or 8bit data samples?

zhou88mang, 2023-11-15:
> Hi, according to the picture you sent, it seems to be a pulse synchronous clock form, and a mono output. The default PCM form in esp-idf is stereo, and you can change the rising edge or the falling [...]

Hi，thank you for your reply, the requirements are as follows:
synchronization clock frequency   8000HZ
data clock frequency    256*8000HZ
mono output
left channel output
pcm as central 
16 bit wide
by the way, i also  need esp32  to receive PCM data from the slave ,and then send to bluetooth headset

esp-cqq, 2023-11-15:
Hi, I want to confirm that the data clock frequency does not seem to be 256*8000HZ, is it OK to use the common 256khz?

zhou88mang, 2023-11-15:
> Hi, I want to confirm that the data clock frequency does not seem to be 256*8000HZ, is it OK to use the common 256khz?

yes，i think 2*16 * 8000 is enough , so data clock frequency is 256000HZ

I forgot to mention that PCM has two modes, mode A and mode B, my requirement is mode B

esp-cqq, 2023-11-16:
Hi, All the previous requirements can be achieved, but can you elaborate on the mode A and mode B you mentioned? Maybe the name is different. There doesn't seem to be a similar definition in esp32, so I'm not quite clear what exactly mode B is?

zhou88mang, 2023-11-16:
> Hi, All the previous requirements can be achieved, but can you elaborate on the mode A and mode B you mentioned? Maybe the name is different. There doesn't seem to be a similar definition in esp32, so I'm not quite clear what exactly mode B is?

Hi, as far as I am concerned，
mode A: After FSYNC is valid, the data is valid on the second rising edge (falling edge) of BCLK (one bit clock delay)
[image]

mode B: After FSYNC is valid, the data is valid on the first rising edge (falling edge) of BCLK (no delay)
[image]

And in i2s_std.h which in esp-idf , macro definition “I2S_STD_PCM_SLOT_DEFAULT_CONFIG”  may be able to control it
```/**
 * @brief PCM(short) format in 2 slots
 * @note  PCM(long) is same as philips in 2 slots
 * @param bits_per_sample i2s data bit width
 * @param mono_or_stereo I2S_SLOT_MODE_MONO or I2S_SLOT_MODE_STEREO
 */
#define I2S_STD_PCM_SLOT_DEFAULT_CONFIG(bits_per_sample, mono_or_stereo)  { \
    .data_bit_width = bits_per_sample, \
    .slot_bit_width = I2S_SLOT_BIT_WIDTH_AUTO, \
    .slot_mode = mono_or_stereo, \
    .slot_mask = (mono_or_stereo == I2S_SLOT_MODE_MONO) ? \
                I2S_STD_SLOT_LEFT : I2S_STD_SLOT_BOTH, \
    .ws_width = 1, \
    .ws_pol = true, \
    .bit_shift = true, \
    .msb_right = (bits_per_sample <= I2S_DATA_BIT_WIDTH_16BIT) ? \
                true : false, \
}```
bit_shift = false  may be  Mode B ?

esp-cqq, 2023-11-16:
Ok, I get it. I will give you the lib file according to your requirements later.

esp-cqq, 2023-11-17:
Hi, According to your requirements, this is the corresponding lib file based on idf version v5.0.4. 
[ sync 8kHZ; clk 256kHZ; mono output / left channel; pcm as central; 16 bit wide; enclosing the first falling (mode B according to your description) ] 
The rest of the requirements can be configured in esp-idf. (including rising edge or falling edge)

[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/13388901/libbtdm_app.zip)
file path: components/bt/controller/lib_esp32/esp32

zhou88mang, 2023-11-17:
ok，thanks! I will have a test.

zhou88mang, 2023-11-20:
> Hi, According to your requirements, this is the corresponding lib file based on idf version v5.0.4. [ sync 8kHZ; clk 256kHZ; mono output / left channel; pcm as central; 16 bit wide; enclosing the [...]

Hi ， i have  tested the static library you provided. I used the HFP_AG demo.
Unfortunately, it reported sone errors during the compilation phase.
And log is as follow:
```[21/1177] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[634/1177] Performing configure step for 'bootloader'
-- Found Git: C:/Users/PC/Desktop/work/dbbz/esp32/esp-tools/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/PC/Desktop/work/dbbz/esp32/esp-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/PC/Desktop/work/dbbz/esp32/esp-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/PC/Desktop/work/dbbz/esp32/esp-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/PC/Desktop/work/dbbz/esp32/esp-examples/hfp_ag/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- App "bootloader" version: v5.0.4-319-ga6afbb38a4-dirty
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader_support C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/efuse C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_app_format C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_common C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_hw_support C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_rom C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_system C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esptool_py C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/freertos C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/hal C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/log C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader/subproject/main C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/newlib C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/partition_table C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/soc C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/spi_flash C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/PC/Desktop/work/dbbz/esp32/esp-examples/hfp_ag/build/bootloader
[693/1177] Performing build step for 'bootloader'
[1/102] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_console_loader.c.obj
[2/102] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/bootloader_sha.c.obj
[3/102] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32/esp_efuse_fields.c.obj
[... cut ...]
C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bt/controller/esp32/bt.c:1556: undefined reference to `sdk_config_set_uart_flow_ctrl_enable'
c:/users/pc/desktop/work/dbbz/esp32/esp-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(bt.c.obj): in function `esp_bt_controller_enable':
C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/bt/controller/esp32/bt.c:1505: undefined reference to `sdk_config_set_bt_pll_track_enable'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.```

esp-cqq, 2023-11-20:
The idf version does not match the lib version. Are you sure that your idf version is v5.0.4?

zhou88mang, 2023-11-20:
> The idf version does not match the lib version. Are you sure that your idf version is v5.0.4?

yes ， i am using esp-idf v5.0.4 
This is using the previous libbtdm_app. a  output of log.
```entry 0x40080650
I (27) boot: ESP-IDF v5.0.4-319-ga6afbb38a4-dirty 2nd stage bootloader
I (27) boot: compile time 10:30:57
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v3.1
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1cfa0h (118688) map
I (144) esp_image: segment 1: paddr=0002cfc8 vaddr=3ffbdb60 size=03050h ( 12368) load
I (149) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7a2b8h (500408) map
I (331) esp_image: segment 3: paddr=000aa2e0 vaddr=3ffc0bb0 size=01308h (  4872) load
I (333) esp_image: segment 4: paddr=000ab5f0 vaddr=40080000 size=19394h (103316) load
I (392) boot: Loaded app from partition at offset 0x10000
I (392) boot: Disabling RNG early entropy source...
I (403) cpu_start: Multicore app
I (403) cpu_start: Pro cpu up.
I (404) cpu_start: Starting app cpu, entry point is 0x400813cc
0x400813cc: call_start_cpu1 at C:/Users/PC/Desktop/work/dbbz/esp32/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (421) cpu_start: Pro cpu start user code
I (422) cpu_start: cpu freq: 160000000 Hz
I (422) cpu_start: Application information:
I (426) cpu_start: Project name:     hfp_ag
I (431) cpu_start: App version:      1
I (436) cpu_start: Compile time:     Nov 20 2023 10:30:22
I (442) cpu_start: ELF file SHA256:  e20d1a3b402e19ec...
I (448) cpu_start: ESP-IDF:          v5.0.4-319-ga6afbb38a4-dirty
I (454) cpu_start: Min chip rev:     v0.0
I (459) cpu_start: Max chip rev:     v3.99 
I (464) cpu_start: Chip rev:         v3.1
I (469) heap_init: Initializing. RAM available for dynamic allocation:
I (476) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (482) heap_init: At 3FFB6BF8 len 00001408 (5 KiB): DRAM
I (488) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (494) heap_init: At 3FFC6750 len 000198B0 (102 KiB): DRAM
I (500) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (507) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (513) heap_init: At 40099394 len 00006C6C (27 KiB): IRAM
I (521) spi_flash: detected chip: generic
I (524) spi_flash: flash io: dio
W (528) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (543) app_start: Starting scheduler on CPU0
I (546) app_start: Starting scheduler on CPU1
I (546) main_task: Started on CPU0
I (556) main_task: Calling app_main()
I (576) BTDM_INIT: BT controller compile version [ec4ac65]
I (576) system_api: Base MAC address is not set
I (576) system_api: read default base MAC address from EFUSE
I (586) BTDM_INIT: Bluetooth MAC: 08:3a:8d:0b:17:de
I (586) phy_init: phy_version 4771,450c73b,Aug 16 2023,11:03:10
I (1216) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1226) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1226) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1236) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1246) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1256) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
n (1266) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

  ==================================================
 |       Steps to test hfp_ag                     |
 |                                                |
 |  1. Print 'help' to gain overview of commands  |
 |  2. Setup a service level connection           |
 |  3. Run hfp_ag to test                         |
 |                                                |
 =================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
hfp_ag> I (1826) main_task: Returned from app_main()```

esp-cqq, 2023-11-20:
Please run the following two commands and then replace the lib file.
`git pull `
 `git checkout 8fbf4ba605`

zhou88mang, 2023-11-20:
> Please run the following two commands and then replace the lib file. `git pull ` `git checkout 8fbf4ba605`

I have run the two commands you provided, and then replaced the lib file,  i successfully got the pcm data waveform what I wanted， thanks  very much ！！

closed 2023-11-20 by zhou88mang

Ageofenigma, 2024-04-11:
Is it possible to build this lib for ESP-IDF v5.2?   ESP32 in my project is a slave PCM data provided by the Simcom modem