espressif/esp-idf #11938: esp32s3box同时初始化i2c和spi，后初始化的会失败，单独初始化某一个则通过

YyeHG, 2023-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3box

### Power Supply used.

USB

### What is the expected behavior?

两者成功配置通过

### What is the actual behavior?

先初始化spi然后初始化i2c，配置失败
E (21162) i2c: i2c_master_cmd_begin(1481): i2c driver not installed

### Steps to reproduce.

   这是spi的配置
 esp_err_t ret;
    spi_bus_config_t buscfg = {
        .miso_io_num = SPI_PIN_MISO,
        .mosi_io_num = SPI_PIN_MOSI,
        .sclk_io_num = SPI_PIN_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 16 * 320 * 2 + 8};
    spi_device_interface_config_t devcfg = {
#ifdef CONFIG_LCD_OVERCLOCK
        .clock_speed_hz = 26 * 1000 * 1000, // Clock out at 26 MHz
#else
        .clock_speed_hz = 10 * 1000 * 1000, // Clock out at 10 MHz
#endif
        .mode = 0,                               // SPI mode 0
        .spics_io_num = SPI_PIN_CS,              // CS pin
        .queue_size = 7,                         // We want to be able to queue 7 transactions at a time
        .pre_cb = lcd_spi_pre_transfer_callback, // Specify pre-transfer callback to handle D/C line
    };
    // Initialize the SPI bus
    ret = spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO);
    ESP_LOGI(TAG, "spi_bus_initialize%x", ret);
    // Attach the LCD to the SPI bus
    ret = spi_bus_add_device(SPI2_HOST, &devcfg, &handle);
    ESP_LOGI(TAG, "spi_bus_add_device%x", ret);

这是i2c的配置
  int ret;
    static i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = GPIO_NUM_41; // SDA IO映射
    conf.sda_pullup_en = GPIO_PULLUP_DISABLE;
    conf.scl_io_num = GPIO_NUM_40; // SCL IO映射
    conf.scl_pullup_en = GPIO_PULLUP_DISABLE;
    conf.master.clk_speed = 400000; // I2C CLK 频率
    conf.clk_flags = 0;
    ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &conf));       // 配置I2C
    ret = i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0); // I2C功能安装使能函数
    ESP_LOGI(TAG, "i2c_driver_install%d", ret);

### Debug Logs.

```plain
I (20952) iic_init: i2c_driver_install-1
E (21162) i2c: i2c_master_cmd_begin(1481): i2c driver not installed
```

espressif2022, 2023-07-24:
测试没问题，是不是 引脚配置有冲突。我用 demo 示例做了以下测试，两个都是 ok 的。

`bsp_display_start(); `（初始化 SPI， 如果用这个 API，注意屏蔽 iic 输入源 indev）
`bsp_i2c_init(); `（初始化两个 IIC）

espressif2022, 2023-07-25:
首先我确认了下，还是 ok 的，初始化顺序，IIC->SPI ->UART
[image]

1： RST，CTS 引脚也看下
2：你这边除了这三个初始化，前面是否还有其他代码，例如正常先操作 box 的初始化

YyeHG, 2023-07-25:
> 
RST，CTS 引脚不使用，用了宏定义忘记说明了不好意思
#define RTS			UART_PIN_NO_CHANGE
#define CTS			UART_PIN_NO_CHANGE
我是在factory_demo中进行编写的，分别进行了串口，spi和i2c的初始化，st7735_show();oled_init();分别是使用spi和i2c协议进行调试oled屏幕
[image]

> 首先我确认了下，还是 ok 的，初始化顺序，IIC->SPI ->UART [image]
> 
> 1： RST，CTS 引脚也看下 2：你这边除了这三个初始化，前面是否还有其他代码，例如正常先操作 box 的初始化

espressif2022, 2023-07-25:
脚肯定和现有代码冲突，app_pwm_led_init 初始化，就会和你的IIC GPIO 40/41 冲突
其他的也都检查下吧，这应该是是工程使用问题，和 IDF 没关系

YyeHG, 2023-07-25:
我认为不是引脚冲突导致的，因为一开始你没告诉我iic与led的初始化冲突的时候，iIC，spi，uart任意两个配置都是没有问题的，是配置三个才会使最后一个初始化的失败。

然后在你的提醒下，我把app_pwm_led_init的引脚更换成了43/44/14，更改引脚配置后，factory_demo工程没有使用其他的可外接引脚，而我配置三个外设也没有引脚冲突，依旧是最后初始化的iic失败了

[image]

espressif2022, 2023-07-25:
方便的话，可以把你工程放在 github 的哪个仓库，我来下载下试试；

YyeHG, 2023-07-25:
> 方便的话,可以把你工程放在开源代码库的哪个仓库,我来下载下试试；

https://github.com/YyeHG/factory_demo.git
初始化的文件在factory_demo/main/app路径下

espressif2022, 2023-07-26:
> > 方便的话,可以把你工程放在开源代码库的哪个仓库,我来下载下试试；
> 
> https://github.com/YyeHG/factory_demo.git 初始化的文件在factory_demo/main/app路径下

首先，我确认了下，确实如此，最后一个分配的任务会失败。
@YyeHG ，如果你这想快速用起来，可以先把上述 factory_demo 的初始化屏蔽了，这样不影响你正常使用开发。

`uart: uart_driver_install(1590): Could not allocate an interrupt for UART`
我这边看了下，最后一个任务都类似错误，iic 或者 SPI 中断分配失败，IDF 的兄弟能帮忙解答下？

larryli, 2023-08-08:
> 首先，我确认了下，确实如此，最后一个分配的任务会失败。 @YyeHG ，如果你这想快速用起来，可以先把上述 factory_demo 的初始化屏蔽了，这样不影响你正常使用开发。
>
> `uart: uart_driver_install(1590): Could not allocate an interrupt for UART` 我这边看了下，最后一个任务都类似错误，iic 或者 SPI [...]

一般来说就是中断资源被用完了，导致最后一个项目初始化失败（其实是后续项目都会失败）。

建议减少不必要的中断资源使用，比如不要使用 GPIO 中断，换成定时器。

espressif2022, 2023-09-21:
> 首先，我确认了下，确实如此，最后一个分配的任务会失败。 @YyeHG ，如果你这想快速用起来，可以先把上述 factory_demo 的初始化屏蔽了，这样不影响你正常使用开发。
>
> `uart: uart_driver_install(1590): Could not allocate an interrupt for UART` 我这边看了下，最后一个任务都类似错误，iic 或者 SPI [...]

我这没有问题了，@Alvin1Zhang

wishinlife, 2023-11-29:
我也遇到类似的问题，像是IDF限制了初始化硬件设备的数量。[Only two SPI, usb, and sdmmc interfaces can be initialized, and the third one fails. (IDFGH-11567)](https://github.com/espressif/esp-idf/issues/12688)

1、初始化了2个SPI后（或1个SDMMC和1个SPI），初始化2个UART，第二个UART失败，只是初始化1个SPI后（或一个SDMMC），2个UART可以成功：
E (1328) uart: uart_driver_install(1614): Could not allocate an interrupt for UART
E (1338) GPS: GPS uart port driver install error: 261(ESP_ERR_NOT_FOUND).

2、先初始化SDMMC（1个），后初始化SPI（2个），再初始化UART（2个），前3个可以成功（SDMMC+2个SPI），两个UART都失败：
E (1454) uart: uart_driver_install(1614): Could not allocate an interrupt for UART
E (1464) [RS485]: Uart port driver install error: 261(ESP_ERR_NOT_FOUND).
I (1469) uart: ALREADY NULL
I (1474) [GPS]: Start gps on UART 2.
I (1479) uart: queue free spaces: 10
E (1484) uart: uart_driver_install(1614): Could not allocate an interrupt for UART
E (1489) [GPS]: GPS uart port driver install error: 261(ESP_ERR_NOT_FOUND).

3、先初始化SDMMC（1个），后初始化UART（2个），最后初始化SPI（2个），前3个可以成功（SDMMC+2个UART），后面SPI失败：
E (1329) w5500.mac: esp_eth_mac_new_w5500(768): adding device to SPI host #2 failed
E (1339) esp_eth: esp_eth_driver_install(189): can't set eth->mac or eth->phy to null
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200ffe0
0x4200ffe0: W5500_init at /home/user/esp-projects/IoT-Station-ESP32-S3-New/build/../main/src/Sys/w5500.c:159 (discriminator 1)

file: "../main/src/Sys/w5500.c" line 159
func: W5500_init
expression: esp_eth_driver_install(&eth_config_spi, &eth_handle_spi[i])

abort() was called at PC 0x40383b1f on core 0
0x40383b1f: _esp_error_check_failed at /home/user/esp/esp-idf/components/esp_system/esp_err.c:50

4、先初始化SDMMC（1个），后初始化SPI（2个），再初始化USB，最后初始化UART（2个），最前3个可以成功（SDMMC+2个SPI），后面USB和UART失败：
I (1519) uart: queue free spaces: 10
E (1524) uart: uart_driver_install(1614): Could not allocate an interrupt for UART
E (1534) [RS485]: Uart port driver install error: 261(ESP_ERR_NOT_FOUND).
I (1539) uart: ALREADY NULL
I (1544) [GPS]: Start gps on UART 2.
I (1549) uart: queue free spaces: 10
E (1554) uart: uart_driver_install(1614): Could not allocate an interrupt for UART
E (1559) [GPS]: GPS uart port driver install error: 261(ESP_ERR_NOT_FOUND).
I (1564) uart: ALREADY NULL
I (1569) [Ftp]: ftp_task start
I (1574) [Ftp]: ftp_user:[hdsolar] ftp_pass:[password]
I (1719) modem_board: Waiting for modem initialize ready
I (6719) USB_HCDC: iot_usbh_cdc, version: 0.2.1
I (6719) IOT_USBH: iot_usbh, version: 0.2.1
E (6719) IOT_USBH: ../managed_components/espressif__iot_usbh/iot_usbh.c:352 (iot_usbh_port_init):HCD Install failed
E (6729) USB_HCDC: ../managed_components/espressif__iot_usbh_cdc/iot_usbh_cdc.c:554 (usbh_cdc_driver_install):usb port init failed
E (6739) esp-modem-dte: esp_modem_dte_new(466): usb driver install failed

以上测试在初始化第一个设备前还初始化了一个I2C的RTC，如果我把RTC初始化注释了，重新运行第4个程序，则第一个UART可以正常初始化，从以上情况基本可以看出前4个设备驱动可以正常初始化，之后的就会失败，我需要用到1个I2C、2个SPI、3个UART（其中一个作为刷程序和日志输出）、1个SDMMC、1个USB，共8个设备，如果刷程序和日志输出的也算成功的话，那也就是idf只可以初始化5个设备。我还使用到了蓝牙，但蓝牙不管是先初始化还是后初始化都没有影响

mythbuster5, 2024-11-06:
Since is one year not updated, I will close this issue. Or  you can report this in esp-iot-sulotion.

closed 2024-11-06 by mythbuster5