espressif/esp-idf #11939: problem with deep sleep time (esp_sleep_enable_timer_wakeup() suddenly/randomly stops)

snehapawar221087, 2023-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello All,

we have designed a data logger using esp32s3, and device wakes up after every set time, and collects data from sensor and goes to deep sleep.
But suddenly system is facing a issue where device is not waking from deep sleep after setting a time of 10mins.

ESP_IDF version is v4.4
please find sdkconfig file in an attachment.

Kindly note the below behaviour:
1] This issue is happening randomly.
2] device is waking up due to any external interrupts which we have set but timer wakeup
3] when device goes to deep sleep, unless and until external interrupt happens, device won't wakeup
4] when issue happens, device won't wakeup and collects data until external interrupt happens, before this issue and after external interrupt while device is in mode, device works fine, goes in deep sleep and comes out of it properly, and no data loss is been seen.

it's been found that when device won't wake up at set time, till external interrupt, that time device remains in deep sleep.

Please find the code below :

<-------------------------------------------------
[sdkconfig_file.txt](https://github.com/espressif/esp-idf/files/12144279/sdkconfig_file.txt)
[sdkconfig_file.txt](https://github.com/espressif/esp-idf/files/12144286/sdkconfig_file.txt)
------------------------------------------------------------->

//set deep sleep time as sleep_time -> seconds is user defined
uint64_t sleep_time = seconds * uS_TO_S_FACTOR;
esp_err_t err = esp_sleep_enable_timer_wakeup(sleep_time);
printf("SLEEPING FOR %d SECONDS: %d\r\n\r\n", seconds, err);

//external interrupts to wakeup device from deep sleep
const int ext_wakeup_pin_1 = 12;
const uint64_t ext_wakeup_pin_1_mask = 1ULL << ext_wakeup_pin_1;
const int ext_wakeup_pin_2 = 14;
const uint64_t ext_wakeup_pin_2_mask = 1ULL << ext_wakeup_pin_2;
const int ext_wakeup_pin_3 = 1;
const uint64_t ext_wakeup_pin_3_mask = 1ULL << ext_wakeup_pin_3;

esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask | ext_wakeup_pin_2_mask | ext_wakeup_pin_3_mask, ESP_EXT1_WAKEUP_ALL_LOW);

gettimeofday(&sleep_enter_time, NULL);

//delete/deinit watchdog timers
esp_task_wdt_reset();
esp_task_wdt_delete(NULL); //Unsubscribe Idle Task from TWDT
esp_task_wdt_status(NULL); //Confirm Idle task has unsubscribed
//Deinit TWDT after all tasks have unsubscribed
esp_task_wdt_deinit();
esp_task_wdt_status(NULL); //Confirm TWDT has been deinitialized

//start deep sleep
esp_deep_sleep_start();

<-------------------------------------------------------------------------------------------------------------->

Please let me know,
1] what is the cause of this issue AND
2] how to resolve this issue.

Thank you in advance.

esp-lis, 2023-07-24:
I guess it may be due to the slow clock source you have chosen. Please refer to the following:
```
# CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC is not set
# CONFIG_ESP32S3_RTC_CLK_SRC_EXT_CRYS is not set
CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC=y
# CONFIG_ESP32S3_RTC_CLK_SRC_INT_8MD256 is not set
```
**CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC** It means you need to provide an external crystal oscillator as the slow clock source. Perhaps you can use Internal 150 kHz OSC (**CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC**) to test.

snehapawar221087, 2023-07-24:
Hello [esp-lis],
Thanks for your reply.

We used external crystal oscillator in our devices, so the setting.

snehapawar221087, 2023-07-25:
Hello [esp-lis],
Thanks for your reply.

Am positive that GPIO functionality of the alternative XTAL PIN functionality is not initialized in the code. 

I will use Internal 150 kHz OSC (CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC) to test whether there is still the same problem.

Thank you.

snehapawar221087, 2023-07-26:
Hello [esp-lis],

I tried Internal 150 kHz OSC (CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC) to test whether there is still the same problem or not 
and I got this data loss issue again with this same setting.

If this is not due to external/internal clock, then is it due to some firmware routine/flow?
what causes this? 

Have already share the flow from timer set for waking up from deep sleep till calling deep_sleep function.

what might be wrong here?

Let me know what I can test/check to resolve this.

Thank you.

esp-lis, 2023-07-26:
> Hello [esp-lis],
>
> I tried Internal 150 kHz OSC (CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC) to test whether there is still the same problem or not and I got this data loss issue again with this same [...]

@snehapawar221087 When Internal 150 kHz OSC (**CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC**) is used as a slow clock source compared to external 32k oscillator (**CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC**), there is a large time drift.

Or use external 32k crystal, e.g. 20 ppm (**CONFIG_ESP32S3_RTC_CLK_SRC_EXT_CRYS**), which we have tested over a long period of time.

snehapawar221087, 2023-07-26:
Hello [esp-lis],
Sorry, but I don't understand what you have said.
I don't see any time drift when I switched from external osc to internal rc clock. 

please explain in brief 

Thank you.

esp-lis, 2023-07-27:
@snehapawar221087 I suspect that the slow clock source may have lost its clock and want to replace a slow clock source test (such as Internal 150 kHz OSC (**CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC**) or external 32k xtal (**CONFIG_ESP32S3_RTC_CLK_SRC_EXT_CRYS**)), which have longer tests

If it (the problem you describe) is still there, it could be something else and we need more information to diagnose it.

For example, this problem can occur in very few products, or it can occur in all products.

snehapawar221087, 2023-07-28:
Hello esp-lis,
 We got this issue in few devices so far.

The device which i have with this issue(other devices are with customers) has frequent data loss issue when clock was external 32k oscillator (CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC).
When i changed clock to Internal 150 kHz OSC (CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC), i saw data loss once, but then onwards for almost 48 hours, data loss is not seen.

Another device which has got this loss issue on field and while with customer's, that device didn't report any data loss issue for more than a 15 days now, when clock was same external 32k oscillator (CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC).

Device serial logs couldn't help as we can see that device went for deep sleep, but never wakeup at set time, but wakeup for set external GPIO wakeup sources.

I don't understand whether it's firmware or hardware issue.
If it's a firmware issue, then we must have seen frequent data losses in multiple devices with that firmware, which is not been seen. But getting loss while EXT as well as INT clock, doubting me on firmware.
If it's hardware issue, then we must get no data loss, when clock is internal RC.(i assumed it that device might work perfectly on internal RC,if it's hardware issue(?????))

What might be the issue here??

Thanks & Regards,
Sneha

snehapawar221087, 2023-07-31:
hello, 
anything that might help me here??? any updated??

esp-lis, 2023-08-01:
> hello, anything that might help me here??? any updated??

we can add some test code to collect some data with the following steps:

1. Use the **external 32k oscillator** as the slow clock source
2. When the data is read after the timer wakes up, call [rtc_time_get()](https://github.com/espressif/esp-idf/blob/28167ea5a3cdd49227216c4131a9df7f4eda260e/components/esp_hw_support/port/esp32s3/rtc_time.c#L170) to record the slow clock tick. For example, output the slow clock tick to log file and record the system timestamp YYYY-MM-DD HH:mm:ss.SSS.
3. Also record the slow clock tick when ext1 wakes up (same as step 2)
4. Compare the system timestamp and expect the slow clock to be continuously timed (tick_count * period_us/tick = system_time_us, which requires the slow clock source to be stable enough).

snehapawar221087, 2023-08-02:
Hello esp-lis,

Please find the log, attached with this reply.

In this log, you can fine, sleep_time_ms as a time for which device was in deep sleep.
kv_save_time as a device's rtc time, read when device wakeup from deep sleep.
Used the external 32k oscillator as the slow clock source

according to log,

set time for      Total time while                 Device time
deep sleep       Device was in deep           when wokeup
(in sec)              sleep(in sec)
   298                      298                               1690256400           Tue Jul 25 2023 09:10:00 GMT+0530
   299                      299                                1690256701          Tue Jul 25 2023 09:15:01 GMT+0530
   298                      5923                              1690262626         Tue Jul 25 2023 10:53:46 GMT+0530

It looks like, device's set rtc time is proper/stable.
But device didn't wakeup on set time using wakeup source as timer.

Please see attached log file, that might help you for analysis.
[loss_issue_log.txt](https://github.com/espressif/esp-idf/files/12239852/loss_issue_log.txt)

[loss_issue_log.txt](https://github.com/espressif/esp-idf/files/12239844/loss_issue_log.txt)

Thanks & Regards,
Sneha

snehapawar221087, 2023-08-02:
set time for               total time while                               Device time when wokeup
deep sleep                 device was in deep sleep
(in seconds)	             (in seconds)
                                  						
298	                          298	                         1690256400	(Tue Jul 25 2023 09:10:00 GMT+0530 (India Standard Time)					
299	                          299	                         1690256701	(Tue Jul 25 2023 09:15:01 GMT+0530 (India Standard Time)					
298	                          5923	                         1690262626	(Tue Jul 25 2023 10:53:46 GMT+0530 (India Standard Time)

snehapawar221087, 2023-08-03:
Hello esp-lis,

According to the logs, device won't loose it's time, when it skips timer wakeup.
But for some reason set timer is not invoked.

wakeup reason   :                                                                Timer
set time to wakeup from deep sleep (in seconds)	: 298 
total time while device was in deep sleep (in seconds)                        :  298 
Device time when wokeup                    :               1690256400 (Tue Jul 25 2023 09:10:00 GMT+0530 (India Standard Time)

wakeup reason                   :                                               Timer
set time to wakeup from deep sleep (in seconds)	:299 
total time while device was in deep sleep (in seconds)               :           299 
Device time when wokeup                        :              1690256701 (Tue Jul 25 20  23 09:15:01 GMT+0530 (India Standard Time)

wakeup reason                                     :                              Button press(external gpio wakeup
set time to wakeup from deep sleep (in seconds)	   :  298 
total time while device was in deep sleep (in seconds)             :             5923 
Device time when wokeup                  :                    1690262626 (Tue Jul 25 2023 10:53:46 GMT+0530 (India Standard Time)

As per the current flow, we set the timer wakeup in deep sleep, and then we set external gpio wakeups, as below

    we have external wakeups on below gpios
    const int ext_wakeup_pin_1 = 12; //light_interrupt
    const uint64_t ext_wakeup_pin_1_mask = 1ULL << ext_wakeup_pin_1;
    const int ext_wakeup_pin_2 = 14;    //button interrupt
    const uint64_t ext_wakeup_pin_2_mask = 1ULL << ext_wakeup_pin_2;
    const int ext_wakeup_pin_3 = 1;    //shock interrupt
    const uint64_t ext_wakeup_pin_3_mask = 1ULL << ext_wakeup_pin_3;
    const int ext_wakeup_pin_4 = 7;    //usb interrupt
	
	
enabling external wakeups using below
esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask | ext_wakeup_pin_2_mask | ext_wakeup_pin_3_mask, ESP_EXT1_WAKEUP_ALL_LOW);
esp_sleep_enable_ext0_wakeup(ext_wakeup_pin_4, 1);

Please note below important observations:
1] We found some of the devices, where loose data because they didn't wakeup for set time, and started again working after sometime to collect data.
2] And some devices, sent data because they had CPU resets, and went in this no wakeup mode, just after that. These devices comes out of this issue, by external GPIO wakeup reason, always with button press(GPIO_14) or USB connection(GPIO_7) for charging.

What might be the issue here?

snehapawar221087, 2023-08-07:
Hello esp-lis,

Below points are noted:
esp32s3 EXTx wake up only support ESP_EXT1_WAKEUP_ANY_HIGH or ESP_EXT1_WAKEUP_ANY_LOW, but not support ESP_EXT1_WAKEUP_ALL_LOW, refer to [sleep_modes.rst#L119-L122]
If you need to wake up with ext0, call esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON); to ensure ESP_PD_DOMAIN_RTC_PERIPH is powered on during sleep.

In addition, I need to know what is the CPU reset reason you mentioned here (And some devices, sent data because they had CPU resets)?
- We found that when device stops waking up for the set time, after a while device wakeups up because of CPU reset.
- we got RESET_REASON from rtc_get_reset_reason(0) as RTC_SW_CPU_RESET, and device starts sending data now onwards.

So two behaviour has been found here:
1] device stops waking up for set timer, and after a while wakes up and starts collecting without any cpu/rtc/poweron resets
2] device stops waking up for set timer, and after a while wakes up due to CPU reset.

What is causing these kind of behaviours?

esp-lis, 2023-08-08:
> Hello esp-lis,
>
> Below points are noted: esp32s3 EXTx wake up only support ESP_EXT1_WAKEUP_ANY_HIGH or ESP_EXT1_WAKEUP_ANY_LOW, but not support ESP_EXT1_WAKEUP_ALL_LOW, refer to [...]

**RTC_SW_CPU_RESET**, It looks like a software-triggered reset, for example, by calling [esp_cpu_reset(int cpu_id)](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/esp_hw_support/cpu.c#L101).

esp-lis, 2023-08-08:
> **RTC_SW_CPU_RESET**, It looks like a software-triggered reset, for example, by calling [esp_cpu_reset(int [...]

It might have called [esp_restart()](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/esp_system/esp_system.c#L48) (**ESP_SYSTEM_MEMPROT_FEATURE** disabled in menuconfig) and triggered a **RTC_SW_CPU_RESET**.

snehapawar221087, 2023-08-08:
Hello esp-lis,

sorry, but may i know what is "ESP_SYSTEM_MEMPROT_FEATURE"?

have used watchdog in code flow, and before restarting due to it or due to fota, we set some flag in nvs, so we get status that this kind of cpu reset is because of watchdog or fota etc.
But we are not getting cpu resets due to fota or watchdog resets, when we found data loss.

While this issue,
1] device stops waking up for set timer, and all of sudden, it wakes up, starts collecting data
OR
2] device stops waking up for set timer, and after a while esp restarts due to cpu reset(am sure it;s not because of watchdog or fota, but unknown to me because those devices are with customers)

i don't understand case 1 and case 2 are happening due to the same reason? 
Is this a hardware issue?? it's not with multiple devices, and it's rarely happens with same devices which has this dataloss issue.

Why esp timer stops working and device won't come out of deep sleep?

esp-lis, 2023-08-09:
> Hello esp-lis,
>
> sorry, but may i know what is "ESP_SYSTEM_MEMPROT_FEATURE"?
>
> have used watchdog in code flow, and before restarting due to it or due to fota, we set some flag in nvs, so we get [...]

@snehapawar221087 Can you provide a simple example code that can reproduce the problem?

snehapawar221087, 2023-08-09:
Hello esp-lis,

am getting this issue in my project, which is huge code. 
Instead, can you tell me what do you require to debug, so I will achieve that with my code, and give to you for analysis?

Thanks

peter3099, 2023-08-31:
Hi guys

I'm getting a similar error, for some reason during deep sleep the RTC generates some issue that alters the time counting

[image]

Now is the time since boot and duration is the time while in deep sleep. It should wake up around every 5 minutes +5% random delay 

only the button being pressed fixes this. I've tried using 5.0.3 and 5.1.1 and I get the same results. It happens randomly and I didn't have this issue with idf v4

Thanks

peter3099, 2023-08-31:
This is my code after wake up 
`void check_wakeup_reason()
{
    ESP_LOGI(TAG, "Check Wake UP reason"); // BTN2 to ask for a new sample
    esp_sleep_wakeup_cause_t wakeup_reason;
    wakeup_reason = esp_sleep_get_wakeup_cause();

    struct timeval timeNow;
    struct timeval timeDiff;
    gettimeofday(&timeNow, NULL);
    timersub(&timeNow, &sleepTime, &timeDiff);
    uptimeInMs += (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000);
    printf("WK reason %d Now: %" PRIu64 "ms, Duration: %" PRIu64 "ms\r\n",
           wakeup_reason, (timeNow.tv_sec * (uint64_t)1000) + (timeNow.tv_usec / 1000), (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000));`

and the deep sleep task `void deep_sleep_swt_task(void *parameter)
{
    uint64_t msToSleep = (uint64_t *)parameter;
    // ESP_LOGW(TAG, "Sleeping swtState: %d for %" PRIu64 "sec NoMotionFactor: %d",
    //          swtState, msToSleep / 1000, get_no_motion_factor());
    if (swtState == SWT_WAIT_JOIN)
    { // deepsleep  to wait for next lora Join request
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        print_lora_vars();
    }
    else if (swtState == SWT_FAIL_JOIN)
    { // deepsleep  to wait for next lora Join request
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        print_lora_vars();
        ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(BTN2, 0));
    }
    else if (swtState == SWT_WAIT_CO2)
    { // deep sleep to wait for C02 sensor measurement
        gpio_deep_sleep_hold_en();
        gpio_hold_en(POWER_VCC);
        gpio_hold_en(CO2_EN);
        // gpio_num_t gpio_num_co2int = CO2_INT;
        ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(CO2_INT, 0)); // wake up when CO2 int is low
    }
    else if (swtState == SWT_SLEEP)
    { // deepsleep  to wait for next sample
        // vTaskDelay()
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        init_ulp_program();
        ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
        if (get_mot_en() == false)
        {
            ESP_LOGE(TAG, "MOT ROUTINE DISABLED");
            // ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(BTN2, 0));
        }
        print_lora_vars();
    }
    esp_deep_sleep_disable_rom_logging(); // suppress boot messages
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(msToSleep * mS_TO_uS_FACTOR));
    // esp_err_t setTimer = esp_sleep_enable_timer_wakeup(msToSleep * mS_TO_uS_FACTOR);
    // if (setTimer != ESP_OK)
    // {
    //     ESP_LOGE(TAG, "SetTimer Failed: %d", setTimer);
    // }
    rtc_gpio_isolate(GPIO_NUM_12);
    // close_rak_uart();
    // rtc_gpio_isolate(GPIO_NUM_15);
    // rtc_gpio_isolate(GPIO_NUM_2);

    gettimeofday(&sleepTime, NULL);
    uptimeInMs += esp_timer_get_time() / 1000;
    // printUptime(uptimeInMs);
    if (swtState == SWT_SLEEP)
    {
        ESP_LOGW(TAG, "### sample Total Time:  %" PRIu64 "ms  Sleeping for %" PRIu64 "sec noMotionFactor: %d",
                 uptimeInMs - sampleInitialUptime, msToSleep / 1000, get_no_motion_factor());
        ESP_LOGW(TAG, "###################################");
    }
    else
    {
        ESP_LOGI(TAG, "Awake Total Time:  %" PRIu64 "ms  Sleeping for %" PRIu64 "sec",
                 uptimeInMs - sampleInitialUptime, msToSleep / 1000);
    }
    esp_deep_sleep_start();
    vTaskDelete(NULL);
}`

peter3099, 2023-08-31:
Code for deep sleep task

`void deep_sleep_swt_task(void *parameter)
{
    uint64_t msToSleep = (uint64_t *)parameter;
    // ESP_LOGW(TAG, "Sleeping swtState: %d for %" PRIu64 "sec NoMotionFactor: %d",
    //          swtState, msToSleep / 1000, get_no_motion_factor());
    if (swtState == SWT_WAIT_JOIN)
    { // deepsleep  to wait for next lora Join request
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        print_lora_vars();
    }
    else if (swtState == SWT_FAIL_JOIN)
    { // deepsleep  to wait for next lora Join request
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        print_lora_vars();
        ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(BTN2, 0));
    }
    else if (swtState == SWT_WAIT_CO2)
    { // deep sleep to wait for C02 sensor measurement
        gpio_deep_sleep_hold_en();
        gpio_hold_en(POWER_VCC);
        gpio_hold_en(CO2_EN);
        // gpio_num_t gpio_num_co2int = CO2_INT;
        ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(CO2_INT, 0)); // wake up when CO2 int is low
    }
    else if (swtState == SWT_SLEEP)
    { // deepsleep  to wait for next sample
        // vTaskDelay()
        turn_off_pin(POWER_VCC);
        turn_off_pin(CO2_EN);
        init_ulp_program();
        ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
        if (get_mot_en() == false)
        {
            ESP_LOGE(TAG, "MOT ROUTINE DISABLED");
            // ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(BTN2, 0));
        }
        print_lora_vars();
    }
    esp_deep_sleep_disable_rom_logging(); // suppress boot messages
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(msToSleep * mS_TO_uS_FACTOR));
    // esp_err_t setTimer = esp_sleep_enable_timer_wakeup(msToSleep * mS_TO_uS_FACTOR);
    // if (setTimer != ESP_OK)
    // {
    //     ESP_LOGE(TAG, "SetTimer Failed: %d", setTimer);
    // }
    rtc_gpio_isolate(GPIO_NUM_12);
    // close_rak_uart();
    // rtc_gpio_isolate(GPIO_NUM_15);
    // rtc_gpio_isolate(GPIO_NUM_2);

    gettimeofday(&sleepTime, NULL);
    uptimeInMs += esp_timer_get_time() / 1000;
    // printUptime(uptimeInMs);
    if (swtState == SWT_SLEEP)
    {
        ESP_LOGW(TAG, "### sample Total Time:  %" PRIu64 "ms  Sleeping for %" PRIu64 "sec noMotionFactor: %d",
                 uptimeInMs - sampleInitialUptime, msToSleep / 1000, get_no_motion_factor());
        ESP_LOGW(TAG, "###################################");
    }
    else
    {
        ESP_LOGI(TAG, "Awake Total Time:  %" PRIu64 "ms  Sleeping for %" PRIu64 "sec",
                 uptimeInMs - sampleInitialUptime, msToSleep / 1000);
    }
    esp_deep_sleep_start();
    vTaskDelete(NULL);
}`

Code after wake up

`void check_wakeup_reason()
{
    ESP_LOGI(TAG, "Check Wake UP reason"); // BTN2 to ask for a new sample
    esp_sleep_wakeup_cause_t wakeup_reason;
    wakeup_reason = esp_sleep_get_wakeup_cause();

    struct timeval timeNow;
    struct timeval timeDiff;
    gettimeofday(&timeNow, NULL);
    timersub(&timeNow, &sleepTime, &timeDiff);
    uptimeInMs += (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000);
    printf("WK reason %d Now: %" PRIu64 "ms, Duration: %" PRIu64 "ms\r\n",
           wakeup_reason, (timeNow.tv_sec * (uint64_t)1000) + (timeNow.tv_usec / 1000), (timeDiff.tv_sec * (uint64_t)1000) + (timeDiff.tv_usec / 1000));
    // rtc_gpio_hold_dis(GPIO_NUM_15);
    // rtc_gpio_hold_dis(GPIO_NUM_2);
    stop_ulp();}`

peter3099, 2023-08-31:
SDK Config
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_MMU_LINEAR_ADDRESS_REGION_NUM=5
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=64
CONFIG_SOC_DAC_PERIPH_NUM=2
CONFIG_SOC_DAC_RESOLUTION=8
CONFIG_SOC_GPIO_PORT=1
CONFIG_SOC_GPIO_PIN_COUNT=40
CONFIG_SOC_GPIO_VALID_GPIO_MASK=0xFFFFFFFFFF
CONFIG_SOC_GPIO_VALID_DIGITAL_IO_PAD_MASK=0xEF0FEA
CONFIG_SOC_I2C_NUM=2
[... 1687 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM=1024
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

peter3099, 2023-08-31:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12488529/sdkconfig.txt)

peter3099, 2023-09-01:
Just to add that I've changed the RTC source to 
[image]

And for the last 12 hours I haven't got the issue, could it be a bug with the 150kHz oscillator?

My problem is that now the RTC is draining more current, I don't need super high precision so for me the RC was ok

snehapawar221087, 2023-09-13:
Hello esp-lis,

We found some devices, where before going into 'not wakeup problem', quectel module stopped responding, and we got battery read from quectel is 65235(garbage). ESP file system didn't have saved files of that time, so it's because device didn't walkups for set time to collect sensor's reading.

can it be the issue due to battery/power related? OR some PCB track related issue??

what causes ESP32s3, not waking by set timer?? But system itself wakes up due to  watchdog-cpu reset.

any help would be appreciated.

snehapawar221087, 2023-09-14:
Hi esp-lis,
Device is in field, and it's a routine where we are turning on the Quectel module to read the battery. We don't verify it while reading whether it's genuine or garbage. 
We have observed that rarely battery voltage reading comes as a garbage, during that time sending is also failed or no reply from AT commands.

I gave you quectel's reference to let you know the status of device while going into 'no wakeup state'(timer wakeup stops working but external interrupts like button press, USB etc working during this time), this will make me think if it's power or pcb level issue?

If we just ignore everything, and focus on this that in what circumstances, esp could go into no wakeup state, even if timer wakeup value is been set, and starts working normally after CPU reset???

esp-lis, 2024-01-18:
> Hello esp-lis,
>
> Please find the log, attached with this reply.
>
> In this log, you can fine, sleep_time_ms as a time for which device was in deep sleep. kv_save_time as a device's rtc time, read when [...]

@snehapawar221087 I have given feedback on this problem. Could you please provide us with samples that can reproduce the problem.

soroosha, 2024-03-01:
I'm facing the same issue with ESP32 and `esp_sleep_enable_timer_wakeup`. The board is powered (red LED is on) but the wake up timer stops randomly after a while. This defeats the whole purpose of the wake up timer and is a big issue. Any updates? Mine is a simple program that wakes up once every 30s, scans WiFi and sets a GPIO pin and goes to sleep.

esp-lis, 2024-03-01:
@soroosha Could you please provide us with samples that can reproduce the problem.

soroosha, 2024-03-06:
> @soroosha Could you please provide us with samples that can reproduce the problem.

Here is the entire code: 

```
gpio_num_t mosfet = GPIO_NUM_4;
// gpio_num_t scan_led = GPIO_NUM_2;
int wifiRssiThreshold = -68;

void setup(void) {
  Serial.begin(115200);

  // wifi
  // Set WiFi to station mode and disconnect from an AP if it was previously connected.
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  pinMode(mosfet, OUTPUT);
  // pinMode(scan_led, OUTPUT);
  
  Serial.println("Setup done");
  esp_sleep_enable_timer_wakeup(20 * 1000000); 
}

// -------------------------------------------------------------------------
// Main loop
// -------------------------------------------------------------------------
void loop()
{
  Serial.println("\nScanning...");
  // digitalWrite(scan_led, HIGH);
  
  // WiFi.scanNetworks will return the number of networks found.
  int ssid_length  = MY_WIFI_SSID.length()+1;    //The +1 is for the 0x00h Terminator
  char ssid_char_array[ssid_length];
  MY_WIFI_SSID.toCharArray(ssid_char_array, ssid_length);
  int n = WiFi.scanNetworks(false, false, false, 100, 0, ssid_char_array);

  int min_rssi = -100;
  int best_rssi = min_rssi;
  for (int i = 0; i < n; ++i) {
    if (String(WiFi.SSID(i).c_str()) == MY_WIFI_SSID) {
      int rssi = int(WiFi.RSSI(i));
      Serial.printf("Net found. rssi: %2d\n", rssi);
      if ( rssi > best_rssi ) {
        best_rssi = rssi;
      }
    }
  }
  // digitalWrite(scan_led, LOW);

  // Delete the scan result to free memory for code below.
  WiFi.scanDelete();

  if(gpio_hold_dis(mosfet) != ESP_OK){
    Serial.println("\nFailed to stop holding pin");
  }

  // Turn mosfet off when home WiFi is in range
  if (best_rssi > wifiRssiThreshold){
    digitalWrite(mosfet, LOW);
  }else{
    digitalWrite(mosfet, HIGH);
  }

  if(gpio_hold_en(mosfet) != ESP_OK){
    Serial.println("\nFailed to hold pin");
  }

  // Wait a bit before scanning again.
  esp_deep_sleep_start();  
}
```

esp-lis, 2024-03-08:
@soroosha 
1. Can you provide a simpler reproduce code?
2. Is this issue on all chips? If it only exists on a small number of chips, you may need to provide a sample.

soroosha, 2024-03-08:
The entire code is 50 lines of code so I'm not sure how simpler it can be. And I have not tested it on a large number of chips. I don't think you'll be able to help me specially since the issue is sporadic and the timer seems to stop working after a few days (making it hard to reproduce the issue quickly). Thanks anyway

esp-lis, 2024-03-11:
@soroosha I have reported this issue back to the digital engineer, we need a sample that can reproduce this issue, and the program to reproduce it is simple enough, such as using the [deep_sleep](https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep) example (disable some options of `CONFIG_EXAMPLE_GPIO_WAKEUP`, `CONFIG_EXAMPLE_EXT0_WAKEUP`, `CONFIG_EXAMPLE_EXT1_WAKEUP` and `CONFIG_EXAMPLE_TOUCH_WAKEUP`) to reproduce it.

closed 2024-08-16 by Sherry616