espressif/esp-idf #11940: Send compressed .gz file from SD card to cloud via HTTPS

Noushadalik, 2023-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
We are using ESP IDF version 4.4.1. We want to send a compressed .gz file from SD card to Azure Cloud using SAS URI over HTTPS, how can it be possible? Can I use the esp_http_server source file to transfer compressed files directly to the cloud?

Noushadalik, 2023-07-26:
hello,
any update?

hmalpani, 2023-07-26:
Hello @Noushadalik 
Apologies for the delayed reply.
If I understand correctly, you want to read the file from sdcard and then send it to the Azure server. I think you can refer to [SD card examples](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card) to read the file from sd card. You can use `esp_http_client` to post the file contents to the server.

Noushadalik, 2023-07-27:
We have already implemented sd card and HTTP client to transfer a plain text file to cloud using http post request. Now we want to send the compressed file in .gz format to the cloud without reading the compressed file. Is it possible for me to send the file without reading it in the esp HTTP client?

hmalpani, 2023-08-01:
Hello @Noushadalik 
You can read the file stored in .gz format on SD card and without decompressing it, you can send the read file to the cloud.

hmalpani, 2023-08-24:
Closing this issue. Please feel free to reopen if you have more updates.

closed 2023-08-24 by hmalpani