espressif/esp-idf #11943: Zigbee color_control_cluster & zigbee2mqtt

elgerg, 2023-07-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm trying to use the following converter to control the RGB state of the onboard LED. Not too complicated but I'm not getting any data through the callback.

The converter is the basic lights example:
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/light.js

The code I'm using to initialize is a modified version of the ha_light_on_off example:

```
/*
 * SPDX-FileCopyrightText: 2021-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: LicenseRef-Included
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice,
 *    this list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form, except as embedded into a Espressif Systems
 *    integrated circuit in a product or a software update for such product,
 *    must reproduce the above copyright notice, this list of conditions and
 *    the following disclaimer in the documentation and/or other materials
 *    provided with the distribution.
 *
 * 3. Neither the name of the copyright holder nor the names of its contributors
 *    may be used to endorse or promote products derived from this software without
 *    specific prior written permission.
 *
 * 4. Any software provided in binary form under this license must not be reverse
 *    engineered, decompiled, modified and/or disassembled.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 * SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 * CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 */

#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_light.h"
#include "nvs_flash.h"

char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

/**
 * @note Make sure set idf.py menuconfig in zigbee component as zigbee end device!
*/
#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
[... 143 lines cut ...]
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_device_add_set_attr_value_cb(attr_cb);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    /* load Zigbee light_bulb platform config to initialization */
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    /* hardware related and device init */
    light_driver_init(LIGHT_DEFAULT_OFF);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

```

Any help to point me in the right direction would be appreciated.

Thanks

elgerg, 2023-07-29:
Ok, slight update. I have it partially working for colour now.

Adding the following:

```
    esp_zb_color_cluster_cfg_t colour_cfg;
    
    //working ish
    //colour_cfg.color_mode = ESP_ZB_ZCL_COLOR_CONTROL_COLOR_MODE_DEFAULT_VALUE;

    //not working
    //colour_cfg.color_mode = ((uint8_t)0x07);
    //colour_cfg.options = ESP_ZB_ZCL_COLOR_CONTROL_OPTIONS_DEFAULT_VALUE;
    //colour_cfg.options = ESP_ZB_ZCL_ATTR_COLOR_CONTROL_OPTIONS_ID;

    colour_cfg.enhanced_color_mode = ESP_ZB_ZCL_COLOR_CONTROL_ENHANCED_COLOR_MODE_DEFAULT_VALUE;
    colour_cfg.color_capabilities = ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID; //((uint16_t)0x0403);
    esp_zb_attribute_list_t *esp_zb_colour_cluster = esp_zb_color_control_cluster_create(&colour_cfg);
```

The following controls now work:
[image]

Does anyone know how I can get color_temp, color_temp_startup and power_on_behavior to work?

Thanks in advance

joshelboy, 2023-08-08:
Any updates on this?

elgerg, 2023-08-08:
@joshelboy no working updates from me. I'm still stuck 😭

xieqinan, 2023-08-10:
Hi @joshelboy ,
The esp-zigbee-sdk only temporarily stores attributes in volatile flash, which means that the color_temp_start and power_on_behavior features are not supported in the current version, and they will be support in this month. 

Color temperature control has been added in esp-zigbee-lib v0.7.4.

We provide more examples in the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk)

joshelboy, 2023-08-10:
Hi @xieqinan,
thank you so much for the response. Totally makes sense. 
Is the new version you were talking about also featuring low-power Zigbee? (Also mentioned in TZ-171)

chshu, 2023-08-31:
@joshelboy The Zigbee sleepy device example is available here: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device

closed 2025-02-26 by chshu