espressif/esp-idf #11947: i2c clock stretching problem

sabbath24, 2023-07-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We tried to use ICM42605 with esp32-s3 and esp-idf 4.4.4. This chip uses clock stretching. We can only read zeros as data from the chip. We can detect the probleam is the esp32-s3 does not handle the i2c clock stretching well. Then we try an older version of esp idf 4.4.0 and we realized the chip is working perfectly. Then we compare the low level i2c driver files in both branch, then in the 4.4.4 we can patch the components/hal/esp32s3/include/hal/i2c_ll.h file to look like the 4.4.0 one. And the problem was solved on the pathced 4.4.4 branch too.
This is the patch that we used:
`
diff --git "a/components/hal/esp32s3/include/hal/i2c_ll.h" "b/components/hal/esp32s3/include/hal/i2c_ll.h"
index 334fe8287a..7c66af44b2 100644
--- "a/components/hal/esp32s3/include/hal/i2c_ll.h"
+++ "b/components/hal/esp32s3/include/hal/i2c_ll.h"
@@ -110,12 +110,11 @@ static inline void i2c_ll_cal_bus_clk(uint32_t source_clk, uint32_t bus_freq, i2
     clk_cal->clkm_div = clkm_div;
     clk_cal->scl_low = half_cycle;
     // default, scl_wait_high < scl_high
-    // Make 80KHz as a boundary here, because when working at lower frequency, too much scl_wait_high will faster the frequency
-    // according to some hardware behaviors.
-    clk_cal->scl_wait_high = (bus_freq >= 80*1000) ? (half_cycle / 2 - 2) : (half_cycle / 4);
-    clk_cal->scl_high = half_cycle - clk_cal->scl_wait_high;
-    clk_cal->sda_hold = half_cycle / 4;
-    clk_cal->sda_sample = half_cycle / 2 + clk_cal->scl_wait_high;
+    clk_cal->scl_high = (bus_freq <= 50000) ?  half_cycle : (half_cycle / 5 * 4 + 4);
+    clk_cal->scl_wait_high = half_cycle - clk_cal->scl_high;
+    clk_cal->sda_hold = half_cycle / 2;
+    // scl_wait_high < sda_sample <= scl_high
+    clk_cal->sda_sample = half_cycle / 2;
     clk_cal->setup = half_cycle;
     clk_cal->hold = half_cycle;
     //default we set the timeout value to about 10 bus cycles
@@ -147,23 +146,18 @@ static inline void i2c_ll_update(i2c_dev_t *hw)
 static inline void i2c_ll_set_bus_timing(i2c_dev_t *hw, i2c_clk_cal_t *bus_cfg)
 {
     HAL_FORCE_MODIFY_U32_REG_FIELD(hw->clk_conf, sclk_div_num, bus_cfg->clkm_div - 1);
-    /* According to the Technical Reference Manual, the following timings must be subtracted by 1.
-     * However, according to the practical measurement and some hardware behaviour, if wait_high_period and scl_high minus one.
-     * The SCL frequency would be a little higher than expected. Therefore, the solution
-     * here is not to minus scl_high as well as scl_wait high, and the frequency will be absolutely accurate to all frequency
-     * to some extent. */
+    //scl period
     hw->scl_low_period.scl_low_period = bus_cfg->scl_low - 1;
     hw->scl_high_period.scl_high_period = bus_cfg->scl_high;
-    hw->scl_high_period.scl_wait_high_period = bus_cfg->scl_wait_high;
     //sda sample
-    hw->sda_hold.sda_hold_time = bus_cfg->sda_hold - 1;
-    hw->sda_sample.sda_sample_time = bus_cfg->sda_sample - 1;
+    hw->sda_hold.sda_hold_time = bus_cfg->sda_hold;
+    hw->sda_sample.sda_sample_time = bus_cfg->sda_sample;
     //setup
-    hw->scl_rstart_setup.scl_rstart_setup_time = bus_cfg->setup - 1;
-    hw->scl_stop_setup.scl_stop_setup_time = bus_cfg->setup - 1;
+    hw->scl_rstart_setup.scl_rstart_setup_time = bus_cfg->setup;
+    hw->scl_stop_setup.scl_stop_setup_time = bus_cfg->setup;
     //hold
     hw->scl_start_hold.scl_start_hold_time = bus_cfg->hold - 1;
-    hw->scl_stop_hold.scl_stop_hold_time = bus_cfg->hold - 1;
+    hw->scl_stop_hold.scl_stop_hold_time = bus_cfg->hold;
     hw->to.time_out_value = bus_cfg->tout;
     hw->to.time_out_en = 1;
 } 
`

o-marshmallow, 2023-07-25:
Hello @sabbath24 ,

On IDF v4.4.4, a bug was present in SDA sampling time, it was fixed on `release/v4.4` after applying commit 6022afc9808b8c644bdda9d6c765bc430fabb6e8 . I invite you to test cherry-picking that commit in your case.

The changes you see in `i2c_ll_update` shall be kept since the values used to not be in line the Technical Reference Manual I2C timings description.

mythbuster5, 2025-01-13:
Any update for this thread or did you try commit above?

mythbuster5, 2025-01-22:
If no, I will close this issue as 1). no further update. 2). v4.4.4 is EOL

closed 2025-01-22 by mythbuster5