espressif/esp-idf #11949: expose mac->write_phy_reg() and read_phy_reg(), with mutex

crackwitz, 2023-07-25:
### Is your feature request related to a problem?

I have several PHYs on the SMI bus. Only one of them is under control of ESP-IDF and its network stack.

The other PHYs are under my control. I access them like so:

```c
// ...
mac = esp_eth_mac_new_esp32(&esp32_emac_config, &mac_config);
// ...
err = mac->read_phy_reg(mac, phyad, regad, &data);
// ...
```

This is not a documented API.

This access may **fail** with [`ESP_ERR_INVALID_STATE`](https://github.com/espressif/esp-idf/blob/9a1cc59338aa151bc6333f85c8169d36a478fe2a/components/esp_eth/src/esp_eth_mac_esp.c#L87C58-L87C79) ("phy is busy") when ESP-IDF's network stack already accesses the SMI bus in that instant.

This also happens in the other direction: if I have an ongoing SMI access, ESP-IDF's access attempt may fail. ESP-IDF seems mildly tolerant of such failure but I've seen it hallucinate an Ethernet connection coming back up after I pulled the network cable. This was with an IP101G (like the Ethernet Kit) and with me nearly filling the SMI bus with nonsense reads.

### Describe the solution you'd like.

I would like 
1. these methods to become public documented API
2. these methods to not fail but _synchronize_ bus access using a mutex or similar suitable operation

### Describe alternatives you've considered.

My own code explicitly handles `ESP_ERR_INVALID_STATE` by retrying, after some FreeRTOS delay, until the call succeeds.

Right now I am _trusting_ that ESP-IDF will not fail or behave erratically under any intermittent failure to poll the PHY.

I have stress-tested ESP-IDF for how it handles congested access to the SMI bus. This is not the use case. My use case is occasional communication on the SMI bus, upto dozens of accesses per second.

crackwitz, 2023-07-28:
While we're at it, I'd like to have a _recursive_ lock on the SMI bus, so I can explicitly lock+unlock it for multi-transaction operations like read-modify-write or MMD access (802.3 Annex 22D: Clause 22 Access to Clause 45 Registers)

- https://www.linecall.de/wp-content/uploads/2016/11/802.3-2015_SECTION2.pdf
- https://www.ieee802.org/3/efm/public/nov02/oam/pannell_oam_1_1102.pdf

kostaond, 2023-07-31:
@crackwitz thanks for the feature request. Could you please be more specific and describe the use case in more details? Currently, it sounds me quite specific and I don't think it would be used by majority of ESP32 users.

crackwitz, 2023-07-31:
Our use cases may be considered specific in that all of this is beyond strictly establishing an Ethernet link for the ESP32.

Some use cases for direct SMI access to PHYs:
- configure registers for interrupt output and LED control (polarity, modes like speed/link/act)
- configure (enable/disable) *Auto-Negotiation*
- read negotiated link speed
- choose to include/exclude link speeds for *Link Speed Advertisement*
- prevent PHY from trying to establish a link

Even if those things were modeled as interfaces to PHY drivers, I am not sure that I as the user could safely access those. As far as I can see, anything ultimately accessing the SMI bus is currently under exclusive control of the ESP-IDF network stack, so those accesses would never fight for access to the SMI bus. Our use cases require the _application_ to access to those PHYs.

My current options are:
- carry a patch for ESP-IDF that adds a (recursive) mutex in the SMI access functions, or
- figure out a way to query FreeRTOS for when ESP-IDF will access the bus (I see accesses every two seconds), and avoid the bus around that time

kostaond, 2023-08-01:
Thanks for the reply. I understand the types of features to be configured in PHY via SMI. However, it's still hard to me to imagine a real device where multiple PHY's are used and managed by ESP32 but the only one is used for data interface to ESP32.

crackwitz, 2023-08-01:
It's an industrial piece of electronics that manages the data path between different ports. One could call it a "media converter" because the ports have *MII in common but aren't all RJ45 Ethernet.

kostaond, 2023-08-01:
So where the data path is connected to from these "ports"? Sorry for asking such questions but I need to understand to the matter so we could prioritize the feature request. I assume you understand that priority goes to requests which have potential to be used by wider audience.

igrr, 2023-08-01:
@crackwitz Just one note regarding

> My current options are:
> carry a patch for ESP-IDF that adds a (recursive) mutex in the SMI access functions, or

If I understand the issue correctly, it's not necessary to patch IDF. You can wrap the write_phy_reg, read_phy_reg functions like so:

```c++
/* declarations */
/* these two variables store the original read_phy_reg and write_phy_reg pointers */
esp_err_t (*read_phy_reg_inner)(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value);
esp_err_t (*write_phy_reg_inner)(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t reg_value);
/* these two functions are the new read_phy_reg and write_phy_reg implementations */
static esp_err_t read_phy_reg_wrap(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value);
static esp_err_t write_phy_reg_wrap(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value);
/* PHY access mutex functions */
static void acquire_mutex(void);
static void release_mutex(void);
static void create_mutex(void);

/* part of your ethernet init code */
void main() {
  create_mutex();
  mac = esp_eth_mac_new_esp32(&esp32_emac_config, &mac_config);
  /* save the pointers to the original functions */
  read_phy_reg_inner = mac->read_phy_reg;
  write_phy_reg_inner = mac->write_phy_reg;
  /* replace them with the wrappers */
  mac->read_phy_reg = read_phy_reg_wrap;
  mac->write_phy_reg = write_phy_reg_wrap;
  /* ... the rest of the initialization... */
}

/* register access wrappers */
static esp_err_t read_phy_reg_wrap(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value)
{
  acquire_mutex();
  esp_err_t result = read_phy_reg_inner(eth_handle, phy_addr, phy_reg, reg_value);
  release_mutex();
  return result;
}
static esp_err_t write_phy_reg_wrap(esp_eth_handle_t eth_handle, uint32_t phy_addr, uint32_t phy_reg, uint32_t *reg_value)
{
  acquire_mutex();
  esp_err_t result = write_phy_reg_inner(eth_handle, phy_addr, phy_reg, reg_value);
  release_mutex();
  return result;
}

static void acquire_mutex() { ... }
static void release_mutex() { ... }
static void create_mutex() { ... }
```

This way, the implementation of PHY register access stays in IDF, but you can wrap it with your own functions which would acquire a mutex. Since both IDF and your code will call mac->read_phy_reg, mac->write_phy_reg, the calls will invoke the wrappers (read_phy_reg_wrap, write_phy_reg_wrap above).

crackwitz, 2023-08-01:
>   mac->read_phy_reg = read_phy_reg_wrap;
>   mac->write_phy_reg = write_phy_reg_wrap;

Thank you! That looks like it will absolutely solve our issue! I never considered replacing the function pointers in that internal structure!

I'll evaluate this in the coming days and most likely close the issue then.

crackwitz, 2023-08-02:
> ```c++
>   mac->read_phy_reg = read_phy_reg_wrap;
>   mac->write_phy_reg = write_phy_reg_wrap;
> ```

Works like a charm. Thanks!

closed 2023-08-02 by crackwitz