espressif/esp-idf #11950: Publish DETAILED example  for Zigbee Gateway

DerPicknicker, 2023-07-25:
### Is your feature request related to a problem?

Not directly - only the fact that the zigbee examples are lacking some information and useful example codes… See the next code what I would expect.

### Describe the solution you'd like.

What I would expect is a documentation and a code which solves the following: 

- How to start a searching for new devices 
- How to successfully pair a device 
- How to change the operation mode (each powered zigbee device should be working as a repeater the esp32 should be able to do the same)
- How to pair devices to a specific repeater / gateway
- A documentation how much devices a single esp32 can Handle 100,200,10.000? 
- How to ensure a stable re-connection if a device is offline 
- How to send data via zigbee send a custom status via zigbee (current status of an variable or an simple string) 

### Describe alternatives you've considered.

Nothing because espressif should provide this.

### Additional context.

This would be great to know if this is already implemented and only the examples aren’t available yet. But for me the zigbee stuff is not really clear and not good to implement as the other examples you provided. 

Please think about to provide more detailed instructions and documents and source code how to build a gateway which have the basic functionalities like:
- Paring 
- Re-Connect 
- Send / Recive Data 
- Route / Pair device to an specific device

Best regards!

DerPicknicker, 2023-07-28:
PUSH

campescassiano, 2023-12-26:
UP

chshu, 2024-06-14:
@campescassiano @DerPicknicker Sorry for the late response. The Zigbee Gateway is a complex device type with no standard implementation, as it is highly dependent on the ecosystem and application design.

For common Zigbee operations such as pairing, discovery, binding, and controlling, please refer to the [Zigbee examples](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples), the [HA_thermostat](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat) and [HA_temperature_sensor](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor)r examples are particularly useful references.

Additionally, we have implemented some gateway examples:

- [Basic Zigbee Gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway): A simple starting example with no application logic.
- [Matter + Zigbee Gateway example](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge): A Zigbee Matter Bridge example, functioning as both a Matter bridge device and a Zigbee Coordinator.
- [Rainmaker + Zigbee Gateway example](https://github.com/espressif/esp-rainmaker/tree/master/examples/zigbee_gateway): An ESP Rainmaker-based Zigbee gateway solution.

Generally, a Zigbee Gateway acts as a Zigbee coordinator, allowing standard Zigbee devices to join its network and mapping the Zigbee device types to another application's data model (such as Matter or Rainmaker).

Feel free to create issues here: https://github.com/espressif/esp-zigbee-sdk/issues, if you have any particular questions during implementation.

DerPicknicker, 2024-06-18:
@chshu .. 

Thank you for the reply. I decide now for another transmission technique. 

I hope you will reply faster in the future ;-)

closed 2024-11-13 by chshu