espressif/esp-idf #11953: ENC28J60  with esp32 checking on sending and receiving data while in a local network.

beginner140BB, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Here i am testing with ENC28J60 with esp32 for sending data between.
Is it possible to make it happen.

whether there any open source exe available to test it from the pc connected to local network where esp32 is connected.
As currently we are using chat server exe that is developed for learning purpose.

when runnning the example code 
code reference (https://github.com/espressif/esp-idf/blob/master/examples/ethernet/enc28j60/main/enc28j60_example_main.c)

I am able to connect to the network and able to ping the ESP32.

So,what if i need to send or receive some data through ethernet interface
(in case of Ethernet Data Frame Format the data i mentioned here will be payload i think )

Handling a high-level API without prior knowledge in the subject feels quiet difficult in arranging testing situation, 
as there are different API from esp-idf available quiet confusing that which API in which order will serve the purpose.

seeking help from anyone expertise or familiarize regarding this situation

nopnop2002, 2023-07-28:
https://github.com/nopnop2002/esp-idf-mirf

nopnop2002, 2023-07-28:
https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60

nopnop2002, 2023-07-28:
This sample uses ENC28J60 to get the IP address.

Once you have an IP address, it's the same as WiFi from then on.

kostaond, 2023-07-28:
Initialize the ENC28J60 as you already did and create a new socket application. [Here](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets) you can find inspiration.

I'm closing this issue since it since it is not an issue at all. Please use [ESP Forum](https://www.esp32.com/) next time for such type of questions. Thank you for understanding.

closed 2023-07-28 by kostaond