espressif/esp-idf #11954: Can serial ports be used for universal DMA transmission. Idf

iotPeanut, 2023-07-26:
Can serial ports be used for universal DMA transmission. Idf

iotPeanut, 2023-07-26:
chip：esp32s3

ESP-Marius, 2023-08-02:
Hi,

Could you explain a bit more in detail what you are looking for?

iotPeanut, 2023-08-04:
I want to bind to UHCI using GDMA, and then bind to UHCI using serial port 2. Because I have a large amount of data to send, using DMA does not take up CPU. Below is my configuration. When I call uart_gdma_write, there is no data output on the sending pin of serial port 2.The serial port configuration section is good, and it has been tested to send separately using serial port 2.  idf version is 4.4.4.Thank you for your answer.
Below is my test code：
dma_descriptor_t    *tx_desc;           ///< DMA descriptors
static gdma_channel_handle_t tx_channel;
uint8_t  *tx_dma_buf;                //dma buffer
uhci_seper_chr_t seper_char ={
    .sub_chr_en =0,
};

static inline void muhci_ll_init(uhci_dev_t *hw)
{
   // typeof(hw->conf0) conf0_reg;
    hw->conf0.clk_en = 1;
    hw->conf0.val = 0;
    hw->conf0.clk_en = 1;
    //hw->conf0.val = conf0_reg.val;
    hw->conf0.tx_rst = 1;
    hw->conf0.tx_rst = 0;
    hw->conf0.rx_rst = 1;
    hw->conf0.rx_rst = 0;
    hw->conf1.val = 0;
}

 esp_err_t uart_gdma_init(void)
{
    esp_err_t ret;
    int rx_ch_id = 0;
    gdma_channel_alloc_config_t channel_config_tx = {
        .direction = GDMA_CHANNEL_DIRECTION_TX,
    };

    ret = gdma_new_channel(&channel_config_tx, &tx_channel);
    if (ret != ESP_OK) {
        goto err;
    }
    gdma_strategy_config_t strategy_config = {
        .auto_update_desc = true,
        .owner_check = true
    };
    gdma_apply_strategy(tx_channel, &strategy_config);
    gdma_connect(tx_channel, GDMA_MAKE_TRIGGER(GDMA_TRIG_PERIPH_UART, 0));//  trig_periph.instance_id ==2
    tx_dma_buf = heap_caps_calloc(1, 2048, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA);
    if (!tx_dma_buf) {
        ret = ESP_ERR_NO_MEM;
        goto err;
    }

   //malloc dma descriptor
    tx_desc = heap_caps_calloc(1, (sizeof(dma_descriptor_t)), MALLOC_CAP_DMA);
    if (!tx_desc) {
        ret = ESP_ERR_NO_MEM;
        goto err;
    }
     tx_desc->dw0.size = 2048;
     tx_desc->dw0.length = 0;
     tx_desc->dw0.suc_eof = 1;
     tx_desc->dw0.owner = 1;
     tx_desc->buffer = tx_dma_buf;
     tx_desc->next = NULL;   
    gdma_get_channel_id(tx_channel, &rx_ch_id);
    ESP_LOGE(TAG, "acquire DMA channel, rx_ch_id=%d", rx_ch_id);
    gdma_ll_tx_reset_channel(&GDMA, rx_ch_id);

    muhci_ll_init(UHCI_LL_GET_HW(0));
    uhci_ll_attach_uart_port(UHCI_LL_GET_HW(0),2);

    uhci_ll_set_seper_chr(UHCI_LL_GET_HW(0), &seper_char);
    return ESP_OK;    
err:
    ESP_LOGE(TAG, "Failed to acquire DMA channel, Err=%d", ret);
    tx_channel = NULL;
    free(tx_dma_buf);
    free(tx_desc);
    return ret;
}

 void uart_gdma_write(dma_descriptor_t *desc,uint32_t len1)
{
    desc->dw0.length = len1;
    gdma_start( tx_channel, (intptr_t )desc);
}

Call in main.c
 uart_gdma_init();
 const char *source_str = "testtest\r\n";
  strcpy((char *)tx_dma_buf, source_str);
uart_gdma_write(tx_desc,10);
**logs:**
uart_gdma: acquire DMA channel, rx_ch_id=0

> Hi,
> 
> Could you explain a bit more in detail what you are looking for? What do you mean with "universal DMA transmission"?