espressif/esp-idf #11955: ESP32-C3 USB example.

v-3n0m, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was diging through esp32s3 devkit in the documentation it is stated as USB serial can use biderctional data transfer from PC to C3 and C3 to PC also. But  there is no example shocasing the example where as in s3 there is an example for using the USB serial device.

nopnop2002, 2023-07-28:
The ESP32-C3 has a USB serial/JTAG controller console.

It supports writing firmware using USB.

v-3n0m, 2023-07-28:
@nopnop2002 can't  I use it to send and receive bytes to and from the app code. Like the usb in s3 or s2 ?

nopnop2002, 2023-07-28:
You can't.

igrr, 2023-07-28:
There's an example of using usb_serial_jtag driver posted in another issue: https://github.com/espressif/esp-idf/issues/11948#issuecomment-1653034451, please take a look. (The topic there is about ESP32-S3, but the `usb_serial_jtag_driver_*` APIs work the same way on C6.)

nopnop2002, 2023-07-29:
@igrr 

It would be nice to have an official sample of usb_serial_jtag_driver_install.