espressif/esp-idf #11956: idf.py monitor baudrate is not applied and will always fall back to 115200 internally

timoxd7, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

After changing the monitor UART port to a custom one (UART0, 1, 3) with a custom baudrate, the IDF Monitor should also use this baudrate accordingly.

### What is the actual behavior?

After changing the monitor UART port to a custom one with a custom baudrate, the monitor only shows one line from bootloader, after that nothing or funny characters, depending on the baudrate.

### Steps to reproduce.

1. Flash project with standard UART
2. Works
3. Flash same project but with custom UART
4. Works
5. Flash same project but with custom UART and custom Baudrate
6. No monitor anymore

### Build or installation Logs.

```plain
idf.py monitor with custom Baudrate:

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 230400 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
f~f~xxxx~`���怘���怘~ff~f0x400810ac: call_start_cpu1 at C:/Users/timo.meyer/esp/esp-idf/components/esp_system/port/cpu_start.c:154

idf.py monitor without custom baudrate

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0038,len:6068
load:0x40078000,len:17276
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3512
entry 0x400805d0
I (85) cpu_start: Multicore app
I (85) cpu_start: Pro cpu up.
I (85) cpu_start: Starting app cpu, entry point is 0x400810ac
0x400810ac: call_start_cpu1 at C:/Users/timo.meyer/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (103) cpu_start: Pro cpu start user code
I (103) cpu_start: cpu freq: 160000000 Hz
I (103) cpu_start: Application information:
I (107) cpu_start: Project name:     esp-uart-empty
I (113) cpu_start: App version:      1
I (117) cpu_start: Compile time:     Jul 26 2023 13:29:55
I (124) cpu_start: ELF file SHA256:  049621c42c279423...
I (130) cpu_start: ESP-IDF:          v5.1-5-g6212e8b0d6-dirty
I (136) cpu_start: Min chip rev:     v0.0
I (141) cpu_start: Max chip rev:     v3.99
I (146) cpu_start: Chip rev:         v3.0
I (150) heap_init: Initializing. RAM available for dynamic allocation:
I (157) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (164) heap_init: At 3FFB27C8 len 0002D838 (182 KiB): DRAM
I (170) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (176) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (183) heap_init: At 4008BB84 len 0001447C (81 KiB): IRAM
I (189) spi_flash: detected chip: generic
I (193) spi_flash: flash io: qio
W (197) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (205) app_start: Starting scheduler on CPU0
32mI ()10) npt_ska t:arted ing sch
r on CPU1
I (210) main_task: Calling app_main()
W (210) ---: BOOTING! Build at:
W (220) ---: Jul 26 2023
W (220) ---: 13:30:14
```

### More Information.

I made an empty example repository: [timoxd7/esp-monitor-bug](https://github.com/timoxd7/esp-monitor-bug)

I was using the ESP32-WROVER-E.

Other tools were able to show the output of the UART port correctly, even after changing the baudrate to other values then 115200. With this in mind, the problem must be inside the "idf.py monitor" command.

dobairoland, 2023-07-26:
I'm sorry but I don't see in the linked example project that you would change the UART or the baudrate. Did you upload the correct version?

The following shows that monitor is connected with 230400  baud rate:
```
--- idf_monitor on \\.\COM6 230400 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
f�~���f~�x��x��x�x��~`�����怘���怘��~��f��f~f�0x400810ac: call_start_cpu1 at C:/Users/timo.meyer/esp/esp-idf/components/esp_system/port/cpu_start.c:154
```

In this state, are you able to connect with another terminal program with baud 230400?

timoxd7, 2023-07-26:
@dobairoland Yes, exactly. The example was started with "idf.py -p COM6 build encrypted-flash monitor", which has automatically set the baud from menuconfig, which was 230400 (but it could be anything above 115200, same result). However, for example with putty, i was able to get a valid UART output, with the same code without flashing again, just exiting the idf.py script and opening the port with putty.

dobairoland, 2023-07-26:
Thanks @timoxd7. Could you please update the example so we could try to reproduce this?

timoxd7, 2023-07-26:
@dobairoland Sorry for the inconvenience, i updated the example. It is just changing the baudrate for IDF monitor in menuconfig. Now the example is in the not-working state. An explicit call to `idf.py -p COM6 -b 230400 monitor` does also not work and produces the same issue.

peterdragun, 2023-07-31:
Hi @timoxd7 ,
I have tried the example provided by you and I was not able to reproduce the issue.
If you are using a custom baud rate you will always see the first characters (boot log) with decoding issues. Those are printed before the baud rate is changed. From your log, it looks like it is your case as well, as after first starting characters there is already a correctly decoded line: `0x400810ac: call_start_cpu1 at C:/Users/timo.meyer/esp/esp-idf/components/esp_system/port/cpu_start.c:154`
Does the output continue from this point?

What is weird is that it does work for you if you set the baud rate to 115200. For me the output looks like this:
```
--- idf_monitor on /dev/cu.usbserial-142140 230400 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
f~�f~��x�x������x�x~`怘�怘���~�f�f�~f������f�f`ff�������`�~f�x�ff�f`��f�f`�����f�����`f�������xf�~��f��~���~�f���f�ff~fxff�������fx�~���fx�~��`�fx�~�����fx�~����`�fx�~��~�fx�~��������`fx�`�������������x��x�~�������x��xx�xx�x���f�����xfx怘�����x������f�������怘�����~f�����x�����怘�����x����f�����������x�����f����x`怘�f��`��f��xf���I (98) cpu_start: Multicore app
I (98) cpu_start: Pro cpu up.
I (98) cpu_start: Starting app cpu, entry point is 0x400810ac
0x400810ac: call_start_cpu1 at /Users/peterdragun/Documents/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (107) cpu_start: Pro cpu start user code
I (107) cpu_start: cpu freq: 160000000 Hz
I (108) cpu_start: Application information:
I (109) cpu_start: Project name:     esp-uart-empty
I (112) cpu_start: App version:      6d8d65d-dirty
I (115) cpu_start: Compile time:     Jul 28 2023 16:06:57
I (118) cpu_start: ELF file SHA256:  9cc8dd8e16e2e86d...
I (121) cpu_start: ESP-IDF:          v5.1-458-g82fde1e823
I (124) cpu_start: Min chip rev:     v0.0
I (126) cpu_start: Max chip rev:     v3.99 
I (128) cpu_start: Chip rev:         v3.0
I (131) heap_init: Initializing. RAM available for dynamic allocation:
I (135) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (137) heap_init: At 3FFB2790 len 0002D870 (182 KiB): DRAM
I (141) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (144) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (147) heap_init: At 4008BB80 len 00014480 (81 KiB): IRAM
I (151) spi_flash: detected chip: generic
I (152) spi_flash: flash io: dio
W (154) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (161) app_start: Starting scheduler on CPU0
I (163) app_start:  rtimI (che) marn_t CPU St0mted on CPU0
I (173) main_task: Calling app_main()
W (173) ---: BOOTING! Build at:
W (173) ---: Jul 28 2023
W (173) ---: 16:07:11

Done
peterdragun@Peters-MacBook-Pro esp-monitor-bug % idf.py monitor -b 115200
/Users/peterdragun/Documents/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Serial port /dev/cu.usbserial-142140
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running idf_monitor in directory /Users/peterdragun/Documents/esp-monitor-bug
Executing "/Users/peterdragun/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/peterdragun/Documents/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-142140 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /Users/peterdragun/Documents/esp-monitor-bug/build/esp-uart-empty.elf -m '/Users/peterdragun/.espressif/python_env/idf5.1_py3.11_env/bin/python' '/Users/peterdragun/Documents/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbserial-142140 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5352
load:0x40078000,len:15084
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3404
entry 0x400805dc

k�`iF�������Bc�bc�F����d���处��
                               �b�*v�cC�F����BH
                                              ��Cv���(hh��b
                                                           k�!#3��R90���!j�1
                                                                            k�`� 3�

```
Which is consistent with starting with the default baud rate and then changing to a custom one.
Note: This is output from my macOS machine, but I have tried using Windows as well with the same results.

Just to be sure, can you please check the version of esp-idf-monitor package by running `pip show esp-idf-monitor`? But there shouldn't be a difference.

timoxd7, 2023-07-31:
Hi @peterdragun,
the output of `pip show esp-idf-monitor` is

```
Name: esp-idf-monitor
Version: 1.1.1
Summary: Serial monitor for esp-idf
Home-page: https://github.com/espressif/esp-idf-monitor
Author: Espressif Systems
Author-email:
License:
Location: C:\Users\timo.meyer\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages
Requires: esp-coredump, pyelftools, pyserial
Required-by:
```

I attached the full output after boot in my post, so there was no other ouput with different baudrates. I noticed the issue after i upgraded from v4.3.4 during migration of our code, as we use a custom baudrate and i suddenly saw nothing anymore printed inside the console. In my example, i only saw application output on the UART with the default baudrate 115200. I tested all common baudrates above, none working.

The line you mentioned that is `0x400810ac: call_start_cpu1 at C:/Users/timo.meyer/esp/esp-idf/components/esp_system/port/cpu_start.c:154` is only the address to function translation as i am guessing. It somehow receives the Pointer and decodes it correctly (the 0x400810ac) and prints the according function using the hex to line tool, but it doesn't even show the according line `I (98) cpu_start: Starting app cpu, entry point is 0x400810ac`. As the decode is successfull, the monitor somehow receives it but does not correctly give the characters from the received data to the monitor's console.

peterdragun, 2023-08-01:
Thanks for your reply. That is strange. Can you try running the monitor with `--no-hints` argument -> `idf.py --no-hints monitor`?

timoxd7, 2023-08-01:
Looks better now, with the `--no-hints` option it prints everything as expected:

```
PS C:\work\esp-uart-custom-broken> idf.py -p COM7 --no-hints monitor
C:\Users\timo\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory C:\work\esp-uart-custom-broken
Executing "C:\Users\timo\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\timo\esp\esp-idf\tools/idf_monitor.py -p COM7 -b 230400 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 C:\work\esp-uart-custom-broken\build\esp-uart-empty.elf --force-color -m 'C:\Users\timo\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe' 'C:\Users\timo\esp\esp-idf\tools\idf.py' '-p' 'COM7' '--no-hints'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM7 instead...
--- idf_monitor on \\.\COM7 230400 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
f~�f~�x�x�����x�x~`怘�怘���~�f�f�~f������f�f`ff������`�~f��ff�f`�f�f`����f����`f�������xf�~��f��~���f�ff~fxff�������fx�~���fx�~��`�fx�~�����fx�~����`�fx�~��~�fx�~�������`fx�`������������x��x�~������x��xx�xx�x���f����������x�����f������x������x���f���怘����x���f����xf��x怘�f��`��f��x���I (85) cpu_start: Multicore app
I (85) cpu_start: Pro cpu up.
I (85) cpu_start: Starting app cpu, entry point is 0x400810ac
0x400810ac: call_start_cpu1 at C:/Users/timo/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (94) cpu_start: Pro cpu start user code
I (94) cpu_start: cpu freq: 160000000 Hz
I (94) cpu_start: Application information:
I (96) cpu_start: Project name:     esp-uart-empty
I (99) cpu_start: App version:      6d8d65d
I (101) cpu_start: Compile time:     Aug  1 2023 16:16:00
I (104) cpu_start: ELF file SHA256:  6ed58398b0b24eaf...
I (107) cpu_start: ESP-IDF:          v5.1-7-g4426b3e3e4-dirty
I (110) cpu_start: Min chip rev:     v0.0
I (113) cpu_start: Max chip rev:     v3.99
I (115) cpu_start: Chip rev:         v3.0
I (118) heap_init: Initializing. RAM available for dynamic allocation:
I (121) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (124) heap_init: At 3FFB27C8 len 0002D838 (182 KiB): DRAM
I (127) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (130) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (134) heap_init: At 4008BB84 len 0001447C (81 KiB): IRAM
I (137) spi_flash: detected chip: generic
I (139) spi_flash: flash io: qio
W (141) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (145) app_start: Starting scheduler on CPU0
I (;32m ma14_tasp: starte Sta CPUg s0m
duler on CPU1
I (147) main_task: Calling app_main()
W (147) ---: BOOTING! Build at:
W (147) ---: Aug  1 2023
W (147) ---: 16:16:16

PS C:\work\esp-uart-custom-broken> idf.py -p COM7 monitor
C:\Users\timo\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory C:\work\esp-uart-custom-broken
Executing "C:\Users\timo\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\timo\esp\esp-idf\tools/idf_monitor.py -p COM7 -b 230400 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 C:\work\esp-uart-custom-broken\build\esp-uart-empty.elf --force-color -m 'C:\Users\timo\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe' 'C:\Users\timo\esp\esp-idf\tools\idf.py' '-p' 'COM7'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM7 instead...
--- idf_monitor on \\.\COM7 230400 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
f~f~xxxx~`���怘���怘~ff~f0x400810ac: call_start_cpu1 at C:/Users/timo/esp/esp-idf/components/esp_system/port/cpu_start.c:154
```

peterdragun, 2023-08-01:
Thanks, that helps! I will investigate more and fix this. For now please use `--no-hints` as a workaround.

peterdragun, 2023-09-20:
Hi, @timoxd7 can you please try again on the latest bug release - 5.1.1? As far as I can tell this should be fixed now, there was an issue with decoding the first characters (still using baud rate 115200) and then the console somehow stopped reading.

dobairoland, 2023-09-25:
I'm closing this for inactivity and under the assumption that the latest bugfix versions of ESP-IDF already addressed it.

closed 2023-09-25 by dobairoland