espressif/esp-idf #11957: See a High Level on GPIO39 on a ESP32-S3

franz-ms-muc, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html)

Type: ESP32-S3-DevKitC-1-N32R8V

i see a High-Level on GPIO39. (just running the Hello World)

how can this be ? 

High Level disappears if i push RESET.

franz-ms-muc, 2023-07-26:
Log of Hello World: 

[https://gist.github.com/franz-ms-muc/f9f981297ba4907e5b4194231c631916](https://gist.github.com/franz-ms-muc/f9f981297ba4907e5b4194231c631916)

franz-ms-muc, 2023-07-28:
0922117CDFA1E0EEDC
This one no issue

09224868B6B33E17A8
This one has this issue. 

Tested on 2 devkits. Behave different

theomeyer-tado, 2023-12-05:
I think this is not obviously documented but at reset GPIO39 is configured as a weak pull-up. See note 8 of table 2-1 :

[image]
[image]