espressif/esp-idf #11958: the bug about lp core i2c  in esp32 c6

DexiangYang, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1449-gd2471b11e7

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

when it call lp_core_i2c_master_write_to_device, it recive the data from i2c slave

### What is the actual behavior?

it blocking in lp_core_i2c_wait_for_interrupt called in reading loop becasue of LP_I2C_TRANS_COMPLETE_INT_ST_M disabled

### Steps to reproduce.

the LP_I2C_TRANS_COMPLETE_INT_ST_M interrupt will be detected, which cause LP_I2C_TRANS_COMPLETE_INT_ST_M disabled, the it will block to wait the trans completion when it reading. in the lp_core_i2c_master_write_to_device in lp_core_i2c.c, how to resolve the problem

DexiangYang, 2023-07-26:
the LP_I2C_TRANS_COMPLETE_INT_ST_M interrupt will be detected after writting some bytes, which cause LP_I2C_TRANS_COMPLETE_INT disabled. Then lp will block to wait the trans completion when it reading. in the lp_core_i2c_master_write_to_device in lp_core_i2c.c, how to resolve the problem.

DexiangYang, 2023-07-26:
it can be resolved add a end cmd when sending last write cmd. without the end cmd, it finish the transformation early. it works, can u try it? @sudeep-mohanty, thanks

DexiangYang, 2023-07-26:
`
esp_err_t lp_core_i2c_master_write_read_device(i2c_port_t lp_i2c_num, uint16_t device_addr,
                                            const uint8_t *data_wr, size_t write_size,
                                            uint8_t *data_rd, size_t read_size,
                                            int32_t ticks_to_wait)
{
    (void)lp_i2c_num;

    esp_err_t ret = ESP_OK;
    uint32_t cmd_idx = 0;

    if ((write_size == 0) || (read_size == 0)) {
        // Quietly return
        return ESP_OK;
    } else if ((write_size > UINT8_MAX) || (read_size > UINT8_MAX)) {
        // HW register only has an 8-bit byte-num field
        return ESP_ERR_INVALID_SIZE;
    }

    /* If SCL is busy, reset the Master FSM */
    if (i2c_ll_is_bus_busy(dev)) {
        i2c_ll_master_fsm_rst(dev);
    }

    /* Reset the Tx and Rx FIFOs */
    i2c_ll_txfifo_rst(dev);
    i2c_ll_rxfifo_rst(dev);

    /* Enable trans complete interrupt and end detect interrupt for read/write operation */
    uint32_t intr_mask = (1 << LP_I2C_TRANS_COMPLETE_INT_ST_S) | (1 << LP_I2C_END_DETECT_INT_ST_S);
    if (s_ack_check_en) {
        /* Enable LP_I2C_NACK_INT to check for ACK errors */
        intr_mask |= (1 << LP_I2C_NACK_INT_ST_S);
    }
    i2c_ll_enable_intr_mask(dev, intr_mask);

    /* Execute RSTART command to send the START bit */
    lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_RESTART, 0, 0, 0, 0);

    /* Write device addr and update the HW command register */
    uint8_t addr_len = 0;
    lp_core_i2c_config_device_addr(cmd_idx++, device_addr, LP_I2C_WRITE_MODE, &addr_len);

    /* Write data */
    uint32_t fifo_available = LP_I2C_FIFO_LEN - addr_len; // Initially, 1 or 2 fifo slots are taken by the device address
    uint32_t fifo_size = 0;
    uint32_t data_idx = 0;
    int32_t remaining_bytes = write_size;

    /* The data to be sent must occupy sequential slots of the Tx FIFO.
     * We must account for FIFO wraparound in case the length of data being sent is greater than LP_I2C_FIFO_LEN.
     */
    while (remaining_bytes > 0) {
        /* Select the amount of data that fits in the Tx FIFO */
        fifo_size = MIN(remaining_bytes, fifo_available);

        /* Update the number of bytes remaining to be sent */
        remaining_bytes -= fifo_size;

        /* Write data to the Tx FIFO and update the HW command register. Expect ACKs from the device */
        i2c_ll_write_txfifo(dev, &data_wr[data_idx], fifo_size);
        lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_WRITE, 0, LP_I2C_ACK, s_ack_check_en, fifo_size);

        // if (remaining_bytes) {
            /* This means we have to send more than what can fit in the Tx FIFO. Insert an End command. */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_END, 0, 0, 0, 0);
            cmd_idx = 0;
        // }

        /* Initiate I2C transfer */
        i2c_ll_update(dev);
        i2c_ll_trans_start(dev);

        /* Wait for the transfer to complete */
        ret = lp_core_i2c_wait_for_interrupt(intr_mask, ticks_to_wait);
        if (ret != ESP_OK) {
            /* Transaction error. Abort. */
            return ret;
        }

        /* Update data_idx */
        data_idx += fifo_size;

        /* We now have the full fifo available for writing */
        fifo_available = LP_I2C_FIFO_LEN;
    }

    /* Reset command index */
    cmd_idx = 0;

    /* Execute RSTART command again to send a START condition for the read operation */
    lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_RESTART, 0, 0, 0, 0);

    /* Write device addr again in read mode */
    lp_core_i2c_config_device_addr(cmd_idx++, device_addr, LP_I2C_READ_MODE, &addr_len);

    /* Read data */
    fifo_size = 0;
    data_idx = 0;
    remaining_bytes = read_size;

    /* The data is received in sequential slots of the Rx FIFO.
     * We must account for FIFO wraparound in case the length of data being received is greater than LP_I2C_FIFO_LEN.
     */
    while (remaining_bytes > 0) {
        /* Select the amount of data that fits in the Rx FIFO */
        fifo_size = MIN(remaining_bytes, LP_I2C_FIFO_LEN);

        /* Update the number of bytes remaining to be read */
        remaining_bytes -= fifo_size;

        /* Update HW command register to read bytes */
        if (fifo_size == 1) {
            /* Read 1 byte and send NACK */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_READ, LP_I2C_NACK, 0, 0, 1);

            /* STOP */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_STOP, 0, 0, 0, 0);
        } else if ((fifo_size > 1) && (remaining_bytes == 0)) {
            /* This means it is the last transaction.
             * Read fifo_size - 1 bytes and send ACKs
             */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_READ, LP_I2C_ACK, 0, 0, fifo_size - 1);

            /* Read last byte and send NACK */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_READ, LP_I2C_NACK, 0, 0, 1);

            /* STOP */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_STOP, 0, 0, 0, 0);
        } else {
            /* This means we have to read data more than what can fit in the Rx FIFO.
             * Read fifo_size bytes and send ACKs
             */
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_READ, LP_I2C_ACK, 0, 0, fifo_size);
            lp_core_i2c_format_cmd(cmd_idx++, I2C_LL_CMD_END, 0, 0, 0, 0);
            cmd_idx = 0;
        }

        /* Initiate I2C transfer */
        i2c_ll_update(dev);
        i2c_ll_trans_start(dev);

        /* Wait for the transfer to complete */
        ret = lp_core_i2c_wait_for_interrupt(intr_mask, ticks_to_wait);
        if (ret != ESP_OK) {
            /* Transaction error. Abort. */
            return ret;
        }

        /* Read Rx FIFO */
        i2c_ll_read_rxfifo(dev, &data_rd[data_idx], fifo_size);

        /* Update data_idx */
        data_idx += fifo_size;
    }

    return ret;
}
`

DexiangYang, 2023-07-26:
> Thanks for reporting the issue and a potential fix @DexiangYang. The changes seem fine as the END command should not alter the I2C protocol itself and should be fine to have it. However, before confirming the change, I would like test it more. May I ask if you have an easy way to reproduce this issue?

thank u for ur help, i try to connect esp32 c6 with sht30 or AT24C16. Neither can be accessed. u can try them. the  probability is 100%.

DexiangYang, 2023-07-26:
`#include <stdint.h>
#include <stdint.h>
#include <stdbool.h>
#include "ulp_lp_core.h"
#include "ulp_lp_core_print.h"
#include "ulp_lp_core_utils.h"
#include "ulp_lp_core_uart.h"
#include "ulp_lp_core_gpio.h"
#include "ulp_lp_core_i2c.h"
#include "hal/uart_hal.h"
#include "hal/i2c_hal.h"
#include "soc/pmu_reg.h"
#include "../bh1750_defs.h"

#define LP_I2C_TRANS_TIMEOUT_CYCLES 5000
#define LP_I2C_TRANS_WAIT_FOREVER   -1

#define LUX_THRESHOLD_LOW           5
#define LUX_THRESHOLD_HIGH          1000

static uint32_t sensor_on = 0;
static uint32_t res_update_done = 0;
volatile uint32_t lux = 0;

int main (void)
{
    uint8_t data_wr = 0;
    esp_err_t ret = ESP_OK;

    uint8_t tx_buf[1 << 4];
    uint8_t rx_buf[1 << 4];

    int64_t temp;

    lp_core_printf("Hello_world\n");

    while (1) {
        tx_buf[0] = 0xF3;
        tx_buf[1] = 0x2D;
        // * (uint16_t *) tx_buf = 0xF3;
        lp_core_printf("tx_buf[0] : %02x\n", tx_buf[0]);
        lp_core_printf("tx_buf[1] : %02x\n", tx_buf[1]);
        if(ESP_OK != (ret = lp_core_i2c_master_write_read_device(LP_I2C_NUM_0, 0x44, tx_buf, 2, rx_buf, 6, 500 * 1000))) {
            lp_core_printf("i2c failed\n");
            lp_core_printf("ret : %d\n", ret);
            while(1);
        }
        for(int i = 0; i < 6; i += 1) {
            lp_core_printf("%x ", rx_buf[i]);
        }
        lp_core_printf("\n");
        ulp_lp_core_delay_us(100 * 1000);
        
    }

    return 0;
}
`

DexiangYang, 2023-07-26:
this is lp code, when lp connected with sht30.

DexiangYang, 2023-07-26:
`

        tx_buf[0] = 0xF3;
        tx_buf[1] = 0x2D;
        // * (uint16_t *) tx_buf = 0x0F32;
        // lp_core_printf("tx_buf[0] : %02x\n", tx_buf[0]);
        // lp_core_printf("tx_buf[1] : %02x\n", tx_buf[1]);
        if(ESP_OK != (ret = lp_core_i2c_master_write_read_device(LP_I2C_NUM_0, 0x50, tx_buf, 1, rx_buf, 3, 500 * 1000))) {
            lp_core_printf("i2c failed\n");
            lp_core_printf("ret : %d\n", ret);
            while(1);
        }
        for(int i = 0; i < 2; i += 1) {
            lp_core_printf("%x ", rx_buf[i]);
        }
        lp_core_printf("\n");
        ulp_lp_core_delay_us(100 * 1000);`

sudeep-mohanty, 2023-07-27:
Hi @DexiangYang,
May I ask you to try the attached patch? It looks like the `lp_core_i2c_master_write_read_device()` did not send out a STOP command after the write operation and before starting the read operation.

[lp_i2c_write_read.patch](https://github.com/espressif/esp-idf/files/12183051/lp_i2c_write_read.patch)

DexiangYang, 2023-07-28:
> Hi @DexiangYang, May I ask you to try the attached patch? It looks like the `lp_core_i2c_master_write_read_device()` did not send out a STOP command after the write operation and before starting the [...]

thank u for ur help, while in my application, there is no stop between writing and reading according at24c16. the url is https://ww1.microchip.com/downloads/en/DeviceDoc/AT24C01C-AT24C02C-I2C-Compatible-Two-Wire-Serial-EEPROM-1Kbit-2Kbit-20006111A.pdf. so i add end instead of stop.