espressif/esp-idf #11960: What does this mean

franz-ms-muc, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

running the file_serving Sample at ESP-IDF 5.1
i get this ins the Log.

```
W (99191) httpd_uri: httpd_uri: Method '3' not allowed for URI '/ubus'
W (99191) httpd_txrx: httpd_resp_send_err: 405 Method Not Allowed - Specified method is invalid for this resource
```

what does it mean ?

Pieter-Vandaele, 2023-08-07:
At the bottom of the file_server.c file, there is a list of ```httpd_register_uri_handler``` the server responds to. If you see that line in your log, it means that the server received a message that is not part in the list.

In the example, there is HTTP_GET on all URIs (```/*```) and HTTP_POST on the URIs ```/upload/*``` and ```/delete/*```.
Your logline ```Method 3``` means HTTP_POST, but the URI ```/ubus``` is not part of the specified URI handlers. To fix, change any of the 2 present HTTP_POST handlers, or add one:

```
/* URI handler for deleting files from server */
    httpd_uri_t file_ubus = {
        .uri       = "/ubus",   // Match URIs of type /ubus
        .method    = HTTP_POST,
        .handler   = ubus_post_handler,
        .user_ctx  = server_data    // Pass server data as context
    };
    httpd_register_uri_handler(server, &file_ubus);
```

Don't forget to add the corresponding handler function if you decide to add a uri_handler

laukik-hase, 2023-08-14:
Thanks for the detailed answer @Pieter-Vandaele!

Hope this helps, @franz-ms-muc!

Closing this issue.

closed 2023-08-14 by laukik-hase

franz-ms-muc, 2023-10-06:
I use the Example: 

[https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving)

franz-ms-muc, 2023-10-06:
looking at a verbose trace, it looks like it is coming from Chrome. 

```

I (68314) wifi softAP: station 00:93:37:e4:ca:37 join, AID=1
D (68321) esp_netif_lwip: esp_netif_dhcps_cb esp_netif:0x3fcb8c98
I (68326) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
D (68334) esp_netif_lwip: Client's MAC: 0:93:37:e4:ca:37
D (68339) event: no handlers have been registered for event IP_EVENT:2 posted to loop 0x3fcb7d70
I (68462) wifi:<ba-add>idx:2 (ifx:1, 00:93:37:e4:ca:37), tid:0, ssn:14, winSize:64
D (68574) httpd: httpd_server: processing listen socket 54
D (68575) httpd: httpd_accept_conn: newfd = 58
D (68575) httpd_sess: httpd_sess_new: fd = 58
D (68578) httpd_sess: httpd_sess_new: active sockets: 1
D (68583) httpd: httpd_accept_conn: complete
D (68588) httpd: httpd_server: doing select maxfd+1 = 59
D (68588) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:2 posted to loop 0x3fcb7d70
D (68593) httpd: httpd_process_session: processing socket 58
D (68609) httpd_sess: httpd_sess_process: httpd_req_new
D (68614) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (68621) httpd_txrx: httpd_recv_with_opt: received length = 109
D (68626) httpd_parse: read_block: received HTTP request block size = 109
D (68633) httpd_parse: cb_url: message begin
D (68637) httpd_parse: cb_url: processing url = /ubus
D (68643) httpd_parse: verify_url: received URI = /ubus
D (68648) httpd_parse: cb_header_field: headers begin
D (68653) httpd_txrx: httpd_unrecv: length = 88
D (68658) httpd_parse: pause_parsing: paused
D (68662) httpd_parse: cb_header_field: processing field = Host
D (68668) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (68674) httpd_txrx: httpd_recv_with_opt: pending length = 88
D (68680) httpd_parse: read_block: received HTTP request block size = 88
D (68687) httpd_parse: continue_parsing: skip pre-parsed data of size = 5
D (68694) httpd_parse: continue_parsing: un-paused
D (68698) httpd_parse: cb_header_value: processing value = 192.168.4.1
D (68705) httpd_parse: cb_header_field: processing field = Content-Length
D (68712) httpd_parse: cb_header_value: processing value = 136
D (68718) httpd_parse: cb_header_field: processing field = Expect
D (68724) httpd_parse: cb_header_value: processing value = 100-continue
D (68731) httpd_parse: cb_header_field: processing field = Connection
D (68737) httpd_parse: cb_header_value: processing value = Keep-Alive
D (68744) httpd_parse: cb_headers_complete: bytes read     = 109
D (68750) httpd_parse: cb_headers_complete: content length = 136
D (68756) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:3 posted to loop 0x3fcb7d70
D (68756) httpd_parse: parse_block: parsed block size = 88
D (68771) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (68924) httpd_txrx: httpd_recv_with_opt: received length = 128
D (68925) httpd_parse: read_block: received HTTP request block size = 128
D (68926) httpd_txrx: httpd_unrecv: length = 128
D (68931) httpd_parse: pause_parsing: paused
D (68935) httpd_parse: cb_on_body: body begins
D (68940) httpd_parse: httpd_parse_req: parsing complete
D (68945) httpd_uri: httpd_uri: request for /ubus with type 3
D (68951) httpd_uri: httpd_find_uri_handler: [0] = /*
D (68956) httpd_uri: httpd_find_uri_handler: [1] = /upload/*
D (68962) httpd_uri: httpd_find_uri_handler: [2] = /delete/*
W (68968) httpd_uri: httpd_uri: Method '3' not allowed for URI '/ubus'
W (68975) httpd_txrx: httpd_resp_send_err: 405 Method Not Allowed - Specified method is invalid for this resource
D (68986) httpd_txrx: httpd_send_all: sent = 78
D (68991) httpd_txrx: httpd_send_all: sent = 2
D (68995) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:4 posted to loop 0x3fcb7d70
D (68995) httpd_txrx: httpd_send_all: sent = 45
D (69010) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb7d70
D (69010) httpd_sess: httpd_sess_delete: fd = 58
D (69020) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:0 posted to loop 0x3fcb7d70
D (69025) httpd_sess: httpd_sess_delete: active sockets: 0
D (69034) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:7 posted to loop 0x3fcb7d70
D (69040) httpd: httpd_server: doing select maxfd+1 = 56
I (69604) main: Core id: 0, prio: 1,  free heap: 8303152 bytes., minimum free heap: 8291040 bytes., minimum free stack: 1288 bytes.
VTC in safe state: SelectPreferredVT check for CF 1400 - Time: 00070327 (00063574) Boottime:  7 VTs:  0
[... 159 lines cut ...]
D (71504) httpd_sess: httpd_sess_process: httpd_req_delete
D (71505) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3fcb7d70
D (71510) httpd_sess: httpd_sess_process: success
D (71525) httpd: httpd_server: doing select maxfd+1 = 60
D (71531) httpd: httpd_process_session: processing socket 58
D (71536) httpd_sess: httpd_sess_process: httpd_req_new
D (71541) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (71548) httpd_txrx: httpd_recv_with_opt: received length = 0
D (71553) httpd_parse: read_block: connection closed
D (71558) httpd_sess: httpd_sess_delete: fd = 58
D (71563) httpd_sess: httpd_sess_delete: active sockets: 1
D (71563) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:7 posted to loop 0x3fcb7d70
D (71569) httpd: httpd_process_session: processing socket 59
D (71584) httpd_sess: httpd_sess_process: httpd_req_new
[... cut ...]
D (71612) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:7 posted to loop 0x3fcb7d70
D (71617) httpd: httpd_server: doing select maxfd+1 = 56
VTC in safe state: SelectPreferredVT check for CF 1400 - Time: 00077089 (00070327) Boottime:  7 VTs:  0
I (79607) main: Core id: 0, prio: 1,  free heap: 8303152 bytes., minimum free heap: 8291040 bytes., minimum free stack: 1288 bytes.
VTC in safe state: SelectPreferredVT check for CF 1400 - Time: 00083843 (00077089) Boottime:  7 VTs:  0
```

xuhongv, 2023-11-27:
@franz-ms-muc 

[image]

[image]

how can i fix it ?