espressif/esp-idf #11961: Error when save AP credentials when SSID size is 18 char wide

pritsios, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Helo
My esp32 is in WIFI_MODE_APSTA and after scanning I want to save the credentials of an AP in NVS with:
    		ESP_ERROR_CHECK(esp_wifi_set_config((wifi_interface_t)ESP_IF_WIFI_STA, &current_conf));
When AP SSID is 16 charaters wide works ok, but when the ssid is 18 characters wide I get an error report and the module crashes

Xiehanxin, 2023-07-27:
hi, @pritsios ,could you please provide your error log?
and please add more information, the IDF version. and the full name of the module you use (ESP32/ ESP32-C3/ ESP32-S3..)

pritsios, 2023-07-27:
Hello Xiehanxin
Thanks for the reply. I should have addressed that issue as a bug and then I saw there are some field to provide these information. Any way.

ESP-IDF = v5.1
module = ESP32-S2-DevCitC-1
IDE = Eclipse
Operating System = windows

Logs when ssid <=16 chars Everything is ok and dot not mind the last 2 lines it is due to I change the mode. This time I successfully connect with AP = "VAFIADIS STUDIOS"
[Logs_SSIDlessthan17.txt](https://github.com/espressif/esp-idf/files/12180837/Logs_SSIDlessthan17.txt)

Logs_SSIDmorethan18 with the error when I am trying to connect with AP = "VAFIADIS STUDIOS 4"
[Logs_SSIDmorethan18.txt](https://github.com/espressif/esp-idf/files/12180830/Logs_SSIDmorethan18.txt)

Thanks