espressif/esp-idf #11963: Build Error: random.cc:452: undefined reference to `getentropy'

franz-ms-muc, 2023-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

using ESP-IDF 5.1 i get the Following Error at a Build in Windows, 
however same Project build fine in Ubuntu: 

```
FAILED: Application.elf 
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/git/hr/LOGIBUS_integration_Hutschienenmoped/Application/build/Application.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T esp32.rom.eco3.ld -T memory.ld -T sections.ld @CMakeFiles\Application.elf.rsp -o Application.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(random.o):(.literal._ZNSt12_GLOBAL__N_117__libc_getentropyEPv+0x4): undefined reference to `getentropy'
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(random.o): in function `std::(anonymous namespace)::__libc_getentropy(void*)':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/random.cc:179: undefined reference to `getentropy'
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(random.o): in function `std::random_device::_M_init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/random.cc:452: undefined reference to `getentropy'
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libc.a(lib_a-arc4random.o):(.literal+0x28): undefined reference to `getentropy'
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libc.a(lib_a-arc4random.o): in function `_rs_stir':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/arc4random.c:89: undefined reference to `getentropy'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

0xjakob, 2023-07-28:
@franz-ms-muc Do you have any reproducing example for this? It seems that the libstdc++ is somehow configured to use `get_entropy()` in your case. But currently in IDF, `get_entropy()` is not used when libstdc++ queries random numbers. See also https://github.com/espressif/esp-idf/issues/11398.

franz-ms-muc, 2023-07-28:
i will make one. 
(i have a Open-Source Version of my Project, so it is easy to do so)

it could come from the 4diac Lib i link to the Project. 
why this not fails in Linux ? 

https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/org.eclipse.4diac.forte/-/blob/develop_me3/src/modules/utils/FB_RANDOM.cpp?ref_type=heads

https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/org.eclipse.4diac.forte/-/blob/develop_me3/src/modules/utils/FB_RANDOM.h?ref_type=heads

this is in fact a Random Number Generator, which is incuded there. 

see this Lines: 
```
  std::random_device mRandomDevice;  // Will be used to obtain a seed for the random number engine
  std::mt19937 mRandomGenerator; // Standard mersenne_twister_engine 

```

0xjakob, 2023-07-31:
> why this not fails in Linux ?

Yes, this is strange. I looked at the [code of random.cc](https://github.com/espressif/gcc/blob/esp-12_2_0/libstdc%2B%2B-v3/src/c%2B%2B11/random.cc#L174C1-L183C7). `getentropy()` shouldn't be used if `_GLIBCXX_HAVE_GETENTROPY` is not defined. I'll ask around if someone knows more about this...

0xjakob, 2023-07-31:
@franz-ms-muc `_GLIBCXX_HAVE_GETENTROPY` is indeed enabled in our toolchain for Windows. But we'll probably leave it as it is and rather implement `getentropy()`. This will be faster than waiting for a new toolchain release. See also https://github.com/espressif/esp-idf/issues/11398.

franz-ms-muc, 2023-07-31:
> @franz-ms-muc `_GLIBCXX_HAVE_GETENTROPY` is indeed enabled in our toolchain for Windows. But we'll probably leave it as it is and rather implement `getentropy()`. This will be faster than waiting for a new toolchain release. See also #11398.

so you mean: 

the Toolchain is configured different in Windows and in Linux ?

0xjakob, 2023-08-01:
> the Toolchain is configured different in Windows and in Linux ?

It pretty much seems so, yes.

closed 2023-08-19 by espressif-bot

0xjakob, 2023-08-21:
@franz-ms-muc We merged a fix. If you can, please try to build again with the latest master version.

andychess, 2024-01-04:
I'm also getting this error using a Mac and ESP32-C6.
[image]

andychess, 2024-01-04:
Here is the relevant code:
[image]

Lapshin, 2024-02-21:
> I'm also getting this error using a Mac and ESP32-C6.

@andychess , sorry for the late response. Please try the latest toolchain version  https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20230928