espressif/esp-idf #11964: ip101: ip101_init(175): failed to init PHY

Sky-Soo-Ap, 2023-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-1709-g4fc2e5cb95

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Init PHY always okay, not 1 time is okay, and 1 time failed to init PHY.
Following is sdkconfig and log for more details.
[sdkconfig.log](https://github.com/espressif/esp-idf/files/12179747/sdkconfig.log)
[0727_Failed_to_init_PHY.log](https://github.com/espressif/esp-idf/files/12179750/0727_Failed_to_init_PHY.log)

### What is the actual behavior?

ESP32 firmware restart, 1 time is okay, and 1 time failed to init PHY.

### Steps to reproduce.

1. Init ethernet PHY
2. Restart

### Debug Logs.

```plain
[Failed to init PHY]
W (13:38:53.683) esp_eth: Call read_phy_reg
W (13:38:53.685) eth_phy_802_3: bmcr.val:65535
W (13:38:53.690) eth_phy_802_3: After write bmcr.val:63487
W (13:38:53.696) eth_phy_802_3: bmcr.power_down:0, bmcr.val:63487
W (13:38:53.712) esp_eth: Call read_phy_reg
W (13:38:53.712) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.722) esp_eth: Call read_phy_reg
W (13:38:53.722) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.732) esp_eth: Call read_phy_reg
W (13:38:53.732) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.742) esp_eth: Call read_phy_reg
W (13:38:53.744) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.752) esp_eth: Call read_phy_reg
W (13:38:53.756) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.772) esp_eth: Call read_phy_reg
W (13:38:53.772) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.782) esp_eth: Call read_phy_reg
W (13:38:53.782) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.792) esp_eth: Call read_phy_reg
W (13:38:53.793) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.802) esp_eth: Call read_phy_reg
W (13:38:53.805) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
W (13:38:53.812) esp_eth: Call read_phy_reg
W (13:38:53.817) eth_phy_802_3: After bmcr.power_down:1, bmcr.val:65535
E (13:38:53.824) eth_phy_802_3: eth_phy_802_3_pwrctl(166): power up timeout, to:10, phy_802_3->reset_timeout_ms / 10:10
E (13:38:53.835) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(333): power control failed
E (13:38:53.844) ip101: ip101_init(175): failed to init PHY

[Init PHY okay]
W (13:38:56.136) esp_eth: Call read_phy_reg
W (13:38:56.138) eth_phy_802_3: bmcr.val:65535
W (13:38:56.143) eth_phy_802_3: After write bmcr.val:63487
W (13:38:56.149) eth_phy_802_3: bmcr.power_down:0, bmcr.val:63487
W (13:38:56.165) esp_eth: Call read_phy_reg
W (13:38:56.166) eth_phy_802_3: After bmcr.power_down:0, bmcr.val:12544
W (13:38:56.175) esp_eth: Call read_phy_reg
W (13:38:56.176) esp_eth: Call read_phy_reg
W (13:38:56.177) esp_eth: Call read_phy_reg
W (13:38:56.182) esp_eth: Call read_phy_reg
W (13:38:56.187) esp_eth: Call read_phy_reg
I (13:38:56.193) esp_eth.netif.netif_glue: 90:38:0c:5e:bf:0f
I (13:38:56.199) esp_eth.netif.netif_glue: ethernet attached to netif
W (13:38:56.206) azp_eth: 0x3ffb4708   90 38 0c 5e bf 0f                                 |.8.^..|
W (13:38:56.215) esp_eth: Call read_phy_reg
W (13:38:56.315) esp_eth: Call read_phy_reg
W (13:38:56.415) esp_eth: Call read_phy_reg
W (13:38:56.515) esp_eth: Call read_phy_reg
W (13:38:56.615) esp_eth: Call read_phy_reg
W (13:38:56.715) esp_eth: Call read_phy_reg
W (13:38:56.815) esp_eth: Call read_phy_reg
W (13:38:56.915) esp_eth: Call read_phy_reg
W (13:38:57.015) esp_eth: Call read_phy_reg
W (13:38:57.115) esp_eth: Call read_phy_reg
W (13:38:57.215) esp_eth: Call read_phy_reg
W (13:38:57.315) esp_eth: Call read_phy_reg
W (13:38:57.415) esp_eth: Call read_phy_reg
W (13:38:57.515) esp_eth: Call read_phy_reg
W (13:38:57.615) esp_eth: Call read_phy_reg
W (13:38:57.715) esp_eth: Call read_phy_reg
W (13:38:57.815) esp_eth: Call read_phy_reg
W (13:38:57.915) esp_eth: Call read_phy_reg
W (13:38:58.015) esp_eth: Call read_phy_reg
I (13:38:58.017) azp_eth: Ethernet Started
```

suda-morris, 2023-07-28:
since it's a customized board, please update the schematic as well, only the Ethernet part. We want to know how esp32 is connected with the PHY, and where the clock comes from.

Sky-Soo-Ap, 2023-07-28:
Hi [suda-morris](https://github.com/suda-morris),
Please note that using the same custom board:
[ESP-IDF 5.0.3 version]
Run 5.0.3 ethernet basic example is okay;
Then esp_restart() is still okay;
Only changed phy_config.phy_addr = -1 (using default value);

[ESP-IDF master version] 
Run master ethernet basic example, first time after flashed is okay;
Then esp_restart() is always failed to init PHY;
Only changed phy_config.phy_addr = 0 (default value -1 & 1 is not work);

Following is the log and source code.
[0728_Master.log](https://github.com/espressif/esp-idf/files/12192159/0728_Master.log)
[0728_5.0.3.log](https://github.com/espressif/esp-idf/files/12192161/0728_5.0.3.log)
[master_basic.zip](https://github.com/espressif/esp-idf/files/12192170/master_basic.zip)
[5.0.3_basic.zip](https://github.com/espressif/esp-idf/files/12192177/5.0.3_basic.zip)

kostaond, 2023-07-28:
It's strange the `bmcr.val:65535`. It should not be. The IP101 is either in some strange state or the SMI is not properly working. Could you please share your schematics?

Sky-Soo-Ap, 2023-07-31:
Hi [kostaond](https://github.com/kostaond),
Please help to take a look and the following is the schematics.
Thank so much.
[ESP32_IP101GR.pdf](https://github.com/espressif/esp-idf/files/12211138/ESP32_IP101GR.pdf)

kostaond, 2023-07-31:
Thanks for the schematics. Do I understand it correctly that ESP32 outputs RMII REF CLK via GPIO0? It's quit unusual solution in configuration such is yours. You use IP101 which does not produce signal on CLKO when in reset so it's safe to use CLK generated by PHY. You would save one component (no need for inverter) and the thing is the RMII REF CLK via GPIO0 is experimental feature.

Anyway, I suspect there might be some issue with clocking. How is your crystal connected, e.g. _"If a 50MHz clock is applied to pin10 TXCLK/50M_CLKI, X1 must be connected to GND."_
Since the nonsense is read from registers, there might be issue with CLK. See, _"For a complete reset function. 25MHz clock (x1) must be active for a minimum of 10 clock cycles before the rising edge of RESET_N. Chip will be able to operate after 2.5ms delay of the rising edge of RESET_N. The 2.5ms extension is to ensure the stability of system power"_ You can try to add delay after `esp_eth_mac_new_esp32`.

Sky-Soo-Ap, 2023-08-01:
Hi [kostaond](https://github.com/kostaond),
I tried X1 connected to GND, it works.
Thanks so much and closed it.

Sky-Soo-Ap, 2023-08-01:
Hi [kostaond](https://github.com/kostaond),
After X1 connects to GND, the Ethernet keeps Link Up and Link Down.

kostaond, 2023-08-01:
Do you use WiFi anong with Ethernet on your board? If so, it can cause trouble. See note in GPIO0 as CLK output configuration: _"If you want the Ethernet to work with WiFi, don't select GPIO0 output mode for stability."_

Sky-Soo-Ap, 2023-08-02:
Hi [kostaond](https://github.com/kostaond),
I tried this method  “RMII Clock Sourced Externally by PHY” to init ethernet, it is okay.
Just the vscode monitor will cause init mac to fail.
Thanks so much and I closed it.

closed 2023-08-02 by Sky-Soo-Ap

confusedsakata, 2024-01-29:
I have a similar issue. I made my phy address -1 so it can search for the phy address but it cannot find the PHY address and then reboots. I'm using a custom pcb that is copying the configuration of the ip101 on this schematic https://dl.espressif.com/dl/schematics/SCH_ESP32-Ethernet-Kit_A_V1.2_20200528.pdf

One difference is R17 and R22 are switched. Originally R48 was missing too but I added it to check however still reboots happening. Using ethernet basic example on esp idf. Will try the delay solution mentioned here and report back

kostaond, 2024-01-29:
@confusedsakata please share schematics and logs so I could help you. Just a note regarding R17/R22, R17 is there to keep the IP101 in reset state during the board reset/power up. If you use IP101 as a source of REF RMII CLK and don't keep it in reset, the CLK is present at GPIO0 at boot. The thing is the GPIO0 is bootstrap pin and so you have 50:50 chance that the ESP32 boots to incorrect mode.

confusedsakata, 2024-01-29:
logs when i use address 1 for phy:

[1B][0;32mI (370) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (370) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;31mE (470) ip101: ip101_pwrctl(314): power up timeout[1B][0m
[1B][0;31mE (470) ip101: ip101_init(376): power control failed[1B][0m
[1B][0;31mE (470) esp_eth: esp_eth_driver_install(205): init phy failed[1B][0m

logs when i use -1 as address to search for phy:

[1B][0;32mI (463) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (463) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;31mE (473) eth_phy_802_3: No PHY device detected[1B][0m
[1B][0;31mE (483) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(332): Detect PHY address failed[1B][0m
[1B][0;31mE (483) ip101: ip101_init(175): failed to init PHY[1B][0m
[1B][0;31mE (493) esp_eth: esp_eth_driver_install(229): init phy failed[1B][0m
[1B][0;31mE (503) example_eth_init: eth_init_internal(90): Ethernet driver install failed[1B][0m

schematics for the relevant part which is the ip101 and esp32:
https://drive.google.com/drive/folders/1kV_ofIRisjIKrsWiI1e2akrAvZ-mmS9-?usp=drive_link

confusedsakata, 2024-01-30:
Ok new issue I have. 

When it works I get these logs:
[1B][0;32mI (363) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (373) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;32mI (393) esp_eth.netif.glue: e0:e2:e6:4c:7b:67[1B][0m
[1B][0;32mI (393) esp_eth.netif.glue: ethernet attached to netif[1B][0m
[1B][0;32mI (4393) server: Ethernet Started[1B][0m

But occassionally it wont work and then the reboots happen again with this log:
[1B][0;32mI (363) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (373) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;31mE (473) ip101: ip101_pwrctl(314): power up timeout[1B][0m
[1B][0;31mE (473) ip101: ip101_init(376): power control failed[1B][0m
[1B][0;31mE (473) esp_eth: esp_eth_driver_install(205): init phy failed[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40084590

Also when I press the reboot button I will get stuck with this log instead even after several reboots until I cycle the power:
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Schematics:
https://drive.google.com/drive/folders/1kV_ofIRisjIKrsWiI1e2akrAvZ-mmS9-?usp=drive_link

Currently R29 is 470 ohms and X1 is to ground now. Otherwise no change.

kostaond, 2024-01-30:
> Also when I press the reboot button I will get stuck with this log instead even after several reboots until I cycle the power:
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Read [my previous post](https://github.com/espressif/esp-idf/issues/11964#issuecomment-1914215211) and focus on note regarding R17/R22. The answer has been already provided there.

confusedsakata, 2024-01-30:
> Read [my previous post](https://github.com/espressif/esp-idf/issues/11964#issuecomment-1914215211) and focus on note regarding R17/R22. The answer has been already provided there.

Hi I switched the resistors and now its back to the reboot loop. This is the log I get:

[1B][0;32mI (363) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (373) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;31mE (473) ip101: ip101_pwrctl(314): power up timeout[1B][0m
[1B][0;31mE (473) ip101: ip101_init(376): power control failed[1B][0m
[1B][0;31mE (473) esp_eth: esp_eth_driver_install(205): init phy failed[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40084590

kostaond, 2024-01-31:
@confusedsakata I've briefly checked your schematics. However, based on the schematics and based on your comments, I'm not sure in which REF RMII CLK configuration you want to use the IP101. Are you aware that [there are multiple options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy)? The ESP32 Ethernet devkit is designed in a way that it can be use in any of those if you do appropriate component reconfiguration (resoldering).

confusedsakata, 2024-01-31:
@kostaond Yes I want to use the default configuration on the ESP32 Ethernet devkit which should be configuration A on those options.

kostaond, 2024-02-01:
You probably have some mistake in CLK configuration. Maybe even in data plane connection. I would suggest the following:
1) I recommend REF RMII CLK configuration _a)_ which is default of the ESP32 Ethernet devkit. Go over reference design once again, check all the comments and focus on CLK signal path (highlighted blue in the schematics).
2) If you go with CLK configuration _a)_, X1 must not be grounded.
3) Make sure the ESP-IDF RMII REF CLK is configured to be provided externally.
4) Once you are able to initialize the Ethernet, connect your device to PC and capture traffic using Wireshark. If you use `basic` example, you should see the device sends DHCP requests.
5) Increase log level to debug or edit [this line](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_esp.c#L335) to see if your device is able to receive something.
6) If you were still not able to get IP address, play with loopback. It will help you to isolate issue - if it is between ESP32 and PHY or PHY and RJ45. [Here](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/test_apps/main/esp_eth_test_apps.c#L312) is example.

confusedsakata, 2024-02-01:
So I took out the ground on X1 but kept the 470 ohm on IO0 of the ESP32. I used menuconfig then went to component config>Ethernet>Support ESP32 internal EMAC controller > RMII clock mode and then selected Input RMII clock from external. Saved it and then built and uploaded my esp idf ethernet basic example program. It caused the reboots again. I uploaded the same program to the devkit afterwards. No issues.

I looked at my schematics CLK signal path and I'm not sure what is different from the signal path from my schematic and the signal path from the devkit. They look the same now that I added 470 ohms at GPIO 0 for my schematic.

Here is the error logs I get:
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7196
ho 0 tail 12 room 4
load:0x40078000,len:14048
load:0x40080400,len:3644
entry 0x40080678
[1B][0;32mI (29) boot: ESP-IDF v4.3.6-142-ga6bcdf9637 2nd stage bootloader[1B][0m
[1B][0;32mI (29) boot: compile time 12:12:55[1B][0m
[1B][0;32mI (29) boot: chip revision: v3.0[1B][0m
[1B][0;32mI (33) boot.esp32: SPI Speed      : 40MHz[1B][0m
[1B][0;32mI (38) boot.esp32: SPI Mode       : DIO[1B][0m
[1B][0;32mI (43) boot.esp32: SPI Flash Size : 16MB[1B][0m
[1B][0;32mI (47) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (53) boot: Partition Table:[1B][0m
[1B][0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (78) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (86) boot: End of partition table[1B][0m
[1B][0;32mI (90) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1015ch ( 65884) map[1B][0m
[1B][0;32mI (123) esp_image: segment 1: paddr=00020184 vaddr=3ffb0000 size=021bch (  8636) load[1B][0m
[1B][0;32mI (126) esp_image: segment 2: paddr=00022348 vaddr=40080000 size=0c0cch ( 49356) load[1B][0m
[1B][0;32mI (149) esp_image: segment 3: paddr=0002e41c vaddr=00000000 size=01bfch (  7164) [1B][0m
[1B][0;32mI (151) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=30614h (198164) map[1B][0m
[1B][0;32mI (232) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (232) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (243) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (243) cpu_start: Starting app cpu, entry point is 0x4008114c[1B][0m
[1B][0;32mI (229) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (258) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (258) cpu_start: cpu freq: 160000000[1B][0m
[1B][0;32mI (258) cpu_start: Application information:[1B][0m
[1B][0;32mI (262) cpu_start: Project name:     ethernet_basic[1B][0m
[1B][0;32mI (268) cpu_start: App version:      1[1B][0m
[1B][0;32mI (272) cpu_start: Compile time:     Feb  1 2024 10:08:17[1B][0m
[1B][0;32mI (278) cpu_start: ELF file SHA256:  c2fa0b9f121d8843...[1B][0m
[1B][0;32mI (284) cpu_start: ESP-IDF:          v4.3.6-142-ga6bcdf9637[1B][0m
[1B][0;32mI (291) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (295) cpu_start: Max chip rev:     v3.99 [1B][0m
[1B][0;32mI (300) cpu_start: Chip rev:         v3.0[1B][0m
[1B][0;32mI (305) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (312) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (318) heap_init: At 3FFB3658 len 0002C9A8 (178 KiB): DRAM[1B][0m
[1B][0;32mI (324) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (331) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (337) heap_init: At 4008C0CC len 00013F34 (79 KiB): IRAM[1B][0m
[1B][0;32mI (345) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (348) spi_flash: flash io: dio[1B][0m
[1B][0;32mI (353) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;31mE (463) emac_esp32: emac_esp32_init(355): reset timeout[1B][0m
[1B][0;31mE (463) esp_eth: esp_eth_driver_install(204): init mac failed[1B][0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400844d8
file: "../main/ethernet_example_main.c" line 241
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x400844db on core 0

Backtrace:0x40081513:0x3ffb5b10 0x40085c81:0x3ffb5b30 0x4008afd2:0x3ffb5b50 0x400844db:0x3ffb5bc0 0x400d6767:0x3ffb5be0 0x400ffe58:0x3ffb5c60 0x40088e41:0x3ffb5c80

ELF file SHA256: c2fa0b9f121d8843

kostaond, 2024-02-02:
> emac_esp32: emac_esp32_init(355): reset timeout

Indicates that EMAC is not clocked...

Delete `ESP_ERROR_CHECK` from [this line](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c#L71C5-L71C20) (keep just `example_eth_init()`) and put `return;` at line 72. We just want to initialize the Ethernet and prevent reset. Now, measure CLK signal at 50M_CLKO, 50M_CLKI and GPIO0 using oscilloscope. Is signal there?

confusedsakata, 2024-02-02:
Ok so my ethernet basic example looks different maybe because I got it from the esp-idf examples thats provided by the expressif extension on vscode. My example doesnt have an example_eth_init function. However I had a second project that I downloaded the ethernet basic example from github. And when I uploaded that it didnt have the emac reset timeout issue when I uploaded it. I made the changes you mentioned by getting rid of esp-error_check and adding return. I see a signal for CLKO, CLKI at the IP101 and I see a signal at GPIO0. As a bonus I checked the crystal and it is giving a waveform as well. Here is the log I get with the code changes; this log looks the same before I made the code changes on this second project. The previous log I showed in the previous comment was the first project that had no example_eth_init. 

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
[1B][0;32mI (29) boot: ESP-IDF 5.0.2 2nd stage bootloader[1B][0m
[1B][0;32mI (29) boot: compile time 09:21:52[1B][0m
[1B][0;32mI (29) boot: chip revision: v3.0[1B][0m
[1B][0;32mI (32) boot.esp32: SPI Speed      : 40MHz[1B][0m
[1B][0;32mI (37) boot.esp32: SPI Mode       : DIO[1B][0m
[1B][0;32mI (41) boot.esp32: SPI Flash Size : 4MB[1B][0m
[1B][0;32mI (46) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (51) boot: Partition Table:[1B][0m
[1B][0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (77) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (85) boot: End of partition table[1B][0m
[1B][0;32mI (89) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b83ch ( 47164) map[1B][0m
[1B][0;32mI (114) esp_image: segment 1: paddr=0001b864 vaddr=3ffb0000 size=01efch (  7932) load[1B][0m
[1B][0;32mI (118) esp_image: segment 2: paddr=0001d768 vaddr=40080000 size=028b0h ( 10416) load[1B][0m
[1B][0;32mI (124) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1ade0h (110048) map[1B][0m
[1B][0;32mI (168) esp_image: segment 4: paddr=0003ae08 vaddr=400828b0 size=0948ch ( 38028) load[1B][0m
[1B][0;32mI (190) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (190) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;32mI (202) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (202) cpu_start: Starting app cpu, entry point is 0x40081674[1B][0m
[1B][0;32mI (0) cpu_start: App cpu up.[1B][0m
[1B][0;32mI (216) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (216) cpu_start: cpu freq: 160000000 Hz[1B][0m
[1B][0;32mI (216) cpu_start: Application information:[1B][0m
[1B][0;32mI (221) cpu_start: Project name:     ESPNOW_Reciever[1B][0m
[1B][0;32mI (226) cpu_start: App version:      6a2c57c-dirty[1B][0m
[1B][0;32mI (232) cpu_start: Compile time:     Feb  2 2024 09:21:05[1B][0m
[1B][0;32mI (238) cpu_start: ELF file SHA256:  cd87304a66c7892e...[1B][0m
[1B][0;32mI (244) cpu_start: ESP-IDF:          5.0.2[1B][0m
[1B][0;32mI (249) cpu_start: Min chip rev:     v0.0[1B][0m
[1B][0;32mI (253) cpu_start: Max chip rev:     v3.99 [1B][0m
[1B][0;32mI (258) cpu_start: Chip rev:         v3.0[1B][0m
[1B][0;32mI (263) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (276) heap_init: At 3FFB2800 len 0002D800 (182 KiB): DRAM[1B][0m
[1B][0;32mI (283) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (289) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (295) heap_init: At 4008BD3C len 000142C4 (80 KiB): IRAM[1B][0m
[1B][0;32mI (303) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (306) spi_flash: flash io: dio[1B][0m
[1B][0;33mW (310) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[1B][0m
[1B][0;32mI (324) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (0) cpu_start: Starting scheduler on APP CPU.[1B][0m
[1B][0;32mI (336) system_api: Base MAC address is not set[1B][0m
[1B][0;32mI (346) system_api: read default base MAC address from EFUSE[1B][0m
[1B][0;31mE (446) eth_phy_802_3: eth_phy_802_3_pwrctl(162): power up timeout[1B][0m
[1B][0;31mE (446) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(335): power control failed[1B][0m
[1B][0;31mE (446) ip101: ip101_init(175): failed to init PHY[1B][0m
[1B][0;31mE (446) esp_eth: esp_eth_driver_install(229): init phy failed[1B][0m
[1B][0;31mE (456) example_eth_init: eth_init_internal(76): Ethernet driver install failed[1B][0m
[1B][0;31mE (466) example_eth_init: example_eth_init(116): internal Ethernet init failed[1B][0m

kostaond, 2024-02-05:
> Ok so my ethernet basic example looks different maybe because I got it from the esp-idf examples thats provided by the expressif extension on vscode. My example doesnt have an example_eth_init function.

You probably selected older version of ESP-IDF to use in VS code extension. However, it doesn't matter, it works normally.

Now, when there is no reset due to EMAC init, do you have PHY address configured correctly? 

I'm afraid I cannot help you more :/ This is obviously hardware issue. I've already shared all the tips I used when I had issues with new board bring up...

kostaond, 2024-02-05:
Please try to apply below patch to the `master` ESP-IDF branch:

[ip101_reset.zip](https://github.com/espressif/esp-idf/files/14165778/ip101_reset.zip)

confusedsakata, 2024-02-06:
> Now, when there is no reset due to EMAC init, do you have PHY address configured correctly?

As in configured correctly on the hardware? It should be. Its configured the same way as the devkit. After all the address seems to work when I ground X1 but doesn't work when X1 is not grounded.

kostaond, 2024-02-07:
Strange... Try to cross check specifications of your crystal with "Table 13 Crystal Specifications" of IP101 datasheet. I briefly went over and it seems you don't meet them (Drive Level, Equivalent Series Resistance). Note that I'm not HW engineer so I don't know the exact implications but I think it worth trying if you don't have any other clue.

confusedsakata, 2024-02-08:
I will test a different crystal. Do you know what the exact name is for the crystal used on the devkit? I tried to find it online and nothing came up.

kostaond, 2024-02-14:
For other design documentation for the board, please contact us at [sales@espressif.com](mailto:sales%40espressif.com).

confusedsakata, 2024-02-15:
Ok so I tried a new crystal that has is closer to the specifications. It didnt fix anything. I uploaded the crystal waveforms I get for the eval board and the custom pcb in the same folder shared earlier https://drive.google.com/drive/folders/1kV_ofIRisjIKrsWiI1e2akrAvZ-mmS9-?usp=drive_link 

The pictures of the custom pcb is using the original crystal I'm using. The new crystal I'm using the waveforms look identical anyways. What could be causing this offset in the custom pcb?

kostaond, 2024-02-16:
It can be anything... I've even seen that manufacturer mounted incorrect part number of a part on one demo board.  Anyway, I'm sorry, I'm not hardware engineer, I cannot help you more. My last shot, please check R97 value and tolerance. It should be 6.19K(1%).

confusedsakata, 2024-02-17:
> It can be anything... I've even seen that manufacturer mounted incorrect part number of a part on one demo board. Anyway, I'm sorry, I'm not hardware engineer, I cannot help you more. My last shot, please check R97 value and tolerance. It should be 6.19K(1%).

Yeah its the correct resistance. Is there a way to communicate to the IP101 without the PHY address? Like make a fresh program where I try to communicate to the IC and then try to do what I want it to do by reading and writing to the registers instead of using the Ethernet basic example. Or since the PHY address is not working with ethernet basic it means theres nothing I can do to get around this issue unless I fix the issue hardware wise?

kostaond, 2024-02-17:
PHY address is always needed to communicate with PHY. See "5.9 Serial Management Interface" section of IP101 datasheet, for example.

confusedsakata, 2024-02-23:
I found that the pin for PHY_ADR3 has a 0.5V for my custom board but on the devkit its 0V. On another custom pcb its 0.7V. Is this something that will cause an issue with regards to the PHY?