espressif/esp-idf #11966: CMock SPI driver

Steven35700, 2023-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I want to use the SPI mock driver provided by ESP-IDF to test one of my components that uses the SPI. For that, I added it in the "CMakeLists.txt" of my project.

### What is the actual behavior?

Unfortunately, during compilation it seems that some dependencies are missing.

```
In file included from D:/Espressif/frameworks/esp-idf-v4.4.5/components/spi_flash/include/esp_flash_internal.h:19,
                 from D:/Espressif/frameworks/esp-idf-v4.4.5/components/spi_flash/esp_flash_api.c:24:
D:/Espressif/frameworks/esp-idf-v4.4.5/components/driver/include/driver/spi_common_internal.h:15:10: fatal error: esp_pm.h: No such file or directory
 #include "esp_pm.h"
          ^~~~~~~~~~
compilation terminated.
[115/1339] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp_flash_spi_init.c.obj
```

### Steps to reproduce.

1. Create a new project/example
2. Add "list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/driver")" in the project "CMakeLists.txt"
3. Run "idf.py build"

### More Information.

Here is my project "CMakeLists.txt" 

```
# This is the project CMakeLists.txt file for the test subproject
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS "../components/schedule"
                        "../components/log"
                        "../components/nvs"
)

list(APPEND EXTRA_COMPONENT_DIRS "../mocks/nvs") # My mock OK
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/driver") # ESP-IDF Mock NOK

set(TEST_COMPONENTS "schedule" CACHE STRING "List of components to test")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(test)
```

0xjakob, 2023-07-27:
@Steven35700 Please note that CMock only works on Linux. This means you need to change the target to "linux":
```bash
idf.py --preview set-target linux
```
This needs to be done before `idf.py build` is called.

Furthermore, it is necessary to set the `COMPONENTS` to `main` in the `CMakeLists.txt` file. This is to avoid drawing in incompatible components when building for Linux. 

Your `CMakeLists.txt` file should look like this:
```cmake
# This is the project CMakeLists.txt file for the test subproject
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

set(EXTRA_COMPONENT_DIRS "../components/schedule"
                        "../components/log"
                        "../components/nvs"
)

list(APPEND EXTRA_COMPONENT_DIRS "../mocks/nvs") # My mock OK
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/driver") # ESP-IDF Mock NOK

set(TEST_COMPONENTS "schedule" CACHE STRING "List of components to test")

project(test)
```

If you want to run unity unit tests as in IDF, you need to enable `CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER` in menuconfig, too.

closed 2023-07-27 by Steven35700

0xjakob, 2023-07-28:
@Steven35700 Happy to hear it works! Yes, I forgot to mention that the Linux target works much more seamlessly since IDF v5.0.