espressif/esp-idf #11967: Does the clock of i2s conflict with the clock of Ethernet ?

ArvinHou, 2023-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using IDF V4.4.5 and ADF V2.5.select lyrat_v4.3 board.
When I debug the network alone，it's OK.
When I debug the es8388 alone，it's OK too .
When I debug together，something wrong.
the es8388 works OK,but the ethernet will link down.

Pins used of my board:

LAN
LAN_RXD0 ---> IO25
LAN_RXD1 ---> IO26
LAN_CRS_DV ---> IO27
LAN_MDC ---> IO23
LAN_MDIO ---> IO18
LAN_TXD1 ---> IO22
LAN_TXD0 ---> IO19
LAN_TX_EN ---> IO21
LAN_RESET ---> IO5
LAN_REF_CLK ---> IO0

es8388
ASDOUT ---> IO35
DSDIN ---> IO33
LRCK ---> IO14
SCLK ---> IO2
MCLK ---> IO3
SDA ---> IO15
SCL ---> IO13

I find the ethernet link down when the bellow code run.  

/* Step 7: Set I2S clocks and start. No need to give parameters since configurations has been set in 'i2s_driver_init' */
    ESP_GOTO_ON_ERROR(i2s_set_clk(i2s_num, 0, 0, 0), err, TAG, "I2S set clock failed");

 Does the clock of i2s conflict with the clock of Ethernet ?How th solve it?

suda-morris, 2023-07-27:
There's only one APLL in ESP32. I2S can use APLL as a clock source and Ethernet can also use APLL to generate the RMII reference clock. If both of them want to be used together, then APLL must be set to 50MHz on the I2S side.
Starting from esp-idf v5.0, there's a lock for the APLL clock, which makes it easier to detect such conflict. 
https://github.com/espressif/esp-idf/commit/4f28b33bbcbcff1a018288cdf9b9bdbec9f1d148

L-KAYA, 2023-07-28:
Yes, on v4.4, the APLL in I2S and Ethernet are calculated separately, but they will try to acquire the same APLL since v5.0. The best way is to migrate your IDF to the later version like v5.0 or v5.1, and initialize Ethernet before I2S, then I2S will try to calculate the clock with 50MHz APLL (will still fail if the required MCLK frequency greater than 50MHz).

Another way is to turn off the  `use_apll` in I2S configuration field, so that I2S will use the 160MHz PLL as the clock source. No worry this clock source might be inaccurate, it has no difference for the normal sample rate like 8KHz ~ 192KHz

ArvinHou, 2023-07-28:
> If both of them want to be used together, then APLL must be set to 50MHz on the I2S side.

Set use_apll=false.But it doesn't work.Ethernet link down.

i2s_cfg.i2s_config.use_apll = false;
i2s_cfg.i2s_config.fixed_mclk = 0;

Set code bellow .Ethernet link down too.

i2s_cfg.i2s_config.use_apll = true;
i2s_cfg.i2s_config.fixed_mclk = 50000000;

In IDF V4.4.5,is there any way to solve it?

If ethernet using IO0 input extern clock  can  solve it?

L-KAYA, 2023-07-28:
As you configure I2S via ADF interface, I don't know too much about how ADF initialize I2S. But the I2S driver in IDF won't configure APLL if `use_apll` is not set.

L-KAYA, 2025-01-13:
Closed since the legacy I2S APIs are not recommended any more.

closed 2025-01-13 by Alvin1Zhang