espressif/esp-idf #11970: What is the status of domain: esp32.info ?

gojimmypi, 2023-07-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have several documentation references to the `esp32.info` domain that appears to now no longer be a valid.

Was that ever an official Espressif domain?

igrr, 2023-07-28:
Probably not an official one, no... I think it was first posted by Neil Kolban here: https://www.esp32.com/viewtopic.php?t=267#p1152, so I guess it was his initiative at that time. The updates you did in wolfSSL/wolfssl look correct to me!

gojimmypi, 2023-07-28:
Ok, thanks for the info! Cheers

closed 2023-07-28 by gojimmypi