espressif/esp-idf #11972: Support phy/cert_test for ESP32 chip

doragasu, 2023-07-28:
### Is your feature request related to a problem?

We are designing a product centered on the ESP32 chip. For the certification tests, we need access to the certification APIs, and currently they are only supported for ESP32-S3 and ESP32-C3.

### Describe the solution you'd like.

We would like the certification APIs to be ported and made available for the ESP32 chip (including the `phy/cert_test` example included in latest esp-idf).

### Describe alternatives you've considered.

We have considered using the certification test Windows application provided in Espressif webpage, but unfortunately it is not possible for us to use it because the system needs to configure additional peripherals for the board to work, and also during operation we have no direct access to the ESP32 UART.

nhhai, 2023-10-02:
Hi,

We're having this problem too. Built on espidf 5.1.1 and platformio 6.4.0:
```
example_app.c:447: undefined reference to `esp_phy_test_start_stop'
example_app.c:450: undefined reference to `esp_phy_ble_tx'
example_app.c:460: undefined reference to `esp_phy_ble_rx'
```

I added these to the include paths already:
```
${IDF_PATH}/components/esp_phy/esp32/include
${IDF_PATH}/components/esp_phy/include
```

Are there any workarounds to get this to work on the good old esp32 board?

rariman, 2023-10-04:
Hello,

I am also trying to get cert_test to work on esp32, any thoughts?

Thanks

jzwyssig, 2024-06-28:
Could you provide more details on the ESP32-C6? I noticed that the RF test tool binary includes an option for testing Thread/Zigbee channels. Are there specific header files available, similar to Bluetooth and Wi-Fi?

closed 2024-09-14 by espressif-bot