espressif/esp-idf #11973: xTASKSTATUS Runtime counter type

EnariSaarBolt, 2023-07-28:
### Is your feature request related to a problem?

FREERTOS task runtime counter overflow.

### Describe the solution you'd like.

configRUN_TIME_COUNTER_TYPE ulRunTimeCounter; to xTASK_STATUS in task.h

### Describe alternatives you've considered.

tracking overflow in expense of RAM

ESP-Marius, 2023-08-01:
Hi @EnariSaarBolt 

I seems like the changes you are requesting is [already a part of FreeRTOS v10.5.1](https://github.com/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel-V10.5.1/include/freertos/task.h#L166), which we are in-progress of updating to.

I'll link to this issue so that we finish the update you'll get a notification here.

AxelLin, 2023-10-03:
4d85cbff6d30cac120e00d53aa1984e026f64a09

closed 2023-11-07 by espressif-bot