espressif/esp-idf #11975: ESP32-C6: Touching GPIO19 causes GPIO20 to flicker/toggle with blank code

sfe-SparkFro, 2023-07-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C6-DevKitC-1-N8

### Power Supply used.

USB

### What is the expected behavior?

With a blank program, touching GPIO19 should not affect GPIO20 in any way.

### What is the actual behavior?

With a blank program, touching GPIO19 causes GPIO20 to flicker/toggle.

### Steps to reproduce.

1. Using the v5.1 release of the IDF, build [sample_project](https://github.com/espressif/esp-idf/tree/master/examples/get-started/sample_project) with no modifications, and target set to `esp32c6`.
2. Flash the example to an ESP32-C6 dev board, with nothing connected to any GPIO pins. I've used the ESP32-C6-DevKitC-1-N8, though I expect other boards will exhibit this behavior as well.
3. GPIO20 starts out as a high-impedance input. This can be verified by connecting an LED, it does not illuminate.
3. Connect GPIO19 to 3.3V, then to GND again. GPIO20 is now a low-impedance output, and will either drive high or low. This can be verified by connecting an LED to GPIO20, it illuminates.

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x17e8
entry 0x4086c410
I (23) boot: ESP-IDF v5.1 2nd stage bootloader
I (24) boot: compile time Jul 27 2023 09:12:31
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 40MHz   
I (31) boot.esp32c6: SPI Mode       : DIO        
I (36) boot.esp32c6: SPI Flash Size : 2MB        
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=42010020 size=07290h ( 29328) map
I (98) esp_image: segment 1: paddr=000172b8 vaddr=40800000 size=00d60h (  3424) load
I (101) esp_image: segment 2: paddr=00018020 vaddr=42000020 size=0cf28h ( 53032) map
I (119) esp_image: segment 3: paddr=00024f50 vaddr=40800d60 size=08308h ( 33544) load
I (128) esp_image: segment 4: paddr=0002d260 vaddr=40809070 size=01054h (  4180) load
I (132) boot: Loaded app from partition at offset 0x10000
I (132) boot: Disabling RNG early entropy source...
I (148) cpu_start: Unicore app
I (149) cpu_start: Pro cpu up.
W (158) clk: esp_perip_clk_init() has not been implemented yet
I (165) cpu_start: Pro cpu start user code
I (165) cpu_start: cpu freq: 160000000 Hz
I (165) cpu_start: Application information:
I (168) cpu_start: Project name:     main
I (172) cpu_start: App version:      1
I (177) cpu_start: Compile time:     Jul 27 2023 09:12:00
I (183) cpu_start: ELF file SHA256:  63832425b3d41a7e...
I (189) cpu_start: ESP-IDF:          v5.1
I (194) cpu_start: Min chip rev:     v0.0
I (198) cpu_start: Max chip rev:     v0.99 
I (203) cpu_start: Chip rev:         v0.0
I (208) heap_init: Initializing. RAM available for dynamic allocation:
I (215) heap_init: At 4080AF60 len 000716B0 (453 KiB): D/IRAM
I (221) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (228) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (235) spi_flash: detected chip: generic
I (239) spi_flash: flash io: dio
W (243) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (256) sleep: Configure to isolate all GPIO pins in sleep state
I (263) sleep: Enable automatic switching of GPIO sleep configuration
I (270) coexist: coex firmware version: ebddf30
I (276) coexist: coexist rom version 5b8dcfa
I (281) app_start: Starting scheduler on CPU0
I (285) main_task: Started on CPU0
I (285) main_task: Calling app_main()
I (285) main_task: Returned from app_main()
```

### More Information.

### Other info

For the record, I do understand floating pins. GPIO19 is clearly floating, and touching it causes it to change the state that it measures. The question I'm asking here is: how is it possible that one pin changing states causes a completely different pin to also change states, when my code doesn't configure either of them?

I recorded a video of this behavior, see below. In this test, the board is flashed with the blank sample_app. I've connected 2 LEDs to GPIO20 (yellow jumper wire); the red LED connects to GND to indicate when GPIO20 goes high, and the blue LED connects to 3V3 to indicate when GPIO20 goes low. I also have a scope probe connected to GPIO20. The white jumper is connected to GPIO19.

https://github.com/espressif/esp-idf/assets/102242586/7f7e59c8-d68e-4ed2-9009-20c4c5455047

When the board powers on (and after pressing the reset button), GPIO20 is a high-impedance input (indicated by neither LED illuminating). As soon as I touch GPIO19 (or even grab the jumper by the insulation), GPIO20 suddenly changes to an output, and changes rapidly between high and low outputs.

### Speculation

I suspect what's actually happening is that the pins are defaulting to the SDIO interface. GPIO19 is SDIO_CLK and GPIO20 is SDIO_D0, so it would make sense that because GPIO19 is floating, touching it causes a random number of "clock pulses", and GPIO20 just outputs data on those clock pulses. To test this, I connected a square wave to GPIO19 while watching with a scope. GPIO19 is the green channel, GPIO20 is the yellow channel:

[image]

The signal coming from GPIO20 repeats indefinitely, repeating every 70 clock cycles (7 low, 63 high). I also measured the other SDIO pins while feeding this square wave into GPIO19. GPIO21 (SDIO_D1) also behaves similarly - it begins as a high-impedance input, then changes to an output; however rather than toggling in a pattern, it just stays high the whole time. All other SDIO pins appear to remain as high-impedance inputs.

### Questions

1. Are the pins actually defaulting to the SDIO interface?
2. If so, is that intentional?
3. If not, is it a bug with the IDF?
4. If not, is it an issue with the C6 itself?
    * Alternatively, do I just have early silicon with a defect in it? (I see in the log that the chip revision is v0.0)

Any insight on this would be much appreciated, thank you!

nopnop2002, 2023-07-31:
>Are the pins actually defaulting to the SDIO interface?

See ESP32-C6 Series Datasheet.

https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf

remenyo, 2024-08-14:
I did not examine this througoutly yet, but just setting SDIO pins with `gpio_set_direction`, do not work with `gpio_set_level`. I have to call `gpio_reset_pin` on pins = 19-23 (Maybe more, but this is what I tested.)
Using ESP32-C6 devkit-M ( and a custom board with the same, ESP32-C6-MINI-1 )

For example, GPIO 0 does not need reset before setting `gpio_set_direction` on it.

sfe-SparkFro, 2024-08-14:
Forgot to close this before, oops! But yes, as described above, the pins default to SDIO, so need to re-configure the pins to avoid this behavior.