espressif/esp-idf #11976: OpenThread Border Router example assert failed: sys_timeout /IDF/components/lwip/lwip/src/core/timeouts.c

DonaldFraser, 2023-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S2-DevKitM-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

Building and running the Espressif Thread Border Router example https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br, I expect the Thread Border Router to run on the ESP-32 S2 devkit as described in the example with no assert failure causing. I have built and run an RCP device on an Espressif ESP32-H2 devkit and this RCP is connected to the ESP32 S2 running this thread border router example by UART as per the example page.

### What is the actual behavior?

ESP32-S2 idf.py monitor indicates an assertion failure, causing a reboot of the ESP32-S2 thread border router
assert failed: sys_timeout /IDF/components/lwip/lwip/src/core/timeouts.c:308 (Timeout time too long, max is LWIP_UINT32_MAX/4 msecs)
This pattern repeats on reboot, leading to continual reboots of the ESP32-S2 running the thread border router example
Looks similar to https://github.com/espressif/esp-idf/issues/10029, but with openthread, not BLE

### Steps to reproduce.

1. Follow the steps in Build,Flash, and Run section of https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br Using a ESP-32 S2 devkit as the thread border router, connected by UART, as shown on the same page to an ESP-H2 devkit already
flashed with the RCP client

### Debug Logs.

```plain
Debug Log

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002558d
0x4002558d: esp_restart_noos_dig at D:/ESP/esp-idf-v5.1/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x17c8
load:0x4004b000,len:0x4
load:0x4004b004,len:0xab4
load:0x4004f000,len:0x3160
entry 0x4004b1c4
I (26) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (26) boot: compile time Jul 27 2023 18:34:38
I (26) boot: chip revision: v0.0
I (30) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
[... 116 lines cut ...]
0x4014b38d: esp_openthread_route_table_add_route_entry at ??:?

0x4014a2e3: icmp6_raw_recv_handler at esp_openthread_infra_if.c.obj:?

0x400a1539: raw_input at D:/ESP/esp-idf-v5.1/components/lwip/lwip/src/core/raw.c:178

0x400abf19: ip6_input at D:/ESP/esp-idf-v5.1/components/lwip/lwip/src/core/ipv6/ip6.c:1055

0x400afb55: ethernet_input at D:/ESP/esp-idf-v5.1/components/lwip/lwip/src/netif/ethernet.c:229

0x4009f691: tcpip_thread_handle_msg at D:/ESP/esp-idf-v5.1/components/lwip/lwip/src/api/tcpip.c:174

0x4009f714: tcpip_thread at D:/ESP/esp-idf-v5.1/components/lwip/lwip/src/api/tcpip.c:148

0x4002f969: vPortTaskWrapper at D:/ESP/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 195e8c95fa07a484

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002558d
0x4002558d: esp_restart_noos_dig at D:/ESP/esp-idf-v5.1/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)
```

zwx1995esp, 2023-07-31:
@DonaldFraser Hi, this RIO `prefix 2A00:23C5:53B8:E01::, lifetime 315360000` is sent by one of your device? Notice, this timelift is not a reasonable time, due to the unit of this variable is second defined by [RFC 4861](https://www.rfc-editor.org/rfc/rfc4861#section-4.6.2). And time `315360000` is almost 36000 years, seems not a reasonable lifetime.

Also I will discuss about this with LWIP team, if the user set the timeout time too long,  should there be an assert or just print some error logs and set the default max timeout instead.

DonaldFraser, 2023-07-31:
Hi, I don't have any experience of RIO, only what I have read at https://openthread.io/reference/group/api-border-routing. I think you are suggesting that the thread border router example on my ESP32-S2 device is receiving an RIO network message externally - so probably from my home wifi router in this case, as that's the SSID I used in menuconfig for this example.
Maybe I can configure my router to send RIO messages with a much shorter timeout, although it's a basic home router I don't know if I can do this. Maybe I can configure my router with IPv4 instead of IPv6. I will try experimenting with router settings.

Donald

DonaldFraser, 2023-08-05:
I was unable to find any router settings on my home wifi router that allowed the thread border router to work. However I did have an old 4G wifi router lying around. I have set that up now, used idf.py menuconfig to change the openthread thread border router example config to use the SSID and password for my 4G router wifi network. The thread border router example running on my esp32-s2 with esp32-hr RCP,  is working with that wifi network, which is great, I can start to experiment and add some more ESP-32H2 devkit Thread end devices now. However it would be good to be able to get this working with my real home wifi network in future. I guess this issue is still relevant in the real world where RIO messages coming from wifi routers that people may have in their homes could specify very long lifetimes. As [zwx1995esp](https://github.com/zwx1995esp) commented, maybe logging an error and using the default max timeout would be more robust for real world wifi networks. I am able to proceed experimenting, using my other wifi router now, but would be good to have a solution for the future.

closed 2024-02-23 by chshu