espressif/esp-idf #11977: Same random number returned for Core 1

reinismu, 2023-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I use rust, but I assume issue is in underlying API

I use std Rust to create my firmware. It uses multiple threads (RTOS tasks) to generate random number. Issue I notice is that only half of my numbers are random. Other are the same number (Different after each reboot). Note that WIFI and BT is disabled.

'''
use esp_idf_sys::rand;
let random_num = unsafe { rand() as u32 };
'''

random_num should be random each call.

### What is the actual behavior?

'random_num' is the same for Core 1.

### Steps to reproduce.

1. Create simple Rust std project
2. Disable Wifi/BT
3. Run multiple threads that call random number
4. Check that 1 Core returns the same number

alberk8, 2023-07-30:
Is the expected behaviour when WIFI or Bluetooth are disabled.  Link [here ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/random.html?highlight=random%20number)

reinismu, 2023-07-30:
> It is the expected behavior when WIFI or Bluetooth are disabled. Link [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/random.html?highlight=random%20number)

As I understand true randomness doesnt work, but pseudo should work. Plus it works fine on 1 core

0xjakob, 2023-07-31:
@reinismu I cannot reproduce your issue on IDF with FreeRTOS tasks. Do you have any code that reproduces your case so that we or the Rust community can work on it?

In general, reading from the RNG always produces a different number, as suggested in the documentation. The function `esp_random()` is not thread-safe, though. It might be possible to read the RNG register at the exact same time, but since `esp_random()` is waiting for many CPU cycles to acquire entropy, this seems pretty unlikely.

0xjakob, 2023-08-08:
@reinismu Are there any updates on this issue or has it maybe been solved eventually?

reinismu, 2023-08-08:
> @reinismu Are there any updates on this issue or has it maybe been solved eventually?

I did hacky fix to add time to numbers on my end. Will try to do minimal project with this issue

0xjakob, 2023-10-17:
@reinismu We will close this issue for now (it is easier for use to deal only with "active" issues in order to not loose track). Please feel free to re-open it if there is any update about this issue which requires additional discussion.

closed 2023-10-17 by 0xjakob