espressif/esp-idf #11979: unrecognized command line option '-std=gnu++2b'; did you mean '-std=gnu++2a'?

krupis, 2023-07-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to run example project for the 2.1 inch round display T-Display RGB development board form LilyGO:
https://github.com/Xinyuan-LilyGO/T-RGB/tree/main/examples/esp-idf-example/T-RGB

and I am getting the following error message:
```
xtensa-esp32s3-elf-gcc.exe: error: unrecognized command line option '-std=gnu++2b'; did you mean '-std=gnu++2a'?
[1242/1359] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/widgets/lv_slider.c.obj
ninja: build stopped: subcommand failed.
```

I am using esp-idf version 5.0.3

and the gcc version:
```
PS D:\Programming\ESP32\RGB_Controller_Round\T-RGB> xtensa-esp32s3-elf-gcc --version
xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2021r2-patch5) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO 
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I would like to understand what is the root cause of this and how to solve this. I normally dont have any problems running any projects on ESP32 or ESP32-S3 its just this particular LilyGO sample project that I have issue building.

Lapshin, 2023-07-30:
@krupis  in your particular case you need just to execute `install.bat` and `export.bat` to fix the problem for idf v5.0.3

For some reason, you have the wrong toolchain to build your project.
Instead version:
```
xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2021r2-patch5) 8.4.0
```
For idf v5.0 it must be:
```
xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2022r1) 11.2.0
```

Don't you get any warnings about using the wrong toolchain?

krupis, 2023-07-31:
> @krupis in your particular case you need just to execute `install.bat` and `export.bat` to fix the problem for idf v5.0.3
>
> For some reason, you have the wrong toolchain to build your project. [...]

Yes this was totally my issue. I think I have accidentally selected esp-idf v4.4 toolchain (I have 2 esp-idf versions installed to support some old projects that we did on v4.4) for the esp-idf v.5.0+ project. 

I have configured to use different toolchain as you have suggested and now I can build the project without any issues.
Thanks

closed 2023-07-31 by krupis