espressif/esp-idf #11981: ESP32-S3-Mini-N8-1 SPI : MOSI data not matching with data sent. (IEP-1012)

radhikaalpha, 2023-07-31:
we are sending 0x48 on MOSI pin. But we are getting following waveform,
The SPI parameters are set as follows,

    spi_device_interface_config_t devcfg={

        .command_bits = 0,

        .address_bits = 0,

        .clock_speed_hz = 2000000

        .mode = 0,          //SPI mode 0

        /*

         * The timing requirements to read the busy signal from the EEPROM cannot be easily emulated

         * by SPI transactions. We need to control CS pin by SW to check the busy signal manually.

         */

        .spics_io_num = -1,

        .queue_size = 1,

        .flags = SPI_DEVICE_HALFDUPLEX,

        .pre_cb = cs_low, //cs_high,

        .post_cb = cs_high, //cs_low,

        .input_delay_ns = 240,

    };

Are we getting correct waveforms (BLUE - MOSI) &(YELLOW - SCK)for 0x48 with the above SPI settings?
Can you please guide us for the same?
[image]

wanckl, 2023-08-02:
@radhikaalpha Hi, can you provide your case more detail, e.g.    
- which idf version you use
- this is device config, what is your transaction code like
- use dma or not
- which API you use

thanks.

Alvin1Zhang, 2025-01-06:
Thanks for reporting, will close due to inactivity, feel free to reopen with more updates/details, thanks.

closed 2025-01-06 by Alvin1Zhang