espressif/esp-idf #11982: [ESP32S3] v4.4 MESH disconnecting periodically

KonssnoK, 2023-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello @zhangyanjiaoesp ,

i'm currently checking one installation, and i see that the devices are constanyl disconnecting and reconnecting to the same mesh layer.

The disconnection seems to occur at mesh lever every hour.
The reported disconnection cause is 102 MESH_REASON_LEAF

```
[27] STA: Send err 0x400A ESP_ERR_MESH_TIMEOUT

[1] <MESH_EVENT_PARENT_DISCONNECTED>reason: 102 MESH_REASON_LEAF

[3] STA: Send err 0x400B ESP_ERR_MESH_DISCONNECTED

[1] MQTT_EVENT_ERROR 32792 78 0 1 0

[1] MQTT task stopped after 1ms

[4] Triggering DYNAMIC MESH handover

[14] <MESH_EVENT_PARENT_DISCONNECTED>reason: 15 WIFI_REASON_4WAY_HAND

[13] <MESH_EVENT_PARENT_DISCONNECTED>reason: 205 WIFI_REASON_CONNECTI

[1] <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAI

[1] <MESH_EVENT_PARENT_CONNECTED>layer:2-->2, parent:ac:17:54:00:0a:

[1] GOT IP from mesh_sta

```

Do you have any idea why this error could trigger every hour?

Codebase 4fc8964ec36c44cd5b959b6b30f4ffb7ccbdcf91

zhangyanjiaoesp, 2023-08-01:
@KonssnoK 
Have checked the code, we didn't find a mesh related behavior that triggered once an hour. 
Maybe it related to the application, we need more information about the disconnecting device and its parent device to debug this issue. We need to know what happens to the mesh network when the device  periodically disconnecting and reconnecting.

KonssnoK, 2023-08-07:
@zhangyanjiaoesp i think the disconnections may be related to the fact that there is no router and the installation is running on fixed root... 
We are investigating (see the other issue we opened)
https://github.com/espressif/esp-idf/issues/12018

zhangyanjiaoesp, 2023-10-13:
@KonssnoK Do you still have this problem ?

KonssnoK, 2023-10-13:
i would think so, yes. 

[image]
A quick look at logs shows this error still being in the field.
We haven't had time to work on mesh lately, we plan on going back on it in a few weeks.

KonssnoK, 2023-10-20:
@zhangyanjiaoesp i have a quesetion because i forgot..

When sending to the external network, we use:
```
    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS | MESH_DATA_NONBLOCK, NULL, 0);
    if (err != ESP_OK) {
        LOGE(TAG, "STA: Send err 0x%X %s", err, esp_err_to_name(err));
    }
```

Sometimes, on the field, we see
`STA: Send err 0x400A ESP_ERR_MESH_TIMEOUT`
I think it's on networks with poor connectivity.

We currently have
`ESP_ERROR_CHECK(esp_mesh_send_block_time(2000));`

Does changing the send_block_time going to impact esp_mesh_send with MESH_DATA_NONBLOCK set??
I remember we already discussed this and the answer was "yes", but maybe i'm wrong! :)

Thanks!

zhangyanjiaoesp, 2023-10-20:
> @zhangyanjiaoesp i have a quesetion because i forgot..
>
> When sending to the external network, we use:
>
> ```
>     esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS | MESH_DATA_NONBLOCK, NULL, [...]

yes, you are right.

KonssnoK, 2023-10-20:
> yes, you are right.

could you tell me a bit more? 
the send_block_time is used for queues before non blocking part?

for example, if i have 3 seconds of block_time but i do a send with "data_nonblock", the function can be block for up to 3 seconds?

zhangyanjiaoesp, 2023-10-23:
@KonssnoK 

When you call `esp_now_send()`, if the `data.tos = MESH_TOS_P2P`, then the transmition will be blocked even you set the  `flag`  with `MESH_DATA_NONBLOCK`. 
If you set the `data.tos = MESH_TOS_DEF` and the `flag` with `MESH_DATA_NONBLOCK`, then the transmition will be unblocked.  And `MESH_TOS_P2P` should be used when sending packets upward.

KonssnoK, 2023-10-23:
Sorry could you map your answer to these? Thanks :)
```
#define MESH_DATA_ENC           (0x01)  /**< data encrypted (Unimplemented) */
#define MESH_DATA_P2P           (0x02)  /**< point-to-point delivery over the mesh network */
#define MESH_DATA_FROMDS        (0x04)  /**< receive from external IP network */
#define MESH_DATA_TODS          (0x08)  /**< identify this packet is target to external IP network */
#define MESH_DATA_NONBLOCK      (0x10)  /**< esp_mesh_send() non-block */
#define MESH_DATA_DROP          (0x20)  /**< in the situation of the root having been changed, identify this packet can be dropped by new root */
#define MESH_DATA_GROUP         (0x40)  /**< identify this packet is target to a group address */

```

zhangyanjiaoesp, 2023-10-24:
In the `esp-idf/components/esp_wifi/include/esp_mesh.h`,

The `data.tos` type is here:
```
/**
 * @brief For reliable transmission, mesh stack provides three type of services
 */
typedef enum {
    MESH_TOS_P2P,    /**< provide P2P (point-to-point) retransmission on mesh stack by default */
    MESH_TOS_E2E,    /**< provide E2E (end-to-end) retransmission on mesh stack (Unimplemented) */
    MESH_TOS_DEF,    /**< no retransmission on mesh stack */
} mesh_tos_t;
```
And the `flag` type is here:
```
/**
 * @brief Flags bitmap for esp_mesh_send() and esp_mesh_recv()
 */
#define MESH_DATA_ENC           (0x01)  /**< data encrypted (Unimplemented) */
#define MESH_DATA_P2P           (0x02)  /**< point-to-point delivery over the mesh network */
#define MESH_DATA_FROMDS        (0x04)  /**< receive from external IP network */
#define MESH_DATA_TODS          (0x08)  /**< identify this packet is target to external IP network */
#define MESH_DATA_NONBLOCK      (0x10)  /**< esp_mesh_send() non-block */
#define MESH_DATA_DROP          (0x20)  /**< in the situation of the root having been changed, identify this packet can be dropped by new root */
#define MESH_DATA_GROUP         (0x40)  /**< identify this packet is target to a group address */
```

KonssnoK, 2023-12-14:
hello @zhangyanjiaoesp .
I have a question on when the device is running in fixed root.
Sometimes the nodes of the network cannot connect to the parent, for various reasons 
( WIFI_REASON_ASSOC_TOOMANY, MESH_REASON_PARENT_IDLE, MESH_REASON_CYCLIC etc).

Now, how can i tell the device that on such errors it need to stop trying to connect to such devices?

To start the connection i use
```
ESP_ERROR_CHECK(esp_mesh_set_parent(
                &best_result.parent,
                (mesh_addr_t*)&best_result.parent_assoc.mesh_id,
                best_result.type,
                best_result.layer
            ));
```
but i don t know how to do something like `esp_mesh_reset_parent`.

Maybe something like a "disconnect" ?

KonssnoK, 2023-12-14:
for reference, esp_mesh_disconnect does not work

```
I (156025) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156027) wifi:state: init -> auth (b0)
I (156031) wifi:state: auth -> init (65c0)
I (156031) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.283) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.287) mesh_hand: Stop mesh auto-reconnect
I (156130) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156131) wifi:state: init -> auth (b0)
I (156142) wifi:state: auth -> init (65c0)
I (156143) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.393) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.395) mesh_hand: Stop mesh auto-reconnect
I (156230) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156231) wifi:state: init -> auth (b0)
I (156235) wifi:state: auth -> init (65c0)
I (156236) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.487) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.492) mesh_hand: Stop mesh auto-reconnect
I (01:02:35.556) processor: send status... (1000ms)
W (156314) wifi:Haven't to connect to a suitable AP now!
I (156332) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156334) wifi:state: init -> auth (b0)
I (156338) wifi:state: auth -> init (65c0)
I (156339) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.590) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.595) mesh_hand: Stop mesh auto-reconnect
I (156435) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156437) wifi:state: init -> auth (b0)
I (156442) wifi:state: auth -> init (65c0)
I (156443) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.693) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.698) mesh_hand: Stop mesh auto-reconnect
I (156537) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156538) wifi:state: init -> auth (b0)
I (156542) wifi:state: auth -> init (65c0)
I (156543) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.795) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.799) mesh_hand: Stop mesh auto-reconnect
I (156639) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156641) wifi:state: init -> auth (b0)
I (156645) wifi:state: auth -> init (65c0)
I (156646) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.897) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.902) mesh_hand: Stop mesh auto-reconnect
I (156738) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156739) wifi:state: init -> auth (b0)
I (156744) wifi:state: auth -> init (65c0)
I (156745) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.996) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.000) mesh_hand: Stop mesh auto-reconnect
I (156844) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156845) wifi:state: init -> auth (b0)
I (156849) wifi:state: auth -> init (65c0)
I (156850) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.102) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.106) mesh_hand: Stop mesh auto-reconnect
I (156945) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156947) wifi:state: init -> auth (b0)
I (156951) wifi:state: auth -> init (65c0)
I (156952) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.203) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.208) mesh_hand: Stop mesh auto-reconnect
I (157049) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157050) wifi:state: init -> auth (b0)
I (157054) wifi:state: auth -> init (65c0)
I (157055) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.307) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.311) mesh_hand: Stop mesh auto-reconnect
I (157151) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157153) wifi:state: init -> auth (b0)
I (157173) wifi:state: auth -> init (65c0)
I (157174) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.424) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.426) mesh_hand: Stop mesh auto-reconnect
I (157256) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157257) wifi:state: init -> auth (b0)
I (157261) wifi:state: auth -> init (65c0)
I (157262) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.514) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.518) mesh_hand: Stop mesh auto-reconnect
I (01:02:36.556) processor: send status... (1000ms)
W (157314) wifi:Haven't to connect to a suitable AP now!
I (157352) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157353) wifi:state: init -> auth (b0)
I (157358) wifi:state: auth -> init (65c0)
I (157359) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.610) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.614) mesh_hand: Stop mesh auto-reconnect
I (157459) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157460) wifi:state: init -> auth (b0)
I (157465) wifi:state: auth -> init (65c0)
I (157466) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.717) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
```

KonssnoK, 2023-12-19:
i will create an additional issue for the 2 previous comments.

KonssnoK, 2024-05-10:
hello @zhangyanjiaoesp, i still see the disconnection related to <MESH_EVENT_PARENT_DISCONNECTED>reason: 102 MESH_REASON_LEAF

more or less 500/hour over thousands of devices
[image]

and i think this happens on networks that do not have WIFI available or the WIFI password is wrong, meaning they work in LTE mode,

zhangyanjiaoesp, 2025-02-13:
@KonssnoK Do you still have this problem? Can we close it?

Alvin1Zhang, 2025-02-25:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens, thanks.

closed 2025-02-25 by Alvin1Zhang

zhangyanjiaoesp, 2025-06-05:
@KonssnoK When a node fails to select a parent, it may also call a function to kick out its own child node, with the reason code being `MESH_REASON_LEAF`.

KonssnoK, 2025-06-05:
> [@KonssnoK](https://github.com/KonssnoK) When a node fails to select a parent, it may also call a function to kick out its own child node, with the reason code being `MESH_REASON_LEAF`.

So it happens when the parent i am already connected to disconnects me because itself gets disconnected.
Can this happen if the parent is already disconnected?

A - B - C(root)
- B is trying to connect to C or is disconnected
- A tries to connect to B and gets disconnected with MESH_REASON_LEAF

Is this a possibility?