espressif/esp-idf #11988: MQTTS OTA random fail

janav1, 2023-08-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

I expect it to download firmware through MQTT successfully without connection lost.

### What is the actual behavior?

Using MQTTS to connect to any broker works fine, but sending OTA firmware through MQTT causes the connecion to break at random progress. ESP then sucesfuly reconnects and whole proces repeats. Sending messages <1kB works fine all the time.

### Steps to reproduce.

1. Step - connect ESP to any MQTT broker using TLS
2. Step - send OTA firmware to ESP

### Debug Logs.

```plain
...
MQTT_EVENT_BEFORE_CONNECT<CR><LF>
MQTT_EVENT_CONNECTED<CR><LF>
MQTT_EVENT_SUBSCRIBED<CR><LF>
...
OTA length: 1154352<CR><LF>
Writing to partition ota_1 at offset 0x1b0000, size 0x190000<CR><LF>
OTA begin succeeded<CR><LF>
MQTT Message: Total len=1154352 ,len=1003 ,offset=0<CR><LF>
OTA Writed 1003 B at 0<CR><LF>
MQTT Message: Total len=1154352 ,len=1024 ,offset=1003<CR><LF>
OTA Writed 1024 B at 1003<CR><LF>
MQTT Message: Total len=1154352 ,len=1024 ,offset=2027<CR><LF>
OTA Writed 1024 B at 2027<CR><LF>
MQTT Message: Total len=1154352 ,len=1024 ,offset=3051<CR><LF>
OTA Writed 1024 B at 3051<CR><LF>
MQTT Message: Total len=1154352 ,len=1024 ,offset=4075<CR><LF>
OTA Writed 1024 B at 4075<CR><LF>
...
MQTT Message: Total len=1154352 ,len=1024 ,offset=31723<CR><LF>
OTA Writed 1024 B at 31723<CR><LF>
W (105174) mbedtls: ssl_msg.c:4639 mbedtls_ssl_fetch_input() returned -29312 (-0x7280)<CR><LF>
W (105174) mbedtls: ssl_msg.c:3918 ssl_get_next_record() returned -29312 (-0x7280)<CR><LF>
W (105184) mbedtls: ssl_msg.c:4639 mbedtls_ssl_fetch_input() returned -76 (-0x004c)<CR><LF>
W (105184) mbedtls: ssl_msg.c:3918 ssl_get_next_record() returned -76 (-0x004c)<CR><LF>
W (105194) mbedtls: ssl_msg.c:5501 mbedtls_ssl_read_record() returned -76 (-0x004c)<CR><LF>
E (105204) esp-tls-mbedtls: read error :-0x004C:<CR><LF>
E (105204) transport_base: esp_tls_conn_read error, errno=Socket is not connected<CR><LF>
E (105214) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=128<CR><LF>
MQTT_EVENT_ERROR<CR><LF>
Free Memory: 153708, Minimum: 146420<CR><LF>
Last error from esp-tls: 0x0<CR><LF>
Last tls stack error number: 0x4c<CR><LF>
E (105234) mqtt_client: mqtt_process_receive: mqtt_message_receive() returned -1<CR><LF>
MQTT_EVENT_DISCONNECTED<CR><LF>
...
MQTT_EVENT_BEFORE_CONNECT<CR><LF>
MQTT_EVENT_CONNECTED<CR><LF>
MQTT_EVENT_SUBSCRIBED<CR><LF>
```

### More Information.

Sometimes the whole process complete successfully, but most of the time it fails at random position.

KaeLL, 2023-08-01:
Can you check if you're not getting dc'ed by the server for lack of PINGREQ? That could be the case if the OTA writes are performed by the MQTT event handler and if the update takes at least Keep-Alive seconds to finish.

euripedesrocha, 2023-08-09:
@janav1 could you provide a reproducer? 

As mentioned, you could be facing disconnection from the Broker due to lack of Keep-Alive.

euripedesrocha, 2024-01-03:
@janav1 I'm closing this one. Feel free to reopen if you didn't solve the issue. 
Provide a small reproducer and MQTT client configuration and logs.

closed 2024-01-03 by euripedesrocha