espressif/esp-idf #11993: stack overflow using SPI

SdvRebel, 2023-08-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp idf 5.0.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Wrower kit

### Power Supply used.

USB

### What is the expected behavior?

I expected to compile flash and run the script because it was already tested in ESP IDF v4.4

### What is the actual behavior?

instead when I flash the code into my ESP32 it enters in stack overflow error

### Steps to reproduce.

1. Step download gc9a01 lib from https://github.com/liyanboy74/gc9a01-esp-idf
2. Step be sure to have ESP IDF > v4.4
3. Step create a new project with visual studio code. the main file shall contain
4. `#include <stdio.h>
#include <string.h>
#include <math.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#include "gc9a01.h"

#define STACK_SIZE              2048

void LCD(void * arg)
{
    uint16_t Color;
    GC9A01_Init();
    for(;;)
    {
        Color=rand();
        GC9A01_FillRect(0,0,239,239,Color);
        GC9A01_Update();
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
[... 260 lines cut ...]
I (282) cpu_start: Compile time:     Aug  1 2023 22:34:22
I (288) cpu_start: ELF file SHA256:  d4ab9bf55dc4f94c...
I (294) cpu_start: ESP-IDF:          v5.0.3
I (299) cpu_start: Min chip rev:     v0.0
I (304) cpu_start: Max chip rev:     v3.99
I (309) cpu_start: Chip rev:         v1.0
I (314) heap_init: Initializing. RAM available for dynamic allocation:
I (321) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (327) heap_init: At 3FFCEB48 len 000114B8 (69 KiB): DRAM
I (333) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (339) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (346) heap_init: At 4008D3F8 len 00012C08 (75 KiB): IRAM
I (353) spi_flash: detected chip: generic
I (356) spi_flash: flash io: dio
I (361) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
enter into LCD INIT
I (386) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

So my question is : is there any other rule I have to follow using GPIO and SPI or driver initialization in general that I don't know or I'm not following to date?
Attached to this post I'm sending my actual working project (without the build folder)

thank you in advace for you answers.

[main_LCD.zip](https://github.com/espressif/esp-idf/files/12233880/main_LCD.zip)

ESP-Marius, 2023-08-02:
You set the stack size to `#define STACK_SIZE 2048`, which was probably very close to the limit in v4.4 and some code changes in v5.0 might have pushed it over the limit.

Try simply increasing the stack size.

SdvRebel, 2023-08-02:
Hi @ESP-Marius you are right, I have just tested with 
`
#define STACK_SIZE 4096
`
and everything works great, thankyou.

So, due to my curiosity I went on esp site and found the function
`uxTaskGetStackHighWaterMark` to analyze the STACK and It gives me that with stack size 4096 bytes the limit was 0x800 = 2048 bytes ahead.
This means that with my previous definition
`
#define STACK_SIZE 2048
`
I was exactly at the limit of my stack.

This is why today I'm simply using
`
#define STACK_SIZE 2080
`
fo my stack sizewith 0x20 from the
`uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
        printf("End of Stack is near: %p \r\n",  (void *)uxHighWaterMark);
`

thankyou.

I have just one question : why the printf in my previous trial was allowing me to avoid the stack overflow?
could someone please explain it to me?
thanks.

ESP-Marius, 2023-08-03:
> and everything works great, thankyou.

That is good to hear.

> This is why today I'm simply using
#define STACK_SIZE 2080

I'd recommend increasing it more than this (e.g. 4000) so you dont risk running into issues as soon as you change your code. You can always optimize for RAM usage later if this turns out to be an actual problems (for most apps it wont be)

> I have just one question : why the printf in my previous trial was allowing me to avoid the stack overflow?

I don't think it avoided it (printing with a stack size of 2000 will cause an overflow by itself, printf requires a lot of stack), but maybe it changed stuff around a bit so we failed to detect it. (the stackover flow checks do not 100% detect all overflows)

closed 2023-08-03 by ESP-Marius