espressif/esp-idf #11995: esp_image: Secure boot signature verification failed

Kampi, 2023-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I´m using an ESP32-C3 with ESP-IDF 5.0.2 in PlatformIO:

```
 - framework-espidf @ 3.50002.230601 (5.0.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 ```

I have two firmware versions that only differ in version number and I´m using the esp-idf OTA library.

After the build, each image is signed with the same key:

```
espsecure.py sign_data --version 2 --keyfile secure_boot_signing_key_1.pem --output 3.0.3-modified.bin 3.0.3.bin
espsecure.py v4.5.1
Padding data contents by 1680 bytes so signature sector aligns at sector boundary
1 signing key(s) found.
Signed 1179648 bytes of data from firmware.bin. Signature sector now has 1 signature blocks.

espsecure.py sign_data --version 2 --keyfile secure_boot_signing_key_1.pem --output 3.0.4-modified.bin 3.0.4.bin
espsecure.py v4.5.1
Padding data contents by 1680 bytes so signature sector aligns at sector boundary
1 signing key(s) found.
Signed 1179648 bytes of data from firmware.bin. Signature sector now has 1 signature blocks.
```

`espsecure` confirms that both firmware versions are valid:

```
espsecure.py verify_signature --version 2 --keyfile secure_boot_signing_key_1.pem 3.0.3.bin
espsecure.py v4.5.1
Signature block 0 is valid (RSA).
Signature block 0 verification successful using the supplied key (RSA).
Signature block 1 invalid. Skipping.
Signature block 2 invalid. Skipping.

espsecure.py verify_signature --version 2 --keyfile secure_boot_signing_key_1.pem 3.0.4.bin
espsecure.py v4.5.1
Signature block 0 is valid (RSA).
Signature block 0 verification successful using the supplied key (RSA).
Signature block 1 invalid. Skipping.
Signature block 2 invalid. Skipping.
```

But the OTA process fails with the following error:
```
I (8646) OTA: Starting OTA...
I (10186) HTTP_CLIENT: Body received in fetch header state, 0x3fcb6d57, 117
I (10186) esp_https_ota: Starting OTA...
I (10186) esp_https_ota: Writing to partition subtype 16 at offset 0x20000
I (10196) OTA: Running firmware version: 3.0.3
I (32396) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=2fd38h (195896) map
I (32546) esp_image: segment 1: paddr=0004fd60 vaddr=3fc95a00 size=002b8h (   696) 
I (32546) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=d7408h (881672) map
I (33216) esp_image: segment 3: paddr=00127430 vaddr=3fc95cb8 size=02c58h ( 11352) 
I (33226) esp_image: segment 4: paddr=0012a090 vaddr=40380000 size=1589ch ( 88220) 
I (33296) esp_image: segment 5: paddr=0013f934 vaddr=50000000 size=00018h (    24) 
[... 1874 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=2560
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
# CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

mahavirj, 2023-08-02:
@Kampi 

```
espsecure.py sign_data --version 2 --keyfile secure_boot_signing_key_1.pem --output 3.0.3-modified.bin 3.0.3.bin
espsecure.py v4.5.1
Padding data contents by 1680 bytes so signature sector aligns at sector boundary
```

Above log shows that the original firmware binary was not generated with secure padding enabled and hence the signing utility had to add padding bytes. Ideally, for secure boot v2 case, the unsigned firmware must be 64K aligned in size, please see documentation details [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/secure-boot-v2.html#secure-padding).

Most likely, this could be an issue with PlatformIO build system where it is not considering the `--secure-pad-v2` argument to `elf2image` utility. You might want to check this by reporting an issue at: https://github.com/platformio/platform-espressif32

Even if we ignore the secure padding part, then also ideally the verification part should have succeeded. But, I suspect an issue with secure OTA without secure boot (`CONFIG_SECURE_SIGNED_ON_UPDATE_NO_SECURE_BOOT`) and the padding (as described above) use-case here. Please try following fix once, please share **verbose** enabled logs if you still run into any issues:

```
--- a/components/bootloader_support/src/esp_image_format.c
+++ b/components/bootloader_support/src/esp_image_format.c
@@ -864,7 +864,7 @@ static esp_err_t verify_secure_boot_signature(bootloader_sha256_handle_t sha_han
         bootloader_munmap(simple_hash);
     }
 
-#if CONFIG_SECURE_BOOT_V2_ENABLED
+#if CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME || CONFIG_SECURE_SIGNED_APPS_ECDSA_V2_SCHEME
     // End of the image needs to be padded all the way to a 4KB boundary, after the simple hash
     // (for apps they are usually already padded due to --secure-pad-v2, only a problem if this option was not used.)
     uint32_t padded_end = ALIGN_UP(end, FLASH_SECTOR_SIZE);
```

Kampi, 2023-08-02:
Hi @mahavirj.

thank you. I have changed `Default log verbosity (Verbose)` and apply the change to `esp_image_format`

```
...
        bootloader_munmap(simple_hash);
    }

#if CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME || CONFIG_SECURE_SIGNED_APPS_ECDSA_V2_SCHEME
    // End of the image needs to be padded all the way to a 4KB boundary, after the simple hash
    // (for apps they are usually already padded due to --secure-pad-v2, only a problem if this option was not used.)
    uint32_t padded_end = ALIGN_UP(end, FLASH_SECTOR_SIZE);
    if (padded_end > end) {
        const void *padding = bootloader_mmap(end, padded_end - end);
        bootloader_sha256_data(sha_handle, padding, padded_end - end);
        bootloader_munmap(padding);
        end = padded_end;
    }
#endif
...
```
After that, I compiled both versions again and the update was executed successfully.

Is this the final solution or only a rough fix? Should I apply the padding after the build process manually by èxecuting?

```
esptool.py --chip ESP32-C3 elf2image firmware.elf --secure-pad-v2
```

Kampi, 2023-08-02:
Hi @mahavirj,

should I create a pull request with this fix or will you do it complete on your side?

mahavirj, 2023-08-03:
No. I have created an internal MR, this issue will be auto-closed when the fix reaches to GH.

closed 2023-08-10 by espressif-bot