espressif/esp-idf #11997: Bluedroid + release/v5.1 + -O2 = build fail

Nikhil-Bollampally, 2023-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1 stable version

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

ESP-IDF Components should build with v5.1, previously did not have any issues similar to this

### What is the actual behavior?

build fails at below step in bluedroid:

C:/Espressif/frameworks/esp-idf-v5.1/components/bt/host/bluedroid/stack/gatt/gatt_sr_hash.c:54:25: error: 'char_uuid.le
' may be used uninitialized [-Werror=maybe-uninitialized]
   54 |                     len += 7 + char_uuid.len;

### Steps to reproduce.

1. Step
2. Step
3. Step
...

caidamao, 2023-08-03:
问题存在3个月了，等修复吧

zhp0406, 2023-08-03:
[0001-bluedroid-fixed-compile-warning-when-optimize-for-pe.zip](https://github.com/espressif/esp-idf/files/12247788/0001-bluedroid-fixed-compile-warning-when-optimize-for-pe.zip)
@caidamao 目前，问题已在 master 分支上得到修复，并即将合并进 release/v5.1 分支。同时，你也可以在 release/v5.1 上手动使用该 patch 来修复问题。

zhp0406, 2023-11-07:
在 v5.1 版本中修复的 commit：62f5905be25e8f096032fb0f62dd768e7f0008de

closed 2023-11-07 by Alvin1Zhang