espressif/esp-idf #11998: LovyanGFX library and uvc_stream library both library sync   error

Nadeemranm, 2023-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idfv5.0.0

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 devkitC-1

### Power Supply used.

USB

### What is the expected behavior?

I am working a project , in this project i am include two library in this project ,but both library not sync together ,
but when i compile the single library in same project code compile susseccfully  , but not two library compile at same time why?
Error msg "main.cpp: error: designator order for field" in main.cpp file 

### What is the actual behavior?

On Wed, Aug 2, 2023 at 12:56 PM Nadeem Kumar` <[nadeemkumar.morp@gmail.com](mailto:nadeemkumar.morp@gmail.com)> wrote:
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:278:5: error: designator order for field 'uvc_config::frame_index' does not match declaration order in 'uvc_config_t' {aka 'uvc_config'}
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:261:18: warning: unused variable 'uvc_config' [-Wunused-variable]
  261 |     uvc_config_t uvc_config = {
      |                  ^~~~~~~~~~

### Steps to reproduce.

1. Step
2. Step
3. Step
...
On Wed, Aug 2, 2023 at 12:56 PM Nadeem Kumar` <[nadeemkumar.morp@gmail.com](mailto:nadeemkumar.morp@gmail.com)> wrote:
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:278:5: error: designator order for field 'uvc_config::frame_index' does not match declaration order in 'uvc_config_t' {aka 'uvc_config'}
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:261:18: warning: unused variable 'uvc_config' [-Wunused-variable]
  261 |     uvc_config_t uvc_config = {
      |                  ^~~~~~~~~~
What is the designator order for failed type error?

### Debug Logs.

```plain
On Wed, Aug 2, 2023 at 12:56 PM Nadeem Kumar` <nadeemkumar.morp@gmail.com> wrote:
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:278:5: error: designator order for field 'uvc_config::frame_index' does not match declaration order in 'uvc_config_t' {aka 'uvc_config'}
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:261:18: warning: unused variable 'uvc_config' [-Wunused-variable]
  261 |     uvc_config_t uvc_config = {
      |                  ^~~~~~~~~~
What is the designator order for failed type error?
```

### More Information.

On Wed, Aug 2, 2023 at 12:56 PM Nadeem Kumar` <[nadeemkumar.morp@gmail.com](mailto:nadeemkumar.morp@gmail.com)> wrote:
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:278:5: error: designator order for field 'uvc_config::frame_index' does not match declaration order in 'uvc_config_t' {aka 'uvc_config'}
/home/nadeem/Documents/esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_lcd_display/main/main.cpp:261:18: warning: unused variable 'uvc_config' [-Wunused-variable]
  261 |     uvc_config_t uvc_config = {
      |                  ^~~~~~~~~~
What is the designator order for failed type error?

Nadeemranm, 2023-08-05:
Pls give  replay

tore-espressif, 2023-11-21:
@Nadeemranm you have a mistake in your main.cpp file which leads to compilation error. 

> main.cpp:278:5: error: designator order for field 'uvc_config::frame_index' does not match declaration order in 'uvc_config_t' {aka 'uvc_config'}

In C++ initialization of struct members must be in the same order as in declaration

closed 2023-11-21 by tore-espressif

Nadeemranm, 2023-11-21:
Thanks for the response
You mean will have to be declared member variable same order . which order
i am using the member variable of the structure