espressif/esp-idf #11999: rst:0x1 (POWERON) with gpio_wakeup_enable and automatic light sleep

JVKran, 2023-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4-386-g9b75e5664e

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board with ESP32-S3-M0N16R8

### Power Supply used.

Battery

### What is the expected behavior?

I expect the program to print ```GPIO[11] intr, val: 1``` whenever the button is pressed and ```GPIO[11] intr, val: 0``` when it is released. It should do so after waking up from automatic light sleep and having the ISR write the pin number into a message queue that's read from a seperate task.

### What is the actual behavior?

It resets. See debug output below.

### Steps to reproduce.

Make sure to compile for the ESP32-S3 and use the attached main.c and sdkconfig.

```#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"
#include "esp_pm.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "esp_intr_alloc.h"
#include "driver/rtc_io.h"

// LED pin info
#define LED_PIN     GPIO_NUM_42
#define LED_PIN_MSK     (1ULL<<LED_PIN)
// Button pin info
#define BUTTON_PIN     GPIO_NUM_11
#define BUTTON_PIN_MSK  (1ULL<<BUTTON_PIN)
// Interrupt flags
#define ESP_INTR_FLAG_DEFAULT 0

static QueueHandle_t gpio_evt_queue = NULL;

static void IRAM_ATTR gpio_isr_handler(void* arg) {
  uint32_t gpio_num = (uint32_t) arg;
  int in_level = gpio_get_level(gpio_num);
  if (in_level == 0) {
[... 150 lines cut ...]
I (199) cpu_start: Project name:     test
I (204) cpu_start: App version:      e02937d-dirty
I (209) cpu_start: Compile time:     Aug  2 2023 13:35:46
I (215) cpu_start: ELF file SHA256:  14d737470223000f...
I (221) cpu_start: ESP-IDF:          v4.4-386-g9b75e5664e
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 3FC964F0 len 00049B10 (294 KiB): D/IRAM
I (241) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (248) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (254) heap_init: At 600FE020 len 00001FE0 (7 KiB): RTCRAM
I (260) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (268) spi_flash: detected chip: gd
I (272) spi_flash: flash io: qio
I (287) esp_core_dump_uart: Init core dump to UART
I (288) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (292) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (302) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:5
```

### More Information.

We're using it on a custom board with a series resistor and capacitor to ground. The button works flawlessly, but not with light sleep enabled. I've stumbled upon [this issue](https://github.com/espressif/esp-idf/issues/8887) that seemed to work, but now it doesn't.

At least I've disabled ```CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND``` altough I'm not really sure if I want to? What are the consequences of disabling it btw?

animevietsub, 2023-08-02:
Make sure you have enabled `"Support for power management" `in **Power Management** and you have enabled `"Tickless idle support"` in **FreeRTOS**.

esp-wzh, 2023-08-03:
@JVKran  Can you have a try on latest v4.4 release? I think this issue has been fixed in https://github.com/espressif/esp-idf/commit/a75a62431a9dab8bffd586ccfacbb5c58c682020.

It is recommended to enable [CONFIG_ ESP_SLEEP_GPIO_RESET_WORKAROUND](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-esp-sleep-gpio-reset-workaround) in your project if you are not filtering the electrostatic pulse on **al**l GPIOs with external circuits. When this option is enabled, IDF will configures non-wakeup pins to be in isolated state (input/output disable and in floating state) during sleep, you can also customize the IOs that do not apply isolation using `gpio_sleep_sel_dis`.

JVKran, 2023-08-07:
Hi @esp-wzh, updating to the latest V4.4 release did the trick. This however, now turned into another issue. When I press the button while the device is booting, the following happens...

```
x��ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375bf0
0x40375bf0: esp_restart_noos at C:/Users/joche/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x127c
load:0x403c9700,len:0xc8c
load:0x403cc700,len:0x2fe8
entry 0x403c9904
I (66) opi psram: vendor id : 0x0d (AP)
I (66) opi psram: dev id    : 0x02 (generation 3)
I (67) opi psram: density   : 0x03 (64 Mbit)
I (70) opi psram: good-die  : 0x01 (Pass)
I (75) opi psram: Latency   : 0x01 (Fixed)
I (84) opi psram: SRF       : 0x01 (Fast Refresh)
I (89) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (94) opi psram: BurstLen  : 0x01 (32 Byte)
I (99) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (105) opi psram: DriveStrength: 0x00 (1/1)
I (110) spiram: Found 64MBit SPI RAM device
I (115) spiram: SPI RAM mode: sram 40m
I (120) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (127) cpu_start: Pro cpu up.
I (130) cpu_start: Starting app cpu, entry point is 0x4037550c
0x4037550c: call_start_cpu1 at C:/Users/joche/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (115) cpu_start: App cpu up.
I (201) cpu_start: Pro cpu start user code
I (201) cpu_start: cpu freq: 240000000
I (201) cpu_start: Application information:
I (203) cpu_start: Project name:     test
I (208) cpu_start: App version:      f2e1795-dirty
I (213) cpu_start: Compile time:     Aug  7 2023 11:41:43
I (220) cpu_start: ELF file SHA256:  36ca3785de0d2dc4...
I (226) cpu_start: ESP-IDF:          v4.4.5-249-gcc71308e2f-dirty
I (232) cpu_start: Min chip rev:     v0.0
I (237) cpu_start: Max chip rev:     v0.99
I (242) cpu_start: Chip rev:         v0.1
I (247) heap_init: Initializing. RAM available for dynamic allocation:
I (254) heap_init: At 3FC97D10 len 00051A00 (326 KiB): D/IRAM
I (260) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (267) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (273) heap_init: At 600FE024 len 00001FDC (7 KiB): RTCRAM
I (279) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (288) spi_flash: detected chip: gd
I (292) spi_flash: flash io: qio
I (296) sleep: Configure to isolate all GPIO pins in sleep state
I (302) sleep: Enable automatic switching of GPIO sleep configuration
I (320) esp_core_dump_uart: Init core dump to UART
I (321) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (335) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:5
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40377058  PS      : 0x00050034  A0      : 0x3fcf0e50  A1      : 0x3fc94210
0x40377058: _xt_lowint1 at C:/Users/joche/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114

A2      : 0x00002200  A3      : 0x8037ec9d  A4      : 0x000637ff  A5      : 0x3fc941f0
A6      : 0x00000000  A7      : 0x3fc94b78  A8      : 0x803761f8  A9      : 0x4037bf86
0x4037bf86: _frxt_int_enter at C:/Users/joche/esp/esp-idf/components/freertos/port/xtensa/portasm.S:119

A10     : 0x3fcea4e8  A11     : 0x00000804  A12     : 0x60004000  A13     : 0x3fc941d0
[... 181 lines cut ...]
.iram0.bss 0x403833e0 0x0 RW
.dram0.heap_start 0x3fc97d10 0x0 RW
.coredump.tasks.data 0x3fcf115c 0x158 RW
.coredump.tasks.data 0x3fcf0e50 0x300 RW
.coredump.tasks.data 0x3fcf406c 0x158 RW
.coredump.tasks.data 0x3fcf3e60 0x200 RW
.coredump.tasks.data 0x3fcf47cc 0x158 RW
.coredump.tasks.data 0x3fcf4560 0x260 RW
.coredump.tasks.data 0x3fcf2924 0x158 RW
.coredump.tasks.data 0x3fcf2690 0x280 RW
.coredump.tasks.data 0x3fcf19c4 0x158 RW
.coredump.tasks.data 0x3fcf1750 0x260 RW
.coredump.tasks.data 0x3fcea37c 0x158 RW
.coredump.tasks.data 0x3fcea100 0x270 RW
.coredump.tasks.data 0x3fcf390c 0x158 RW
.coredump.tasks.data 0x3fcf35a0 0x360 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='faa12739'
I (3021) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...
x��ESP-ROM:esp32s3-20210327
```

esp-wzh, 2023-08-07:
@JVKran  Because you are using level type interrupt: `.intr_type = GPIO_INTR_HIGH_LEVEL`. Since the level cannot be read and returned to low, it is blocked in `gpio_isr_loop` in isr, I think set `.intr_type = GPIO_INTR_ANYEDGE`  should be OK.

JVKran, 2023-08-07:
Okay, well, that's stupid of me. Thanks. 

If you don't mind, I have one more question. After I've updated to release/v4.4 I'm seeing a higher current consumption by around 75uA more in deep-sleep. Are there any things that have changed? I'm using WiFi, Bluetooth, GPIO and ADC in my application.

esp-wzh, 2023-08-08:
@JVKran  In our release process. Will cover power consumption testing in some application scenarios, but still in the process of refinement. The tests will at least ensure that the power consumption of this [example](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/system/deep_sleep_wake_stub) power consume is as expected, you can compare your application with this example to locate the module that is causing the power consumption anomaly.

closed 2023-11-15 by Alvin1Zhang