espressif/esp-idf #12001: language standard

podaen, 2023-08-02:
### Is your feature request related to a problem?

Just found that the Arduino core is running now on the IDF component v4.4.5, I checked the documentation but I can't find anything about the gnu it is using for C and C++

Finally there is a written a document about the c++ code that is in use , but they forgot to tell the C version for the latest IDF 5.2...

### Describe the solution you'd like.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/c.html

### Describe alternatives you've considered.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html

0xjakob, 2023-08-03:
@podaen You are right, the C standard isn't really mentioned anywhere. We will find a place in the documentation to mention the used C standard. Meanwhile, you can check it in [build.cmake](https://github.com/espressif/esp-idf/blob/master/tools/cmake/build.cmake#L134).  `gnu17` basically means the ISO C17 standard with gnu extensions.

podaen, 2023-08-03:
Hi,

Thanks to acknowledging it. It will be a huge step forward in programming with the IDF component and will lead in less miss understandings in problems that do or do not occur based on the language that's in use. I will wait to release specific documentation for the latest in Arduino core in a related comments about that topic until it is final.

Right now the Arduino core uses v4.4.5. If I understand it right it uses C++23 and C17 by default. Knowing that I could start to adapted locally in advance. 

Kr
Dave

closed 2023-11-20 by espressif-bot