espressif/esp-idf #12002: req->aux == null

baldhead69, 2023-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am having some issues with my websocket secure server running in esp32-s3.

When i sent a websocket message to client synchronously, the req->aux sometimes is null and sometimes in not null.

In httpd_ws.c file:
```
esp_err_t httpd_ws_send_frame(httpd_req_t *req, httpd_ws_frame_t *frame)
{
    ESP_LOGI(TAG, "[%s] req: %p\n", __func__, req );
    ESP_LOGI(TAG, "[%s] req->aux: %p\n", __func__, req->aux );
    
    esp_err_t ret = httpd_ws_check_req(req);
    if (ret != ESP_OK) {
        return ret;
    }
    return httpd_ws_send_frame_async(req->handle, httpd_req_to_sockfd(req), frame);
}
```

```
I (110924) httpd_ws: [httpd_ws_send_frame] req: 0x3fcc2c7c
I (110927) httpd_ws: [httpd_ws_send_frame] req->aux: 0x0

W (110934) httpd_ws: httpd_ws_check_req: Argument is null
```

What can be the causes for req->aux to be null in httpd_ws.c file ?

Stack corruption ?
Do i need to increase the httpd task stack ?

baldhead69, 2023-08-04:
I changed compiler optimization performance level -O2 to debug -O0 and at runtime i get:
"A stack overflow in task sys_evt has been detected".

I removed 2 local buffers with 128 bytes each(consumed 256 bytes task stack) from a ota callback function "ota_event_handler" and the stack overflow still occurs.
CODE: [SELECT ALL](https://www.esp32.com/viewtopic.php?f=13&t=34999&p=118011#)

ESP_ERROR_CHECK(esp_event_handler_register(ESP_HTTPS_OTA_EVENT, ESP_EVENT_ANY_ID, &ota_event_handler, NULL));
then i increased the stack from 2304 to 3072 bytes in the task sys_evt and the stack overflow it disappeared(worked).

I thought i found the problem of req->aux = null, but doesn't, the problem still persists.

baldhead69, 2023-08-04:
Maybe i cant send a message from "Event loop task" to http task ?

wss_server.c file:
```
void resp_Ws_With_Message_Sync( httpd_req_t *req, char* message, uint8_t command )
{
    ESP_LOGI(TAG, LOG_USER("FUNCTION: %s"), __func__ ); 
    ESP_LOGI(TAG, LOG_USER("Message: %s\n"), message);

    ESP_LOGI(TAG, LOG_USER("[%s] req: %p"), __func__, req );
    ESP_LOGI(TAG, LOG_USER("[%s] req->aux: %p\n"), __func__, req->aux );

    size_t strSize = strlen(message);
    init_ws_pkt( 1 + strSize );  // command size + error message size 
    ws_pkt.payload[0] = command;       
    
    for( uint32_t i = 0 ; i < strSize ; i++ )
    {
        ws_pkt.payload[i+1] = message[i];  // add a message after the command.
    }

    httpd_ws_send_frame( req, &ws_pkt );
}
```

my_ota.c file:
```
static httpd_req_t* http_req = NULL;

////////////////////////////////////////////////////////////////////////////////////////////
// Send eventhandler messages to phone app and lcd display.

static void send_ota_message( char* message )
{
    ESP_LOGI(TAG, LOG_USER("FUNCTION: %s"), __func__ ); 
    ESP_LOGI(TAG, LOG_USER("[%s] http_req: %p\n"), __func__, http_req );

    if( http_req != NULL )
    { 
        resp_Ws_With_Message_Sync( http_req, message, req_command );

        // send message to lcd display
    }
    else
    {
        // send message to lcd display
    }
}

////////////////////////////////////////////////////////////////////////////////////////////
// Event handler for catching ota system events

static void ota_event_handler(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data)
{
    ESP_LOGI(TAG, LOG_USER("FUNCTION: %s"), __func__ ); 
    
    if( event_base == ESP_HTTPS_OTA_EVENT )
    {
        switch( event_id )
        {
            case ESP_HTTPS_OTA_START:
            {                
                ESP_LOGI(TAG, TEXT_OTA_START);
                send_ota_message( TEXT_OTA_START );  // calback need to return quickly, try create a task to send the message to websocket client. Send message from this task to new task using queue.               
                break;
            }

            case ESP_HTTPS_OTA_CONNECTED:
            {                
                ESP_LOGI(TAG, TEXT_OTA_CONNECTED);
                send_ota_message( TEXT_OTA_CONNECTED ); 
                break;
            }
            
            case ESP_HTTPS_OTA_GET_IMG_DESC:
            {               
                ESP_LOGI(TAG, TEXT_OTA_GET_IMG_DESC);
                send_ota_message( TEXT_OTA_GET_IMG_DESC ); 
                break;
            }

            case ESP_HTTPS_OTA_VERIFY_CHIP_ID:
            {               
                ESP_LOGI(TAG, TEXT_OTA_VERIFY_CHIP_ID"%d", *(esp_chip_id_t *)event_data);
                                
                char* str = NULL; 
                if( concat_text_int( TEXT_OTA_VERIFY_CHIP_ID, *(esp_chip_id_t *)event_data, &str ) != ESP_OK ) return;
                send_ota_message( str );                
                free(str);

                break;
            }

            case ESP_HTTPS_OTA_DECRYPT_CB:
            {          
                ESP_LOGI(TAG, TEXT_OTA_DECRYPT_CB);
                send_ota_message( TEXT_OTA_DECRYPT_CB );            
                break;
            }

            case ESP_HTTPS_OTA_WRITE_FLASH:
            {            
                ESP_LOGI(TAG, LOG_USER1(TEXT_OTA_WRITE_FLASH"%d"), *(int *)event_data);
                                
                char* str = NULL; 
                if( concat_text_int( TEXT_OTA_WRITE_FLASH, *(int *)event_data, &str ) != ESP_OK ) return;
                send_ota_message( str );                
                free(str);        

                break;
            }

            case ESP_HTTPS_OTA_UPDATE_BOOT_PARTITION:
            {                               
                ESP_LOGI(TAG, TEXT_OTA_UPDATE_BOOT_PARTITION"%d", *(esp_partition_subtype_t *)event_data);  
                
                char* str = NULL; 
                if( concat_text_int( TEXT_OTA_UPDATE_BOOT_PARTITION, *(esp_partition_subtype_t *)event_data, &str ) != ESP_OK ) return;
                send_ota_message( str );                
                free(str); 

                break;
            }

            case ESP_HTTPS_OTA_FINISH:
            {                
                ESP_LOGI(TAG, TEXT_OTA_FINISH);
                send_ota_message( TEXT_OTA_FINISH );                                                
                http_req = NULL;                
                break;
            }

            case ESP_HTTPS_OTA_ABORT:
            {
                ESP_LOGI(TAG, TEXT_OTA_ABORT);
                send_ota_message( TEXT_OTA_ABORT );                
                http_req = NULL;                
                break;
            }
        }
    }
}
```

baldhead69, 2023-08-08:
some suggestion ?

ESP-YJM, 2023-08-09:
@baldhead69 It seems `http_req ` has freed in `esp_http_server` and you still use it. So `req->aux` is NULL. Where do your `http_req` set?

baldhead69, 2023-08-09:
> @baldhead69 It seems `http_req ` has freed in `esp_http_server` and you still use it. So `req->aux` is NULL. Where do your `http_req` set?

No because req is not null.

http_req is passed by http uri handler.

https://www.esp32.com/viewtopic.php?f=13&t=34999&sid=b94760bbf6fe41e9f04beeb69fd76d0a

ESP-YJM, 2023-08-09:
@baldhead69 Yes, `http_req` is not NULL, but `req->aux` it is freed by internal. You can add address print in `esp_http_server` where `req->aux` set 0 or NULL.

baldhead69, 2023-08-10:
> @baldhead69 Yes, `http_req` is not NULL, but `req->aux` it is freed by internal. You can add address print in `esp_http_server` where `req->aux` set 0 or NULL.

Where can i put this ?

file, line ?

Here:
```
httpd_main. c
static void httpd_delete(struct httpd_data *hd)
```

Thank's.

ESP-YJM, 2023-08-11:
@baldhead69 You can add print here https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_parse.c#L724C19-L724C19 to check aux whether has set NULL here.

baldhead69, 2023-08-11:
> @baldhead69 You can add print here https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_parse.c#L724C19-L724C19 to check aux whether has set NULL here.

@ESP-YJM ,

With your suggestion i found the problem.

The req->aux is set to null every time the uri handler returns. 
This is not persistent until the "close socket", it is valid only per "client message reception".

This way i can't use "httpd_ws_send_frame( req, &ws_pkt );" that is "sync".

I have to use "httpd_ws_send_frame_async(server, sockfd, &ws_pkt);", that is async .

Now i understand why there is send message sync and send message async in espressif websocket server.

Thank's.

ESP-YJM, 2023-08-14:
@baldhead69 Yes, thanks for sharing your found.

closed 2023-08-14 by Alvin1Zhang