espressif/esp-idf #12003: An assertion occurring in esp_bignum.

imammy-hacomono, 2023-08-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

No assertion occurs.

### What is the actual behavior?

 am performing encrypted MQTT communication using TLS with the (esp-aws-iot library)[https://github.com/espressif/esp-aws-iot/tree/202210.01-LTS-release], coreMQTT, and the W5500 Ethernet chip. Approximately every one minute, I send JSON data of a few tens of bytes to the MQTT broker. (The data is encrypted by TLS and may become slightly larger.)

After leaving the system idle for about one day, I observed the following log indicating an assertion:

```
assert failed: mpi_mult_mpi_failover_mod_mult esp_bignum.c:651 (mpi_words(Z) == z_words)

Backtrace: 0x40081b86:0x3ffcbaa0 0x40087db5:0x3ffcbac0 0x4008e409:0x3ffcbae0 0x40127119:0x3ffcbc00 0x401271d9:0x3ffcbc30 0x4011d354:0x3ffcbc60 0x4011d3d1:0x3ffcbcf0 0x40125665:0x3ffcbd10 0x40125cb3:0x3ffcbdd0 0x40125d53:0x3ffcbdf0 0x40123daf:0x3ffcbe20 0x40123bc9:0x3ffcbe50 0x4011ad9d:0x3ffcbe80 0x40118c96:0x3ffcbf20 0x40118cf5:0x3ffcbf40 0x401141c1:0x3ffcbf60 0x40113f0d:0x3ffcbf80 0x40113fe3:0x3ffcbfc0 0x400eb5ca:0x3ffcc000 0x400e491a:0x3ffcc080 0x400e4139:0x3ffcc0f0
0x40081b86: panic_abort at esp/esp-idf/components/esp_system/panic.c:452

0x40087db5: esp_system_abort at /Users/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4008e409: __assert_func at /Users/esp/esp-idf/components/newlib/assert.c:81

0x40127119: mpi_mult_mpi_failover_mod_mult at /Users/esp/esp-idf/components/mbedtls/port/esp_bignum.c:651
 (inlined by) mbedtls_mpi_mul_mpi at /Users/esp/esp-idf/components/mbedtls/port/esp_bignum.c:522

0x401271d9: mbedtls_mpi_mul_int at /Users/esp/esp-idf/components/mbedtls/port/esp_bignum.c:557

0x4011d354: mbedtls_mpi_div_mpi at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1430

0x4011d3d1: mbedtls_mpi_mod_mpi at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1499

0x40125665: rsa_prepare_blinding at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:807
 (inlined by) mbedtls_rsa_private at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:950

0x40125cb3: mbedtls_rsa_rsassa_pkcs1_v15_sign at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:1981

0x40125d53: mbedtls_rsa_pkcs1_sign at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2022

0x40123daf: rsa_sign_wrap at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:393

0x40123bc9: mbedtls_pk_sign_restartable at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:648

0x4011ad9d: ssl_write_certificate_verify at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3335
 (inlined by) mbedtls_ssl_handshake_client_step at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3553

0x40118c96: mbedtls_ssl_handshake_step at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3880
 (inlined by) mbedtls_ssl_handshake_step at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3832

0x40118cf5: mbedtls_ssl_handshake at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3943
 (inlined by) mbedtls_ssl_handshake at /Users/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3920

0x401141c1: esp_mbedtls_handshake at /Users/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:192

0x40113f0d: esp_tls_handshake at /Users/esp/esp-idf/components/esp-tls/esp_tls.c:114 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at /Users/esp/esp-idf/components/esp-tls/esp_tls.c:493 (discriminator 15)

0x40113fe3: esp_tls_conn_new_sync at /Users/esp/esp-idf/components/esp-tls/esp_tls.c:522

0x400eb5ca: xTlsConnect at /Users/workspace/test-application/components/esp-aws-iot/libraries/coreMQTT/port/network_transport/network_transport.c:31

```

This issue seems similar to a past Issue (https://github.com/espressif/esp-idf/issues/10403).

### Steps to reproduce.

1. Step
2. Step
3. Step
...

imammy-hacomono, 2023-08-03:
Is it incorrect that this configuration is enabled?
[image]

AxelLin, 2023-08-03:
https://github.com/espressif/esp-idf/issues/11850
This should be fixed by 8e25eb1e08dcb5a346248451ea0b75d64295ddfe

imammy-hacomono, 2023-08-03:
Hi @AxelLin 
You responded to my issue, and I want to say thank you.

If it has not been merged into the release branch yet?
Is the workaround to disable Hardware MPI Acceleration?

mahavirj, 2023-08-04:
@imammy-hacomono Yes the fix has been merged in the release/v5.1 branch: https://github.com/espressif/esp-idf/commit/8e25eb1e08dcb5a346248451ea0b75d64295ddfe. Either you may cherry-pick the fix or wait for the next bugfix release v5.1.1. Hope this helps!

imammy-hacomono, 2023-08-04:
@mahavirj 
Understood!
 I will switch to using IDF "release/v5.1" and give it a try!
 However, I find it a bit difficult to understand why the fix is not included in "tag v5.1."

mahavirj, 2023-08-04:
>  However, I find it a bit difficult to understand why the fix is not included in "tag v5.1."

The issue was discovered after v5.1 release (tag) and hence it will be part of the next bugfix release (v5.1.1). I will close this issue, please feel free to comment if you still face any issue with latest `release/v5.1` branch.

closed 2023-08-04 by mahavirj

imammy-hacomono, 2023-08-04:
@mahavirj 
Thank you.
Is it okay to keep Hardware MPI Acceleration enabled?

mahavirj, 2023-08-04:
>  Is it okay to keep Hardware MPI Acceleration enabled?

Yes, provided you have the fix mentioned above.