espressif/esp-idf #12007: ld_sco_frm_isr_wrapper  ?

zhou88mang, 2023-08-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.02

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitCV4

### Power Supply used.

USB

### What is the expected behavior?

Run hfp_ag demo, excepted behavior  : It runs normally and does not restart.

### What is the actual behavior?

Sometimes it reports errors and restarts , 
and error : 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008588c  PS      : 0x00060334  A0      : 0x800867b0  A1      : 0x3ffc1da8  
0x4008588c: ld_sco_frm_isr_wrapper at ??:?

A2      : 0x00017337  A3      : 0x3ffe9a78  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffb8d40  A7      : 0x00000000  A8      : 0x3ffb8258  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x3ffd1654  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000b3  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x40085889:0x3ffc1da8 |<-CORRUPTED
0x40085889: ld_sco_frm_isr_wrapper at ??:?

### Steps to reproduce.

1. run hfp_ag demo
2. connect my earphone
3. Establishment of sco connection
...

### Debug Logs.

```plain
W (67338) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67344) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67352) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67356) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67362) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67368) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67374) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67380) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67386) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67392) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67400) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67404) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67410) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67416) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67422) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67428) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67434) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67440) BT_APPL: bta_ag_sco_read_cback: status(2)
W (67446) BT_APPL: bta_ag_sco_read_cback: status(2)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008588c  PS      : 0x00060334  A0      : 0x800867b0  A1      : 0x3ffc1da8  
0x4008588c: ld_sco_frm_isr_wrapper at ??:?

A2      : 0x00017337  A3      : 0x3ffe9a78  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffb8d40  A7      : 0x00000000  A8      : 0x3ffb8258  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x3ffd1654  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000b3  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x40085889:0x3ffc1da8 |<-CORRUPTED
0x40085889: ld_sco_frm_isr_wrapper at ??:?
```

shangke1112, 2023-08-04:
Please upload the elf file with the error ,and tell me the commit id of the IDF .

shangke1112, 2023-08-06:
[libbtdm.tar.gz](https://github.com/espressif/esp-idf/files/12268789/libbtdm.tar.gz)
Please use this lib and try again . If there is any error, let me know .

shangke1112, 2023-08-13:
Any error ?

Indastri, 2024-11-25:
Since this issue was resolved on our side and there was no response from the user -> closing. If you encounter any further issues please reopen or create a new issue.

Cheers!

closed 2024-11-25 by Indastri