espressif/esp-idf #12008: How to enable Secure Boot v2 AND Flash Encryption (+ NVS encryption) on the ESP32-S3 with user-generated keys?

robokrishan, 2023-08-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to figure out how to enable both security features on my ESP32-S3 (N8R8). I bought 5x boards, but I have burned through 4 of them. I have ONE fresh board left which I don't want to brick.

## Chip

ESP32-S3 (N8R8)

## IDF Version

IDF 5.1 (stable)

## Host Machine

Ubuntu 22.04 + vsCode

## Problem

I have successfully tested both features independently on separate boards. Flash encryption works when I have just flash encryption enabled and configured. Secure Boot V2 works when I have just secure boot v2 enabled and configured. When I try to enable both on a single board, the device bricks because anything I flash no longer boots.

## Procedure followed for Flash Encryption

1. We burn the XTS-AES key into the ESP32's eFUSE. Run the following command to burn the flash encryption key to the eFUSE:

        $ espefuse.py burn_key --show-sensitive-info BLOCK_KEY0 ./encryption/flash_encryption_key.bin XTS_AES_128_KEY

    - The `--show-sensitive-info` option will make the burned key human-readable in the prompt that follows.
    - The key will be burned to `BLOCK_KEY0` which is `BLOCK4` of the eFUSE.
    - The flash encryption key is in `XTS_AES_128` format.
    - Type `BURN` in all capitals and press enter to continue.

2. Now flash encryption must be enabled in the project configuration. Open `menuconfig` and navigate to `Security Features`. Check the following box:

        [ ] Enable flash encryption on boot (READ DOCS FIRST)

    - When you enable this option, additional options will show up

3. Under `Security Features`, set the `Enable usage mode` to `Development (NOT SECURE)`

4. Save and exit `menuconfig`.

5. Use the following command to flash the encrypted version of just the app image:

        $ idf.py encrypted-app-flash monitor

6. Use the following command to flash all data in encrypted format:

        $ idf.py encrypted-flash monitor
        
When I follow the instructions above, the flash encryption is configured upon first boot. I added the `encrypted` keyword to 3 extra partitions in the partition table i.e. partitions `3`, `5` and `8`. The partition encryption can be seen in the logs below:

```
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 otadata          OTA data         01 00 00011000 00002000
I (77) boot:  1 phy_init         RF data          01 01 00013000 00001000
I (84) boot:  2 coredump         Unknown data     01 03 00014000 00010000
I (92) boot:  3 nvs_keys         NVS keys         01 04 00024000 00001000
I (99) boot:  4 nvs              WiFi data        01 02 00025000 000d0000
I (106) boot:  5 storage          Unknown data     01 ff 000f5000 00001000
I (114) boot:  6 ota_0            OTA app          00 10 00100000 00200000
I (122) boot:  7 ota_1            OTA app          00 11 00300000 00200000
I (129) boot:  8 logs             WiFi data        01 02 00500000 00100000
I (137) boot:  9 files            Unknown data     01 ff 00600000 00200000
[... 123 lines cut ...]
14. If the previous command executed without problems, run the following command to flash the signed bootloader:
    
        $ idf.py bootloader-flash

    - If this command fails, then the bootloader needs to be force-flashed. 
    - Scroll to the bottom of the failure message and copy the failed command. Paste it into the terminal.
    - Navigate to the function call `write_flash` and type `;--force;` as demonstrated in the following point.
    - Here is the command for reference. It will vary from system to system:
```
        $ cd /home/xxxxx/esp/esp-idf/components/bootloader && /usr/bin/cmake -D IDF_PATH=/home/xxxxx/esp/esp-idf -D "SERIAL_TOOL=/home/xxxxx/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/home/xxxxx/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=no_reset;--no-stub;write_flash;--force;@bootloader-flash_args" -D WORKING_DIRECTORY=/home/xxxxx/esp/project_dir/build -P /home/xxxxx/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
```

15. Since we enabled binary signing in `menuconfig`, we can simply proceed as usual to build application binaries using the following command:
        $ idf.py build

    - You will be able to see in the build logs if a signature was found and if the compiled binary is signed or not.

16. If the previous command executed without problems, run the following command to flash the signed application binary:
        $ idf.py flash monitor
    
**After following the above steps, I am able to build and flash signed binaries to the chip. If I flash unsigned binaries, the image fails to boot which is expected. I don't have the logs to show anymore (because I bricked it trying to also enable flash encryption).**

 I would appreciate it if someone could point out any mistakes in my procedure. I have written the whole procedure myself after following Espressif's documentation for IDF 5.1 (stable) and ESP32-S3. The documentation does not explain how to enable both features together. I know the procedure is not perfect so please let me know the things I missed. I want to know the correct order of steps from the procedures above so that I can have both features enabled on my board.
 
 TIA

mahavirj, 2023-08-04:
@robokrishan 

I see some discrepancy in the configuration that you have specified and the logs that you shared above.

>  Under Security Features, set the Enable usage mode to Development (NOT SECURE)

You mentioned that you enabled the Flash Enc feature in development mode but the log says that the feature was enabled in the release mode.

> I (40657) flash_encrypt: Setting CRYPT_CNT for permanent encryption

Please check if you are flashing the bootloader generated with right configuration case here.

To enable both secure boot and flash enc together, you can probably enable all relevant configurations in one go under menuconfig and then flash the artifacts as highlighted during the build process (separate commands for application and bootloader). After this on the first boot-up the bootloader shall enable both features, program efuse configurations, encrypt images and then proceed for normal boot-up.

If you would like to enable both secure boot and flash enc with external host based assistance (and not through bootloader) then please follow guide available here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/host-based-security-workflows.html#enable-flash-encryption-and-secure-boot-v2-externally

If you would like to experiment on the different workflows then I suggest evaluating this [ESP32 Qemu guide](https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/esp32/README.md) with ESP32 qemu release available [here](https://github.com/espressif/qemu/releases). Please note that there is no support for S3 target here but the security workflows are more or less between ESP32 and S3.

Hope this helps!

robokrishan, 2023-08-08:
@mahavirj Thanks for the link, that's what I was looking for ;)

closed 2023-08-08 by robokrishan

PerSmitt, 2024-05-01:
I saw the issue was closed but i found one issue in the list of instructions. For secure boot step 10 the public key should be used, not the private key. This does give the same result as the tool is smart enough to see that its a private or public key. Still, the private key should never be accessible for this tool once the product is put in production.

mahavirj, 2024-05-08:
@PerSmitt 

There are 2 ways to sign the artifacts for secure boot case:

1. Through IDF build system by supplying the secure signing private key
2. Managing the signing part outside of the IDF build system (e.g., remote signing case)

Based on the selected approach, private key may or may not be accessible to the IDF build system (or tools).

> This does give the same result as the tool is smart enough to see that its a private or public key.

Right. The command to generate the digest of the public key works by specifying either public or private key. We will add more clarification regarding this in our docs.