espressif/esp-idf #12009: ESP_LOGx argument `format` cannot be const char*

amrsoll, 2023-08-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The `format` variable in `ESP_LOGx(tag, format, ...)` cannot be used as a runtime variable.

I expect `ESP_LOGx` to behave similarly `printf()`, at least when it comes to the `format` argument, because the documentation for [`ESP_LOG_LEVEL`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#c.ESP_LOG_LEVEL) refers to `printf()`

[image]

### What is the actual behavior?

Above fails because of the macro 

```c
#define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
```

The macro does not allow for `format` to be a variable as it is concatenated at compile time.

Similar issue : https://github.com/espressif/esp-idf/issues/3069 (The OP closed the issue because he found a workaround, not a fix)

### Steps to reproduce.

```c
#include "esp_log.h"

const char* TAG = __FILE__;
const char* format = "%s";

void app_main() {
    ESP_LOGI(TAG, format, __func__);
}
```

### More Information.

I understand the concatenation is necessary to keep the functionality of the format that is injected to `vprintf` later. 

### Use Case

I would like to forward the format through multiple functions that can log the function caller

```c
// test_file.h
#define str(x) #x
#define xstr(x) str(x)

void _my_logging_func(const char *format, const char *func_name);

#define my_logging_func() _my_logging_func("Function called from : %s "__FILE__":"xstr(__LINE__), __func__)

// test_file.c
#include "esp_log.h"
#define TAG "TEST"

void _my_logging_func(const char *format, const char *func_name) {
    ESP_LOGW( TAG, _format, func_name);
}
```

### Suggestion

It could be possible to make multiple calls to achieve the same result, although with a risk of getting overlapping asynchronous print statements.

[`components/log/include/esp_log.h`](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/components/log/include/esp_log.h`)

**Before**

```c
#define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"

#define ESP_LOG_LEVEL(level, tag, format, ...) do {   \
    if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR, tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \

    else if (level==ESP_LOG_WARN ) { ... } \
    else if (level==ESP_LOG_DEBUG) { ... } \
    else if (level==ESP_LOG_VERBOSE) { ... } \
    else  { ... } \
} while(0)
```

**After**

```c
#define LOG_FORMAT(letter)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: "

#define ESP_LOG_LEVEL(level, tag, format, ...) do {   \
    if (level==ESP_LOG_ERROR )   { esp_log_write(ESP_LOG_ERROR, tag, LOG_FORMAT(E), esp_log_timestamp(), tag); } \
    else if (level==ESP_LOG_WARN ) { ... } \
    else if (level==ESP_LOG_DEBUG) { ... } \
    else if (level==ESP_LOG_VERBOSE) { ... } \
    else  { ... } \
    esp_log_write(format, ##__VA_ARGS__); \
    esp_log_write(LOG_RESET_COLOR "\n"); \
} while(0)
```

amrsoll, 2023-08-03:
@igrr Yep it was simplified

My intention was to define various macros that have different string `format`s. To avoid repetition, I would prefer to pass the `format` as a dependency injection.

So imagine multiple 

```c
#define my_macro() _my_logging_func("some %s format", "other")
```

Now the only reason that I categorise it as 'bug' is because the usage differs from `printf`, otherwise you can consider it a feature request :)

igrr, 2023-08-03:
Thanks, that makes it a bit more clear. Could you show how you imagine `_my_logging_func` _function_ would be implemented in this case? You can use `printf` instead of ESP_LOG just to demonstrate the idea.

The reason I'm asking, I'm not sure how you plan to pass `va_list` you get inside `_my_logging_func` over to printf, other than interpolating the format string, first... in which case, it would look like `printf("%s", str)` which is equally possible with ESP_LOG...

amrsoll, 2023-08-07:
I actually don't have a practical example as of now. I had a workaround for my previous example similar to what you highlighted.

However, it would make it easier to build higher order utilities on top of esp_log, e.g. if you want to wrap `ESP_LOGx` to add other functionalities

igrr, 2023-08-08:
I'm not sure simply allowing the format string to be a const char* is enough.

It looks like you will need something like this to wrap ESP_LOGx with a function (rather than a macro):

```c++
void custom_log(const char* fmt, ...) {
  va_list list;
  va_start(list, fmt);
  char buf[BUF_SIZE];
  int len = vsnprintf(temp, sizeof(temp) - 1, fmt, list);
  if (len > sizeof(temp) - 1) {
    // handle truncation
  }
  ESP_LOGI(TAG, "%s", buf);
  va_end(list);
}
```

If you only use macros as wrappers around ESP_LOG, then you can forward the variable length argument list directly to ESP_LOG. But in the case of macro wrappers, there is no a problem with concatenation of format strings.

amrsoll, 2023-08-15:
You're right. In the end, the ESP_LOGx are not black magic and it wouldn't take a lot of effort to implement one's own flavour if I wanted them to behave differently.

If you could just put a disclaimer [in the doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html?highlight=esp_log#c.ESP_LOG_LEVEL). As of writing, it refers to the `printf ()` docs, and it creates a huge confusing stacktrace when trying to pass a `format` variable. I could make a PR too.

closed 2024-05-16 by KonstantinKondrashov