espressif/esp-idf #12010: examples/bluetooth/esp_ble_mesh/ble_mesh_wifi_coexist: How to adjust power levels?

rretanubun, 2023-08-03:
### Is your feature request related to a problem?

We are using the `ble_mesh_wifi_coexist` example program to obtain RF certification testing. We received a request to provide a method to set the power level of the WLAN/BLE radio.

I looked at the provided APIs here:
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=wifi%20low%20power#esp32-wi-fi-power-saving-mode
- https://github.com/espressif/esp-idf/blob/53ff7d43dbf/components/bt/esp_ble_mesh/api/core/include/esp_ble_mesh_low_power_api.h

Unless I am missing something, other than `esp_wifi_set_ps(WIFI_PS_MAX_MODEM)`  I do not see an obvious method to set the radio power level. Even then, based on the documentation, this will only influence the Wifi time slice of the coexistence operation mode.

### Describe the solution you'd like.

A console cli to `get and set` the `radio power levels for WLAN and BLE Radio` for this example project.

gnameg, 2023-08-07:
Pehaps, you can try to use this API
```
esp_err_t esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level);
```

MaxwellAlan, 2023-08-11:
Hi @rretanubun 
Do you want related to change the WLAN tx power in runtime?
Usually we can limit the max tx power by config the menuconfig->PHY->Max WiFi TX power.
Thanks.

rretanubun, 2023-08-14:
Hi alan, changing TX power at runtime for wifi will be ideal, in case you
know is there a reason it is only available via menu config (and thus
require a different build each time)? Thanks for your time!