espressif/esp-idf #12012: During Ethernet communication, the W5500 fails to acquire a lock in w5500_lock() and triggers a reboot.

imammy-hacomono, 2023-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

There is no reboot occurring due to the failure of spinlock_acquire().

### What is the actual behavior?

I am using the [esp-aws-iotlibrary](https://github.com/espressif/esp-aws-iot/tree/202210.01-LTS-release) in combination with esp-idf to perform MQTT communication over Ethernet using the W5500 Ethernet chip.

The communication involves sending JSON data of a few tens of bytes to the MQTT broker approximately every one minute.
(The data is encrypted using TLS, so it may be larger than the original size.)

After leaving the system running for about one day, it encountered an assertion with the following log and triggered a reboot.

~~~
assert failed: spinlock_acquire spinlock.h:122 (lock->count == 0)

0x40081b86: panic_abort at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/panic.c:452
0x40087db5: esp_system_abort at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4008e409: __assert_func at /Users/keigo.imaizumi/esp/esp-idf/components/newlib/assert.c:81
0x4008a65a: spinlock_acquire at /Users/keigo.imaizumi/esp/esp-idf/components/esp_hw_support/include/spinlock.h:90
 (inlined by) xPortEnterCriticalTimeout at /Users/keigo.imaizumi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:501
0x40088525: vPortEnterCritical at /Users/keigo.imaizumi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:575
 (inlined by) xQueueSemaphoreTake at /Users/keigo.imaizumi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1671
0x40115afb: w5500_lock at /Users/keigo.imaizumi/esp/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:57
 (inlined by) w5500_read at /Users/keigo.imaizumi/esp/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:100
0x40115c7c: w5500_get_rx_received_size at /Users/keigo.imaizumi/esp/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:165
0x40116952: emac_w5500_alloc_recv_buf at /Users/keigo.imaizumi/esp/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:539
 (inlined by) emac_w5500_task at /Users/keigo.imaizumi/esp/esp-idf/components/esp_eth/src/esp_eth_mac_w5500.c:701
~~~

I am currently creating a separate [issue](https://github.com/espressif/esp-idf/issues/12003) using the same procedure. 
The only change made from this point is enabling Hardware MPI Acceleration.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

kostaond, 2023-08-04:
It fails on 
```c
assert(lock->count > 0 && lock->count < 0xFF);    // Bad count value implies memory corruption
```
https://github.com/espressif/esp-idf/blob/v5.1/components/esp_hw_support/include/spinlock.h#L90C9-L90C103

Therefore it could indicate some problem with memory. Could you please enable stack overflow detection -CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK, (https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack — link for C3 since you didn't mention which chip you use). You can also enable heap corruption detection (https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/heap_debug.html#heap-corruption-detection) and see if that exposes any issue.

You can also blindly try to increase stack size for `emac_w5500_task` by configuring `eth_mac_config_t` (https://github.com/espressif/esp-idf/blob/v5.1/components/esp_eth/include/esp_eth_mac.h#L407).

imammy-hacomono, 2023-08-07:
@kostaond 
[Here's the problem](https://github.com/espressif/esp-idf/issues/12003), and I changed the IDF version from "tag v5.1" to "release/v5.1", and also increased the stack size of the emac_w5500_task from 2048 to 8192.

Currently, the issue of not being able to lock and causing a reboot has not occurred. However, the following log still occurs, and communication becomes unavailable:

~~~
E (404317) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 10
~~~
At this state, even pinging the device does not work.

kostaond, 2023-08-07:
Have you tried to run your app with stack overflow detection?

```
E (404317) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 10
```

It is really strange failure which should not happen since the W5500 should discard runt frames. Therefore, it might be SPI or memory issue... Hard to say without being able to reproduce though.

imammy-hacomono, 2023-08-08:
@kostaond 
I have enabled stack overflow detection, but no relevant logs are being displayed.

By the way, do I need to implement vApplicationStackOverflowHook? I remember that if it is not implemented, a default function will be called, so I have not implemented it.

I will share my sdkconfig.

~~~
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.0 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_CPU_CORES_NUM=2
[... 1786 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=8192
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

~~~

imammy-hacomono, 2023-08-08:
@kostaond 

Here are my SPI device interface settings. The clock_speed_hz value might be too fast.

When I searched the internet, I found varying values like "25MHz" or "12MHz." I'm not sure what value I should set it to.

Could you advise on the appropriate value?

~~~
 spi_device_interface_config_t spi_devcfg = {
        .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
        .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
        .spics_io_num = GPIO_NUM_5,
        .mode = 0,
        .clock_speed_hz = 33 * 1000 * 1000,
        .queue_size = 20
    };
~~~

kostaond, 2023-08-10:
> I remember that if it is not implemented, a default function will be called, so I have not implemented it.

Yes, default function should be used in that case.

If you have properly designed your PCB, you can set up to 40 MHz as SPI frequency. Note that some additional SPI bus configuration might be required to address the real bus properties (e.g. `.input_delay_ns`).  I rethought your issue and this is probably not an issue of SPI since the issue is consistent. If it was SPI, the error would be more random (it would fail on read/writing different registers and it would appear sooner). The memory corruption issue is still under consideration though.

What chip do you use? Would it be possible you tried to run your app and to communicate via e.g. WiFi or other Ethernet  PHY (if you have access to a board with different PHY like ESP32 Ethernet devkit)?

kostaond, 2024-09-30:
Closing... Feel free to reopen or open a new issue if you gather more details about the issue.