espressif/esp-idf #12013: mdns does not work in AP mode.

felixcollins, 2023-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-698-gcc71308e2f-dirty

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

I expect that mdns requests from my windows 10 pc for host "amybeakbox" will be resolved to 192.168.4.1 when my pc is connected via wifi to the esp32 running in AP mode.

### What is the actual behavior?

The pc can not resolve the name, even though the requests are visible in the log. If the esp32 and pc are both connected to another wifi AP then the mdns resolution does work.

### Steps to reproduce.

1. Start esp32 wifi with no AP configured to connect to in STA mode.
2. esp32 runs as soft AP
3. pc tries to connect to "amybeakbox.local" or "amybeakbox" host http server.

### Debug Logs.

```plain
RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[444780]: 
    Q: DESKTOP-U6D7POA...local. ANY IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[444780]:
    Q: DESKTOP-U6D7POA...local. ANY IN

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[444790]: AUTHORITATIVE
    A: DESKTOP-U6D7POA.local. AAAA IN 60 [16] fe80:0000:0000:0000:9789:64f1:46b7:6288
    A: DESKTOP-U6D7POA.local. A IN 60 [4] 192.168.4.2

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[444810]: AUTHORITATIVE
    A: DESKTOP-U6D7POA.local. AAAA IN 60 [16] fe80:0000:0000:0000:9789:64f1:46b7:6288
    A: DESKTOP-U6D7POA.local. A IN 60 [4] 192.168.4.2

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[449390]: AUTHORITATIVE
    A: 2.4.168.192.in-addr.arpa. PTR IN FLUSH 120 [23] DESKTOP-U6D7POA...local.
    A: 8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa. PTR IN FLUSH 120 [2] DESKTOP-U6D7POA...local.
    A: DESKTOP-U6D7POA.local. A IN FLUSH 120 [4] 192.168.4.2
    A: DESKTOP-U6D7POA.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:9789:64f1:46b7:6288
    X: .2.4.168.192.in-addr.arpa. NSEC IN FLUSH 120 [6] .2.4.168.192.in-addr.arpa.  00 02 00 08
    X: .8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa. NSEC IN FLUSH 120 [6] .8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa.  00 02 00 08
    X: DESKTOP-U6D7POA...local. NSEC IN FLUSH 120 [8] DESKTOP-U6D7POA...local.  00 04 40 00 00 08
W (451746) WIFI: No access point configured
[... 159 lines cut ...]
I can not run on IDF 5 because I'm running ADF.

I set up the mdns both in the wifi init code immediately after calling esp_netif_init();. I also call it when an SYSTEM_EVENT_AP_STACONNECTED event fires.

The mdns set up code is here:
`void init_my_mdns(void)
{
	// initialize mDNS service
	ESP_LOGW(TAG, "starting MDNS");
	if (ESP_OK == ESP_ERROR_CHECK_WITHOUT_ABORT(mdns_init()))
	{
		ESP_LOGW(TAG, "Setting host name");
		// set hostname
		mdns_hostname_set("amybeakbox.local");
		// set default instance
		mdns_instance_name_set("My Local BeakBox");

		// add our services
		mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0);

		// NOTE: services must be added before their properties can be set
		// use custom instance for the web server
		mdns_service_instance_name_set("_http", "_tcp", "Local BeakBox Web Server");
	}
}`

felixcollins, 2023-08-04:
Found this issue https://github.com/espressif/esp-idf/issues/4369, but as far as I can tell this should be fixed in 4.4.4

gabsuren, 2023-08-15:
@felixcollins thank you for rising the issue. I tried to reproduce it by the steps you provided, please check the code below

```
static void initialise_mdns(void)
{
    char * hostname = "esp-32";

    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
    //set mDNS hostname (required if you want to advertise services)
    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
    ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);
    //set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set(EXAMPLE_MDNS_INSTANCE) );
        
    //structure with TXT records
    mdns_txt_item_t serviceTxtData[3] = {
        {"board", "esp32"},
        {"u", "user"},
        {"p", "password"}
    };

    //initialize service
    ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3) );

    //add another TXT item
    ESP_ERROR_CHECK( mdns_service_txt_item_set("_http", "_tcp", "path", "/foobar") );

ESP_ERROR_CHECK( mdns_service_txt_item_set_with_explicit_value_len("_http", "_tcp", "u", "admin", strlen("admin")) );
    free(hostname);
}

void wifi_init_sta_ap() { // Initialize esp32 to STA + AP mode
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &wifi_event_handler,
                                                        NULL,
                                                        NULL));
        
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        
    wifi_config_t sta_config = {
        .sta = {
            .ssid = "WIFI SSID to which connect",
            .password = "password",
        },
    };  
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config));
            
    wifi_config_t ap_config = {
        .ap = {
            .ssid = "MY_SSID",
            .password = "123456789",
            .max_connection = 1,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK,
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &ap_config));

    ESP_ERROR_CHECK(esp_wifi_start());
}

void app_main(void)
{       
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    initialise_mdns();
            
            
    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */ 
    
    wifi_init_sta_ap();
} 
```
After connection from PC to esp32 AP, I was able to resolve the hostname  by the following 
`dns-sd -G v4v6 esp32-mdns.local` 
command. 

Timestamp     A/R    Flags if Hostname                               Address                                      TTL
13:32:38.589  Add        2  5 esp32-mdns.local.                      192.168.4.1                                  120

I used `esp-idf:v4.4`

In order to move forward could you please try the code below and let us know if it works or not for you? 

Regards,
Suren

felixcollins, 2023-08-17:
Hi Suren, Sorry, I was not very clear in my report. I am talking about running the ESP32 in exclusively AP mode. The client device will have to connect to the ESP32 wifi AP before making the MDNS request. I am trying to create a reproduction using your code but some of it is missing (event_handler and wifi_event).  Also, in the initialise_mdns() function the last line is wrong "free(hostname);" as hostname was not heap allocated.

felixcollins, 2023-08-17:
Okay, I got the code sample running by pasting it into my main and modifying it. I also turned on MDNS logging in the components\mdns\private_include\mdns_private.h file. When I run "C:\dns-sd -G v4v6 esp-32.local" it never returns. I do see activity in the ESP32 log though. I'm not a DNS expert so I'm not sure what to look for. My setup is very similar to this example. For my setup I get mdns name resolution when both devices are on the same (third device) AP. When a device is connected to ESP32 soft AP then MDNS fails.

IDF version from log `I (340) cpu_start: ESP-IDF:          v4.4.5-249-gcc71308e2f-dirty`

Here is the relevant code

`
static void initialise_mdns(void)
{
	char *hostname = "esp-32";

	// initialize mDNS
	ESP_ERROR_CHECK(mdns_init());
	// set mDNS hostname (required if you want to advertise services)
	ESP_ERROR_CHECK(mdns_hostname_set(hostname));
	ESP_LOGI(TAG, "mdns hostname set to: [%s]", hostname);
	// set default mDNS instance name
	ESP_ERROR_CHECK(mdns_instance_name_set("Test MDNS"));

	// structure with TXT records
	mdns_txt_item_t serviceTxtData[3] = {
			{"board", "esp32"},
			{"u", "user"},
			{"p", "password"}};

	// initialize service
	ESP_ERROR_CHECK(mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3));

	// add another TXT item
	ESP_ERROR_CHECK(mdns_service_txt_item_set("_http", "_tcp", "path", "/foobar"));

	ESP_ERROR_CHECK(mdns_service_txt_item_set_with_explicit_value_len("_http", "_tcp", "u", "admin", strlen("admin")));
	//    free(hostname);
}

const char *system_event_strings[] =
		{
				"SYSTEM_EVENT_WIFI_READY ",
				"SYSTEM_EVENT_SCAN_DONE",
				"SYSTEM_EVENT_STA_START",
				"SYSTEM_EVENT_STA_STOP",
				"SYSTEM_EVENT_STA_CONNECTED",
				"SYSTEM_EVENT_STA_DISCONNECTED",
				"SYSTEM_EVENT_STA_AUTHMODE_CHANGE",
				"SYSTEM_EVENT_STA_GOT_IP",
				"SYSTEM_EVENT_STA_LOST_IP",
				"SYSTEM_EVENT_STA_BSS_RSSI_LOW",
				"SYSTEM_EVENT_STA_WPS_ER_SUCCESS",
				"SYSTEM_EVENT_STA_WPS_ER_FAILED",
				"SYSTEM_EVENT_STA_WPS_ER_TIMEOUT",
				"SYSTEM_EVENT_STA_WPS_ER_PIN",
				"SYSTEM_EVENT_STA_WPS_ER_PBC_OVERLAP",
				"SYSTEM_EVENT_AP_START",
				"SYSTEM_EVENT_AP_STOP",
				"SYSTEM_EVENT_AP_STACONNECTED",
				"SYSTEM_EVENT_AP_STADISCONNECTED",
				"SYSTEM_EVENT_AP_STAIPASSIGNED",
				"SYSTEM_EVENT_AP_PROBEREQRECVED",
				"SYSTEM_EVENT_ACTION_TX_STATUS",
				"SYSTEM_EVENT_ROC_DONE",
				"SYSTEM_EVENT_STA_BEACON_TIMEOUT",
				"SYSTEM_EVENT_FTM_REPORT",
				"SYSTEM_EVENT_GOT_IP6",
				"SYSTEM_EVENT_ETH_START",
				"SYSTEM_EVENT_ETH_STOP",
				"SYSTEM_EVENT_ETH_CONNECTED",
				"SYSTEM_EVENT_ETH_DISCONNECTED",
				"SYSTEM_EVENT_ETH_GOT_IP",
				"SYSTEM_EVENT_ETH_LOST_IP",
};
[... 267 lines cut ...]
    X: DESKTOP-U6D7POA...local. NSEC IN FLUSH 120 [8] DESKTOP-U6D7POA...local.  00 04 40 00 00 08

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[230330]:
    Q: esp-32.local. A IN
    Q: esp-32.local. AAAA IN

TX[1][0]: To: 224.0.0.251:5353, Packet[230450]: AUTHORITATIVE
    A: esp-32.local. A IN FLUSH 120 [4] 192.168.4.1

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[237090]: AUTHORITATIVE
    A: 2.4.168.192.in-addr.arpa. PTR IN FLUSH 120 [23] DESKTOP-U6D7POA...local.
    A: 8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa. PTR IN FLUSH 120 [2] DESKTOP-U6D7POA...local.
    A: DESKTOP-U6D7POA.local. A IN FLUSH 120 [4] 192.168.4.2
    A: DESKTOP-U6D7POA.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:9789:64f1:46b7:6288
    X: .2.4.168.192.in-addr.arpa. NSEC IN FLUSH 120 [6] .2.4.168.192.in-addr.arpa.  00 02 00 08
    X: .8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa. NSEC IN FLUSH 120 [6] .8.8.2.6.7.B.6.4.1.F.4.6.9.8.7.9.0.0.0.0.0.0.0.0.0.0.0.0.0.8.E.F.ip6.arpa.  00 02 00 08
    X: DESKTOP-U6D7POA...local. NSEC IN FLUSH 120 [8] DESKTOP-U6D7POA...local.  00 04 40 00 00 08

RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[473840]:
    Q: esp-32.local. A IN
    Q: esp-32.local. AAAA IN

TX[1][0]: To: 224.0.0.251:5353, Packet[473950]: AUTHORITATIVE
    A: esp-32.local. A IN FLUSH 120 [4] 192.168.4.1
`

chipweinberger, 2023-08-20:
I used to have the same problem, but it works in v5.1 these days.

felixcollins, 2023-08-20:
@chipweinberger I am stuck using the ADF so no support for IDF 5.1 yet. I wonder if the fix (if there is one) could be back ported to 4.4

gabsuren, 2023-08-21:
@chipweinberger thank you for the log. From it I see 
`
RX[1][0]: From: 192.168.4.2:5353, To: 224.0.0.251, Packet[230330]:
    Q: esp-32.local. A IN
    Q: esp-32.local. AAAA IN
`

`TX[1][0]: To: 224.0.0.251:5353, Packet[230450]: AUTHORITATIVE
    A: esp-32.local. A IN FLUSH 120 [4] 192.168.4.1
    `
    
It means you are requesting to resolve hostname `esp-32.local` from PC (`192.168.4.2`)and devices resolves your answer to `192.168.4.1` which is your `esp32` device running on AP mode, right?

So in which case you don't see the answer? According to your log it sends answer correctly. 
Could you please provide more details?

felixcollins, 2023-08-21:
@gabsuren The log was actually from me. I have tried again. I small experiment - I removed the code to make the esp32 run in APSTA mode and instead just run as AP. Running like this, I get the following from dns-sd as I connect and disconnect from the esp32 AP on my windows PC. This is what I would expect, so it is working for AP mode Yay!

```
C:\Users\felix>dns-sd -G v4 esp-32.local
Timestamp     A/R Flags if Hostname                  Address                                      TTL
11:11:14.804  Add     2 18 esp-32.local.             192.168.4.1                                  120
11:12:13.382  Rmv     0 18 esp-32.local.             192.168.4.1                                  0
11:12:27.281  Add     2 18 esp-32.local.             192.168.4.1                                  120
11:14:37.048  Rmv     0 18 esp-32.local.             192.168.4.1                                  0

```

However, when I reenable the APSTA mode, connect to the esp32 AP and try dns-sd again I get:
```
C:\Users\felix>dns-sd -G v4 esp-32.local
```

...nothing.

So it seems that something is going wrong with the response when both AP and STA are enabled on esp32. Does that give you a clue? I'll change the title of the issue to reflect this new information.

felixcollins, 2023-08-21:
Okay, I was a bit hasty, I can't actually reproduce that success. It seems like either dns-sd util on windows is unreliable or more likely the mdns system in windows is not reliable. I'll try and query from a linux box.

felixcollins, 2023-08-22:
I'm closing this as I can not reliably test that the mdns server is working or not.

closed 2023-08-22 by felixcollins

gabsuren, 2023-08-22:
@felixcollins Just one last hint (probably not the issue, but worth checking): 
Did you check your Wi-Fi connection from your PC to the ESP access point during the request? Did the connection remain stable throughout the request?"