espressif/esp-idf #12014: memory leak in wifi init-start-stop-deinit calls

AxelLin, 2023-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-480-g79b1379662

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

No memory leak with wifi init-start-stop-deinit calls.

### What is the actual behavior?

Use getting_started/station/main/station_example_main.c with below modified code:

The change is adding a while loop to repeat wifi init-start-stop-deinit calls.
Also show free memory via esp_get_free_heap_size(), it shows the symptom of a small memory leak in each iteration.
[wifi-memory-leak.txt](https://github.com/espressif/esp-idf/files/12256592/wifi-memory-leak.txt)

### Steps to reproduce.

Just run example code with above modification.

AxelLin, 2023-08-06:
v4.4 branch does not has this memory leak.
This is a regression in v5.0, v5.1, master branches.

Xiehanxin, 2023-08-08:
hi， @AxelLin I have seen that it is fixed in latest master. I have tested and find that there is no memory leak. and it might be backpoer to V4.4. Please try the latest master

AxelLin, 2023-08-08:
> hi， @AxelLin I have seen that it is fixed in latest master. I have tested and find that there is no memory leak. and it might be backpoer to V4.4. Please try the latest master

I don't get your point. There is no problem with v4.4.
The issue happens on v5.0+.
If it is fixed in master, could you share the commit id?

mhdong, 2023-08-08:
@AxelLin 691eed15ccf91b367df8935bd9529a19c3e80720

mhdong, 2023-08-08:
Hi @AxelLin  If there is still a memory leak issue, please provide the log and corresponding commit information. We will reproduce it as soon as possible and try our best to fix the issue.

AxelLin, 2023-08-08:
> Hi @AxelLin v5.1-480-g79b1379662 has memory leak issue caused by assoc IE. We have fixed this issue recently. I have tried to reproduce using 5aa9881478 and confirm that there is no issue.

Thanks for the update.
I just realize I stop testing master due to another issue (https://github.com/espressif/esp-idf/issues/11883)  in master tree.
At the moment 5aa9881478 is not yet available in github v5.1 branch, so I think I have to wait for the fix.

closed 2023-08-15 by AxelLin